blackmagic-debug/blackmagic #955: STM32L1 stuck in WFI after flash error

tomasvr, 2021-12-03:
Dear community,
I have been using a BMP to program and debug the stm32l162 processor with GDB. Most of the time this works without problems, but now I have ran into a problem during flashing which has left the mcu unresponsive. 

I was trying to flash new firmware and during the load phase the following error occured:
`Error writing data to flash`

This has happened before, but usually I would just try to flash again and that would work.

This time however, when I retry to flash firmware onto the mcu I keep getting the following response:

```
Target voltage: 1.79V
Timeout during scan. Is target stuck in WFI?
Timeout during scan. Is target stuck in WFI?
SW-DP scan failed!
gdb/gdb.gdb:3: Error in sourced command file:
.gdb/target:4: Error in sourced command file:
Attaching to Remote target failed
```

UweBonnes, 2021-12-03:
Please compile hosted. First run "blackmagic -t", If it fails, run, post the output. If it syucceeds, try to flash "blackmagic <bla.bin>.  Also state if WFI/WFE or deep sleep is involved, or a remap of the SWJ pins.

tomasvr, 2021-12-03:
Sorry, I accidentally pressed enter while my post was still incomplete and you had already responded. 

I do make use of deep sleep and WFI. My first thought was also that the MCU might immediately enter WFI after boot which means it would be unreachable for the probe. Thus I tried to enable the `connect_srst` option but that had no effect. Furthermore I do not see any output on my serial debug line so I suspect the application is not being run at all. 
Lastly, I am not aware of any SWJ pin remapping.

I will try your suggestion, thank you!

UweBonnes, 2021-12-03:
Just a remainder: connect_srst needs NRST connected. What BMP platforms do you use?

dragonmux, 2024-10-10:
👋🏼 Apologies that this one appears to have gotten away from us. We believe that our issue may have been fixed in some of the work we did in v1.10 or since on `main` during the overhaul of the STM32L0/L1 support. Please give the latest firmware a test and let us know how you get on.

dragonmux, 2024-11-09:
Given it's been a month and we've still not heard anything, we're going to close this as fixed. If it turns out it is not fixed, please let us know and we'll re-open the issue.

closed 2024-11-09 by dragonmux