blackmagic-debug/blackmagic #961: How to Change SSID & PASSWORD of blackmagic?

Kothiyaparth, 2021-12-26:
I want to change default SSID name and password how can I do this and how to rebuild it?

DX-MON, 2021-12-26:
If you're taking about the Blackmagic Design cards, and one which has WiFi then you have come to the wrong repo.
This repo is for the BlackMagic Probe JTAG-based debugger which has no affiliation with Blackmagic Design.

UweBonnes, 2021-12-26:
There are also some WIFI enabled debuggers, based on BMP, but with a different codebase. Get in contact with those projects.

esden, 2021-12-27:
@Kothiyaparth if you are referring to [ctxLink](https://www.crowdsupply.com/sid-price/ctxlink), you will have to ask @sidprice, he maintains his probes [firmware separately](https://github.com/sidprice/blackmagic/tree/ctxLink_master) and you will have to contact him. Or if you are talking about [other an ESP32](https://github.com/Ebiroll/esp32_blackmagic) you should also contact those specific projects.

In any case. As this does not seem to be related to the official firmware that we are maintaining here I will close this issue. If we made a mistake here, please feel free to comment and/or reopen this issue. :)

closed 2021-12-27 by esden