blackmagic-debug/blackmagic #963: Add support for SAMS70 (C-M7)

perigoso, 2021-12-29:
Just showing interest for a particular family i use, and cortex-m7 support in general (no rush)

https://www.microchip.com/en-us/product/ATSAMS70N19

Dev boards for this are few, expensive, and probably out of stock, but I have a custom board and couple spare ICs if needed

Open to help support this

UweBonnes, 2021-12-29:
Start by posting the output of bmp/hosted "blackmagic -t".Debugging is already supported. Study the flashing algorithm and compare it to the already implemented algos.

esden, 2022-01-08:
Hi @perigoso this is an interesting addition.

It would be nice to have an eval board for a chip from that family added to the test collection. They seem to be available and the E70 eval board is around $150. A decent way to get those for free is if you talk to the sales rep from Microchip. I might look into contacting them to get one for our testing purposes.

perigoso, 2022-01-08:
Getting an eval from them would be great, they are pretty expensive, as a last resort the [boards](https://github.com/openinput-fw/big-sammy) I used these for are OSH and I have spares and chips

esden, 2022-01-08:
That is definitely an option. But as far as I can see this board only exposes SWD interface and not the full JTAG interface. For testing purposes it would be good to have both accessible so that we can make sure that both work. Also it would be nice to have them on a standard Cortex-Debug connector so it is easy to leave hooked up to the HITL CI system in the future.

perigoso, 2022-01-08:
Ah, yes it only exposes SWD, and the connector is inconvenient for the HITL CI for sure

UweBonnes, 2022-01-09:
As I read the datasheet, SAMS70 only has one JTAG TAP used for boundary scan. Debug only happens via SWD.

perigoso, 2022-05-20:
@esden did you ever reach out to microchip for a dev board?

esden, 2022-05-20:
Thanks for the reminder. Sorry, I did not come around to reaching out to them. I will try doing that.

perigoso, 2022-05-21:
Ok first run on it
```
[perigoso@perigoso-desktop blackmagic]$ ./src/blackmagic -t
INFO: Open USB 1532:0091 class  0 failed
BMP hosted v1.7.1-416-g29dc94c
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 0669FF564949775087212225 Black Sphere Technologies
 Black Magic Probe (STLINK/V2) v1.7.1-415-gc4869a5
Running in Test Mode
Target voltage: 3.30V Volt
Speed set to  3.2727 MHz for SWD
DPIDR 0x0bd11477 (v1 MINDP rev0)
Timeout on read RESP
remote_adiv5_low_access error -4
Timeout on read RESP
remote_adiv5_dp_read error -4
DEBUG Power-Up failed
Scan SWD failed, trying JTAG!
Timeout on read RESP
Update Firmware to allow to set max SWJ frequency
Timeout on read RESP
Device has fixed frequency for JTAG
Resetting TAP
Timeout on read RESP
Timeout on read RESP
Change state to Shift-IR
Timeout on read RESP
Scanning out IRs
Timeout on read RESP
jtag_scan: Sanity check failed: IR[0] shifted out as 0
Can not attach to target 1
```

This is  blackmagic hosted on a blackmagic probe, but maybe I should start working on blackmagic hosted with a probe that supports this device?

perigoso, 2022-05-21:
ok I need some guidance here, how do I go about connecting to a ft2232h as a probe with blackmagic hosted? I have it configured as a swd debugger with the resistor shebang, it works by itself as its what I usually use to develop on the platform

I tried calling blackmagic with a bunch of different "cable" options, but it seems to always exit silently

UweBonnes, 2022-05-21:
Please  copy in ftdi_bmp.c the ft232h_resistor_swd section  as ft2232h_resistor_swd, adapte the PID, recompile, try and report.

dragonmux, 2022-05-21:
BMP Debug App on BMP native is actually picking up the device ("DPIDR 0x0bd11477 (v1 MINDP rev0)"), ignore the switch to JTAG as this is something that's getting patched out in v1.9.

As for the FT2232H probe, could you please share some the command lines you've tried to BMPDA? Could you also share the result of trying `src/blackmagic -lv 15` so we can get a better idea of what's going on

UweBonnes, 2022-05-21:
Right, as  DPIDR is picked up, basic SWD communication works. Perhaps NRST disturbs things, eventually keeping NRST low or some extended_reset procedure is needed as with other SAM devices

perigoso, 2022-05-21:
Will play with this later today

perigoso, 2022-05-21:
ok, I tried to add 2232 to the cable list, but it seems nothings happens still, I also noticed other cables in the list had the correct configurations, so I tried with the "usbmate" too

the device is present: 
`Bus 003 Device 015: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC`

and in the cable it has the correct VID/PID, this is the "usbmate":
```
.vendor  = 0x0403,
.product = 0x6010,/*FT2232H*/
```

Logs:
```
[perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -l
Generic FTDI MPSSE VID/PID found. Please specify exact type with "-c <cable>" !
No usable debugger found
```
```
[perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cft2232h_resistor_swd
Generic FTDI MPSSE VID/PID found. Please specify exact type with "-c <cable>" !
```
```
[perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cusbmate -t
```
```
[perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cusbmate -l
No usable debugger found
```

Note I'm cheched out to master
`v1.7.1-416-g29dc94c`
29dc94c3d619c388b06e6b0e09d2e4cb07a0ae34

perigoso, 2022-05-21:
anyway, what should I start with here, I'm assuming my first step would be to get the device to attach to swd successfully, do I need to setup some probe procedure or so? should i try to capture a successful attach from another debugger to see what they do?

UweBonnes, 2022-05-22:
The FTDI has no advantage  and your setup additional does not pick it up. So stay with the BMP probe and post a more verbose log.

dragonmux, 2022-05-22:
> ok, I tried to add 2232 to the cable list, but it seems nothings happens still, I also noticed other cables in the list had the correct configurations, so I tried with the "usbmate" too
>
> the device [...]
There needs to be a space between the -c option and its parameter, same with -v and 31 - this is a limitation of the option parser.
> ```
> [perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cusbmate -t
> ```
>
> ```
> [perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cusbmate -l
> No usable debugger found [...]

dragonmux, 2022-05-22:
With regards to getting started, yes, your plan of attack starting with getting SWD scan working is about what we'd do too. Most likely it needs a probe_sams70 type routine written - probably modeled on the probe_samd routines - and suitable Flash support code too

perigoso, 2022-05-22:
> @perigoso
> [...]
> > ```
> > [perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -l
> > Generic FTDI MPSSE VID/PID found. Please specify exact type with "-c <cable>" !
> > No usable debugger found
> > ```
> > 
> > ```
> > [perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cft2232h_resistor_swd
> > Generic FTDI MPSSE VID/PID found. Please specify exact type with "-c <cable>" !
> > ```
> There needs to be a space between the -c option and its parameter, same with -v and 31 - this is a limitation of the option parser.
> > 
> > ```
> > [perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cusbmate -t
> > ```
> > 
> > ```
> > [perigoso@perigoso-desktop blackmagic]$ sudo ./src/blackmagic -v31 -cusbmate -l
> > No usable debugger found
> > ```
> [...]
> 

I did try with spaces between the options before, to see of it made a difference, but that is good to know

Syncena, 2022-05-23:
Just a quick heads-up that devices in the SAMx70 series have more in common with the SAM4S series (supported in src/target/sam3x.c) than SAMD. For example, there's no DSU, so no need to fiddle with nRST on connect.

You'll also find the Flash controller is pretty much identical to that found on SAM4S.

perigoso, 2022-05-23:
Thanks, i was aware of some similarities with the sam3u for example, as I've played with the flash controller and usb peripheral on both and they similar indeed, i don't remember much of the samd

dragonmux, 2022-05-23:
Thanks for the extra information Syncena - hadn't yet delved into that unlike the SAMD code which is why we suggested what we knew as a basis, though it doesn't much change our advice, only where to look. We look forward to seeing Perigoso's contribution

perigoso, 2022-05-23:
ok here's a verbose log with the BMP
```
[perigoso@perigoso-desktop blackmagic]$ ./src/blackmagic -v 31 -t
INFO: Open USB 046d:c31d class  0 failed
BMP hosted v1.7.1-416-g29dc94c-dirty
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 0669FF564949775087212225 Black Sphere Technologies
 Black Magic Probe (STLINK/V2) v1.7.1-415-gc4869a5
+#!GA#
       K(STLINK/V2) v1.7.1-415-gc4869a5
Remote is (STLINK/V2) v1.7.1-415-gc4869a5
!GZ0#
       K0
Running in Test Mode
 !GV#
       K3.30V
Target voltage: 3.30V Volt
!GF003d0900#
       K0
!Gf#
       K17f03100
!Gf#
       K17f03100
Speed set to  3.2727 MHz for SWD
remote_swdptap_init
!SS#
       K0
swdptap_seq_out        32 ticks: ffffffff
!So20ffffffff#
       K0
swdptap_seq_out        32 ticks: ffffffff
!So20ffffffff#
       K0
swdptap_seq_out        32 ticks: 6209f392
!So206209f392#
       K0
swdptap_seq_out        32 ticks: 86852d95
!So2086852d95#
       K0
swdptap_seq_out        32 ticks: e3ddafe9
!So20e3ddafe9#
       K0
swdptap_seq_out        32 ticks: 19bc0ea2
!So2019bc0ea2#
       K0
swdptap_seq_out        12 ticks: 000001a0
!So0c1a0#
       K0
swdptap_seq_out        32 ticks: ffffffff
!So20ffffffff#
       K0
swdptap_seq_out        32 ticks: 0fffffff
!So20fffffff#
       K0
swdptap_seq_out         8 ticks: 000000a5
!So08a5#
       K0
!Si03#
       K1
swdptap_seq_in          3 ticks: 00000001
!SI20#
       Kbd11477
swdptap_seq_in_parity  32 ticks: 0bd11477 OK
DPIDR 0x0bd11477 (v1 MINDP rev0)
!HC#
       K1
!Hd00ff0004#
       K200000f0
dp_read addr f0000020: 00000004
Read  CTRL/STAT: 0xf0000020
Write CTRL/STAT: 0xf0000020
!HL00000004f0000020#
Timeout on read RESP
remote_adiv5_low_access error -4
!Hd00ff0004#
Timeout on read RESP
remote_adiv5_dp_read error -4
dp_read addr ff001d: 00000004
Read  CTRL/STAT: 0x00ff001d
DEBUG Power-Up failed
Scan SWD failed, trying JTAG!
!GF003d0900#
Timeout on read RESP
Update Firmware to allow to set max SWJ frequency
!Gf#
Timeout on read RESP
Device has fixed frequency for JTAG
Resetting TAP
+#!JS#
Timeout on read RESP
+#!JR#
Timeout on read RESP
Change state to Shift-IR
!JT043#
Timeout on read RESP
Scanning out IRs
!JN01#
Timeout on read RESP
jtag_scan: Sanity check failed: IR[0] shifted out as 0
Can not attach to target 1
```

perigoso, 2022-05-23:
```
[perigoso@perigoso-desktop blackmagic]$ ./src/blackmagic -t
INFO: Open USB 046d:c31d class  0 failed
BMP hosted v1.7.1-416-g29dc94c-dirty
 for ST-Link V2/3, CMSIS_DAP, JLINK and LIBFTDI/MPSSE
Using 1d50:6018 0669FF564949775087212225 Black Sphere Technologies
 Black Magic Probe (STLINK/V2) v1.7.1-415-gc4869a5
Running in Test Mode
Target voltage: 3.30V Volt
Speed set to  3.2727 MHz for SWD
DPIDR 0x0bd11477 (v1 MINDP rev0)
RESET_SEQ failed
AP   0: IDR=04770041 CFG=00000000 BASE=e00fd003 CSW=03000040 (AHB-AP var4 rev0
Halt via DHCSR: success 01030003 after 0ms
ROM: Table BASE=0xe00fd000 SYSMEM=0x00000001, designer  1f Partno   0
ROM: Table BASE=0xe00fe000 SYSMEM=0x00000001, designer 43b Partno 4c8
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c7
  0 0xe000e000: Generic IP component - Cortex-M4 SCS (System Control Space) (PIDR = 0x04000bb00c  DEVTYPE = 0x00 ARCHID = 0x0000)  -> cortexm_probe
Silicon bug: Single stepping will enter pending exception handler with this M7 core revision!
CPUID 0x410fc271 (M7 var 0 rev 1)
  1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
  2 0xe0002000: Generic IP component - Cortex-M7 FBP (Flash Patch and Breakpoint) (PIDR = 0x04000bb00e  DEVTYPE = 0x00 ARCHID = 0x0000)
  3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04000bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
  4 Entry 0xfff41002 -> Not present
  5 Entry 0xfff42002 -> Not present
  ROM: Table END
 1 0xe0041000: Debug component - Cortex-M7 ETM (Embedded Trace) (PIDR = 0x04000bb975  DEVTYPE = 0x13 ARCHID = 0x4a13)
 2 Entry 0xfff44002 -> Not present
 3 Entry 0x1ff02002 -> Not present
 ROM: Table END
1 0xe0040000: Debug component - Cortex-M7 TPIU (Trace Port Interface Unit) (PIDR = 0x04000bb9a9  DEVTYPE = 0x11 ARCHID = 0x0000)
2 Entry 0x1ff03002 -> Not present
3 Entry 0x1ff03002 -> Not present
ROM: Table END
***  1      SAMS70J19A M7
RAM   Start: 0x20000000 length = 0x40000
Flash Start: 0x00400000 length = 0x80000 blocksize 0x1000
```

perigoso, 2022-05-23:
ah dam, i forgot gdb is borked, here I was perplexed at the core dumps (I'm on gdb 12.1)

UweBonnes, 2022-05-24:
What was the trick/pitfall that now detection proceeds? Before, detection hanged after reading DPIDR, now it proceeds.

perigoso, 2022-05-24:
well before It didn't know this particular target, I honestly don't know what it tried to after failing to probe it, not yet familiar enough with the code base

perigoso, 2022-05-24:
I basically just added a probe method for this family and it seemed to behave

dragonmux, 2022-05-25:
We'll get this reviewed later today. We're happy that this should work as intended and at the very least can be improved on later if needed.

PR #1021 (2022-06-01): target: add samx7x probe method
adds support for SAME70, SAMS70, SAMV71 and SAMV70

closes #963
closed 2022-06-01 by dragonmux