blackmagic-debug/blackmagic #966: Tags are not annotated tags

schodet, 2022-01-04:
Regarding 0ae65cc10fd2667d4c945c233c5ce35a84f416c5

Tags used for releases are supposed to be annotated tags, it was the case before 1.7, you can see the difference with git show:

```
git show v1.6.1
git show v1.7
```

You can use `git tag -a` to make annotated tags.

schodet, 2022-01-04:
This is confusing when running `git describe` and not having the same result as `mon version`.

esden, 2022-01-04:
Thanks for noticing. I will look into it tomorrow and correct the issue.

esden, 2022-01-06:
Ok, I have created annotated tags. I hope this fixes this issue. Can you confirm?

UweBonnes, 2022-01-06:
How should the change be visible? ```> git fetch upstream
t520:~/devel/blackmagic/src> git describe upstream/master
v1.6.1-917-g5c07d617```

DX-MON, 2022-01-06:
you have to `git pull --tags`/`git fetch --tags` to update your tags @UweBonnes

esden, 2022-01-06:
Thanks @DX-MON The tags are a bit tricky. They don't update as easily on their own when you pull. Some other project decided to stop using tags because of that. We will need to be much more careful creating new tags going down the road or find a different solution. None the less I hope the `git describe` is fixed. At least it is on my test machines.
```
❯ git describe
v1.7.1-335-g5c07d61
```

schodet, 2022-01-06:
It’s difficult to update a tags because clients will not update them, but it will work for new clones, and for future tags.

```
nicolas@boktor:~/work/stm32/blackmagic% git fetch origin 
nicolas@boktor:~/work/stm32/blackmagic% git fetch --tags origin
remote: Enumerating objects: 2, done.
remote: Counting objects: 100% (2/2), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 2 (delta 0), reused 2 (delta 0), pack-reused 0
Unpacking objects: 100% (2/2), 282 bytes | 282.00 KiB/s, done.
From https://github.com/blacksphere/blackmagic
 ! [rejected]        v1.7       -> v1.7  (would clobber existing tag)
 ! [rejected]        v1.7.1     -> v1.7.1  (would clobber existing tag)
nicolas@boktor:~/work/stm32/blackmagic% git tag -d v1.7 v1.7.1
Deleted tag 'v1.7' (was c5a0271)
Deleted tag 'v1.7.1' (was 0ae65cc)
nicolas@boktor:~/work/stm32/blackmagic% git fetch origin       
From https://github.com/blacksphere/blackmagic
 * [new tag]         v1.7       -> v1.7
 * [new tag]         v1.7.1     -> v1.7.1
nicolas@boktor:~/work/stm32/blackmagic% git describe    
v1.7.1-335-g5c07d61
nicolas@boktor:~/work/stm32/blackmagic%
```

esden, 2022-01-06:
Thank you @schodet for confirming. I am glad that the tags are correctly set now. Thus I am closing this.

closed 2022-01-06 by esden

UweBonnes, 2022-01-06:
How to proceed with the next tagged release for people following git? Should " git fetch --tags origin" or something similar be called from the Makefile "version.h" target?

schodet, 2022-01-06:
Just use annotated tags with `git tag -a`, then do as usual.

When users will fetch, they will get the tags too:

> By default, any tag that points into the histories being fetched is also fetched; the effect is to fetch tags that point at branches that you are interested in.