blackmagic-debug/blackmagic #968: Cortex M DHCSR.C_MASKINTS may be handled incorrectly

pavel-kirienko, 2022-01-05:
I suspect that Cortex M's bit `DHCSR.C_MASKINTS` is subtly mishandled, which leads to unstable behavior if semihosting call `SYS_EXIT` (and its extended counterpart) is used. I must admit here that I don't fully comprehend what's happening under the hood yet but I want to report this anyway because this issue had cost me a few days of debugging and tearing my hair off.

I am working on a baremetal application based on STM32F446 that heavily relies on semihosting (it is actually a unit testing suite, not that it matters here). At the end of the execution, the application makes the final semihosting call `SYS_EXIT` with the appropriate exit code indicating whether the test suite is executed successfully. Everything was functioning normally until I had to add a peripheral interrupt handler to the application, whereupon I discovered that under some mysterious circumstances the handler might not be invoked despite the appropriate pending/enable flags being set in the NVIC registers. I was never able to find a robust way of reproducing this problem: sometimes it would work reliably across power cycling and software resets, sometimes it would fail. Fast forward a few days, the culprit has been tracked down to `DHCSR.C_MASKINTS`; here's my best understanding of what is happening.

First, I power on the board, upload and launch the application:

```gdb
shell ln -fs /dev/serial/by-id/usb-*Black_Magic_Probe*-if00 .gdb_port.tmp
target extended-remote .gdb_port.tmp
monitor connect_srst enable
monitor swdp_scan
attach 1
set mem inaccessible-by-default off
set remote system-call-allowed 1
set confirm off
load
compare-sections
run
```

The versions involved are:

- GDB: `GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git`
- Black Magic (native): `Black Magic Probe v1.7.1-131-g560a046, Hardware Version 3`

The original state of DHCSR after a power-on reset before the application is started is reliably observed as 0x30003, which makes sense.

Now, wait for the application to finish. The first run after a power-on reset appears to always succeed (I can't be 100% sure due to the sporadic nature of the problem though).

As I wrote above, at the end of the execution the application calls SYS_EXIT. The handler of that system call [invokes `cortexm_halt_resume` with step=true](https://github.com/blacksphere/blackmagic/blob/5c07d6170fce8404ffd5cf841eb5ec9a1e36c222/src/target/cortexm.c#L1612), which causes the debugger to set `DHCSR.C_MASKINTS`.

The debugger is disconnected at this point --- never physically, but logically, by way of exiting GDB.

Next, without changing anything, I re-run the suite. The debugger is attached again (successfully), but this time occasionally you might observe the original state of DHCSR to be not 0x30003 (where C_MASKINTS is reset) but 0x3000B (where C_MASKINTS is set but C_STEP is not). If DHCSR reads as 0x3000B, the application is also able to function except that the interrupt handlers are masked, which is an error because `run` is expected to unmask interrupts.

Here are the most relevant excerpts from the arm-v7m tech reference manual:

[image]

[image]

What I suspect might be happening is that after SYS_EXIT, the target is left with C_MASKINTS=1. When the debugger is reattached later, the new value is written to DHCSR that changes C_MASKINTS bypassing due sequence, triggering the unpredictable behavior that happens to leave C_MASKINTS set only occasionally. If this is correct, then a possible remedy is to read DHCSR before the first write to ensure the bits are not modified until C_HALT & C_DEBUGEN are guaranteed to be set. I would perhaps also harden `cortexm_halt_resume()` against the risk of state divergence between `t->stepping` and the actual bits stored in the register by always fetching the current value before writing a new one.

While googling around I stumbled upon this random tweet that seems to provide a plausible explanation, but I am not sure if it's valid for Black Magic: https://mobile.twitter.com/azonenberg/status/1162402192793358341

As a tentative workaround, I made a script that resets this bit at every opportinity -- it appears to be effective:

```gdb
define clear_dhcsr_c_maskints
    set {unsigned}0xE000EDF0 = 0xA05F0000 | ({unsigned}0xE000EDF0 & 0b1111111111110111)
    p/x {unsigned}0xE000EDF0
end

catch signal all
commands
    clear_dhcsr_c_maskints
end

define hook-run
    clear_dhcsr_c_maskints
end

define hook-continue
    clear_dhcsr_c_maskints
end
```

UweBonnes, 2022-01-05:
Dear Pavel,
v1.7.1-131 is no so recent, recent git head is at about 370. A lot has been changes around DHCSR, but probably not regarding MASKINT. However lately it was notices that reset   CORTEXM_DHCSR  needs first CORTEXM_DHCSR_C_DEBUGEN set before the next write can take away  C_DEBUGEN . Can you please try PR 969 (https://github.com/UweBonnes/blackmagic/tree/fixes) that beside other fixes really cleans up DHCSR on a normal exit and report back?

pavel-kirienko, 2022-01-06:
Thanks, Uwe. I flashed the firmware from the branch you suggested, removed my workaround, and ran a few tests, so far it seems to be holding up well. As you understand, I can't be certain whether it works yet because the problem cannot be reproduced consistently. I will continue using this new configuration and if I see the problem again, I will report it here.

I have two tangential questions:

1. The last release was over a year ago, and the codebase has changed considerably since then. Are there plans to publish a new release anytime soon?
2. The branch you pointed me to reports its version as `Black Magic Probe v1.6.1-917-g5c07d61, Hardware Version 3`, notice that the minor version is 1.6 whereas the latest release is 1.7. How come?

UweBonnes, 2022-01-06:
BMP needs a release asap, but  is also constantly evolving, so it is hard to decide when a release is to be done. Input welcome. Regarding the tagging, #966 discussed the issue and hopefully fixes it for new releases

pavel-kirienko, 2022-01-06:
I was unable to reproduce the DHCSR.C_MASKINTS problem so far so I would cautiously say that it is, indeed, fixed since v1.7-131. However, on your branch, I found a regression that may not be related to DHCSR and is certainly not related to semihosting. Steps to reproduce (no semihosting needed):

1. Define a catchpoint as follows:

```gdb
catch signal SIGTRAP
commands
    continue
end
```

2. Attach, load, run.

3. Make the firmware execute a soft reset at one point by calling `NVIC_SystemReset()`.

4. Observe the notification from GDB that the catchpoint was hit on reset. Execution will continue as expected.

5. Go to step 3. At approx. 3rd-6th iteration the target hangs and it is no longer possible to attach (see error messages below) until the target is power cycled.

```
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
```

This problem can be reproduced reliably with `v1.6.1-917-g5c07d61` and never with `v1.7.1-131-g560a046`.

UweBonnes, 2022-01-07:
Regarding the original problem, the master branch should work the same way too. I tried to reproduce the situation that made me write  107f65d02def555dc022f07c2b0ce2e48ded4f75, but could not do it. Probably I missread something, saw a phantom, had a MCU in a bad state or a MCU with bad debug implementation. 107f65d02 does no harm, but adds cruft and I will drop it.

Regarding your new problem: Best is to open a new issue. Did you keep power to the debugger or was it power cycled too. In the later case, I suspect a memory problem, as hints from another issue indicate too. I will try to reproduce the next days. When I can reproduce, I will try to reproduce with hosted. If it is seen on hosted, bisect is easier than a bisect in firmware, no need for constant reflash. If it is a memory problem, compiling hosted with libasan will help too.

pavel-kirienko, 2022-01-07:
>I tried to reproduce the situation that made me write 107f65d, but could not do it.

The tech reference says that incorrect usage of DHCSR leads to unpredictable behavior, so is it not possible that your later attempts to reproduce the problem merely did not reveal it while the problem might actually still be there, latent? I am getting a bit paranoid perhaps but I did spend way too much time hunting down this elusive issue.

>Did you keep power to the debugger or was it power cycled too.

The debugger was kept powered on and physically connected. I only power cycled the target.

I am going to file a new issue.

dragonmux, 2023-02-27:
We've done some analysis and concur that DHCSR is incorrectly handled both in cortexm_halt_resume() and originally in adiv5_initial_halt(). A further reading of the documentation on DHCSR says that it's not even valid for C_MASKINTS to be set for C_STEP unless C_HALT is also set either before hand or in the same write, which of course is extra problematic. Probably the right thing to do here is to ensure C_MASKINTS is definitely cleared during detach having ensured C_HALT is set

dragonmux, 2023-02-27:
This should now be properly addressed by the linked branch (fix/cortexm-halt-resume-correctness) - we will open a PR with these fixes once the ones the fix branch depends on are merged

closed 2023-04-04 by esden