blackmagic-debug/blackmagic #970: Target hangs after several software resets, not possible to attach afterwards

pavel-kirienko, 2022-01-07:
This ticket is split from https://github.com/blacksphere/blackmagic/issues/968#issuecomment-1006961836. The target is STM32F446, GDB version `GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git`.

1. Define a catchpoint as follows:

```gdb
catch signal SIGTRAP
commands
    continue
end
```

2. Attach, load, run.

3. Make the firmware execute a soft reset at one point by calling `NVIC_SystemReset()`.

4. Observe the notification from GDB that the catchpoint was hit on reset. Execution will continue as expected.

5. Go to step 3. At approx. 3rd-6th iteration the target hangs and it is no longer possible to attach (see error messages below) until the target is power cycled.

```
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
```

This problem can be reproduced reliably with https://github.com/blacksphere/blackmagic/pull/969 and never with `v1.7.1-131-g560a046` (i.e., the latest release).

UweBonnes, 2022-01-07:
I try to reproduce with a minimum program on a nucleo-f411 and BMP firmware from the fixes branch, no semihosting used. NVIC Reset triggered by the blue nucleo button first . Multiple button presses did not show the problem. Now the program reset itself after some 100 milliseconds from start. No problem so long:
```(gdb) info br
Num     Type           Disp Enb Address    What
1       catchpoint     keep y              signal "SIGTRAP" 
        catchpoint already hit 933 times
        continue
(gdb) mon
Black Magic Probe(STLINK-V2) v1.7.1-340-g17c5ae28, Hardware Version 2
```
Will keep it running for some time...

There must be more to reproduce  that problem.

pavel-kirienko, 2022-01-07:
Can you share your minimal program, please? I will see if I can make it work on my target.

UweBonnes, 2022-01-07:
The program uses the Ethernut Framework https://sourceforge.net/projects/ethernut/ so it uese a programm that is (mostly) target independent and linked against  a library that is target dependent. Do you have a Nucleo-F446? I can easy compiled for that target, as the library has the deinitions for that. But in the end ```
int main(void)
{
 volatile unsigned int i;
 for (i =0; i <1000*1000; i++);
NVIC_SystemReset();
}```
should work too.

pavel-kirienko, 2022-01-07:
Okay, I am going to look into that. Couldn't find any F4xx Nucleo nearby so far.

What if you add an interrupt and a WFI? Would that change things?

UweBonnes, 2022-01-07:
My Nut/Os test program uses interrupts, e.g. for serial output and timerticks. Idle loop uses WFI. So interrupts and WFI where active.

dragonmux, 2023-04-03:
👋🏼 sorry it's been so long since you were last contacted about this issue @pavel-kirienko - could you test against latest main and let us know if this is still an issue as we'd like to get this resolved for v1.10

dragonmux, 2023-12-31:
Just checking back in with this issue as now that our and ALTracer's fixes for semihosting have landed in main and we've been able to confirm semihosting is actually working now, we would appreciate this being re-tested and updated on if the issue is fixed or still need work.

pavel-kirienko, 2023-12-31:
I can tentatively confirm that the problem is addressed, but it's hard to be certain due to its stochastic nature. I will conduct further testing later on; should the issue re-appear, I will re-open this ticket. Thank you!

closed 2023-12-31 by pavel-kirienko