blackmagic-debug/blackmagic #971: Use malloc or alloca for memory only used inside  function?

UweBonnes, 2022-01-07:
Some recent issues makes me think that BMP runs out of memory to allocate  and than hangs. Semihosting code uses malloc inside its call and frees the memory  immediate.  Semihosting calls may occur at high rate. Isn' t there a chance that mempool fragmentation happens. Isn't alloca() better suited for embedded system when memory is only used inside a function.

DX-MON, 2022-01-07:
Are the structures semihosting is allocating for only used within the one function? If so, the ideal would be no allocation and let them be stack structures by writing them as so. This would eliminate the fragmentation worry while simplifying the code

UweBonnes, 2022-01-07:
Looking closer, the malloc with regard to semihosting only happens with hosted. So no chance that his is the cause of a possible memory allocation failure as a cause of the reported  problems.

closed 2022-01-07 by UweBonnes

DX-MON, 2022-01-08:
It would still be interesting to us if you could link which files are doing these allocations so we can review if they're necessary or could be converted to stack usage please?

UweBonnes, 2022-01-08:
CortexM semihosting is handled in target/cortexm.c. As said, this is with hosted, running on a big machine with much more RAM than the 20 kByte our little STM32F103 has.

DX-MON, 2022-01-08:
The issue isn't the RAM usage - the issue we're looking to ensure is the proper paring of malloc with free and the performance penalty it causes and if paying that price is even necessary

DX-MON, 2022-01-08:
As it happens, we're happy now as all the allocations there are done to avoid VLAs (good) and aren't really avoidable as they are all truly dynamic.