blackmagic-debug/blackmagic #979: support for Sparkfun Artemis line of product

jerabaul29, 2022-01-25:
Being curious: is the Sparkfun Artemis line of product supported? I do not see the underlying MCU listed in the supported list (it is an Ambiq Apollo 3 BLU, which is an implementation of ARM Cortex M4F). See: https://www.sparkfun.com/artemis .

A bit naive, but I would have guessed that processor of a given family (like cortex M4 F) may share some "high level" way to debug, but maybe I am wrong? :) .

jerabaul29, 2022-01-25:
Ok, thanks for the indication :) . I wanted to have a bit of an idea if I could expect it to work or not at all on a "non officially supported chip that implements a CortexM" before buying a probe. Will order, try, and let you know then.

UweBonnes, 2022-01-25:
Have a look at https://github.com/blackmagic-debug/blackmagic/wiki. Hosted can talk to some other debugger hardware for a first try.

esden, 2022-08-02:
As sparkfun has the module on backorder and does not seem to know when they will be available again, I have the feeling that this module might be obsolete due to chip shortage. I am closing this issue for now, if someone decides to add support for it please feel free to reopen.

closed 2022-08-02 by esden

alistair23, 2023-06-22:
It would be great to see support for the Apollo3. Currently Blackmagic will connect to the device and single step through existing code, but breakpoints and other useful features don't work

dragonmux, 2023-06-23:
Please open issue tickets with specific information on the issues you're seeing, with BMDA logs (using verbosity flag `-v 29`) if possible, and part numbers.

dragonmux, 2024-09-02:
Happy news, basic support is now merged in #1587 - while this doesn't implement support for the Flash, everything else should work now.