blackmagic-debug/blackmagic #981: LPC and MCULINK broken since b343ebe06f303ebe

UweBonnes, 2022-01-25:
Following hotfixes were done to get bmp/hosted working with Dragonprobe https://git.lain.faith/sys64738/DragonProbe.git

commit 166eb3ee3eb736ece3e874738e8eb564e3f08cbb
Author: Uwe Bonnes <bon@elektron.ikp.physik.tu-darmstadt.de>
Date:   Mon Oct 11 20:57:49 2021 +0200

    cmsis_dap: Always transfer 65 bytes with hid_read|write()
    
    CMSIS-DAP Dragonprobe works on some devices. E.g. it does not work
    on NUCLEO-F103

commit b343ebe06f303ebe5d46e4a05aaceafcf0a04b3d
Author: Uwe Bonnes <bon@elektron.ikp.physik.tu-darmstadt.de>
Date:   Thu Oct 14 11:45:36 2021 +0200

    cmsis-dap/linux: Link against libhidapi-hidraw
    
    libhidapi-libusb detaches a kernel module but does not reattach again.
    This may wrack interaction with other programs using hidapi.

LPC-Link and MCU Link transfer hangs. Other CMSIS probes, e.g. GD32VF103C-Start seem not affected. Reverting both patches gets things going again for the LPC- and MCU-Link

UweBonnes, 2022-05-22:
I can not reproduce.

closed 2022-05-22 by UweBonnes

dragonmux, 2022-05-22:
Does that mean the associated PR can be closed too, or?