blackmagic-debug/blackmagic #988: Using an alternate toolchain and I need to specify --specs=nosys.specs in the makefile somewhere

embeddedpenguin, 2022-02-26:
Hi, I am using a toolchain produced by crossdev on gentoo, and there is currently an issue with the toolchain where nano_libc stuff isn't being found. This is a toolchain problem, not a blackmagic debugger problem. Bug report [here](https://bugs.gentoo.org/834052) I'm trying to test a fix, but I the makefile structure in the project is a little confusing for me. Where can I safely change the --specs flag so that when blackmagic.elf is being created, it uses --specs=nosys.specs?

UweBonnes, 2022-02-27:
What about in the  makefiles
SPECS =? nano.specs
LDFLAGS_BOOT := ... -specs=$(SPECS) ...

and you call "make SPECS=nosys.specs .."
?

embeddedpenguin, 2022-02-28:
> What about in the makefiles SPECS =? nano.specs LDFLAGS_BOOT := ... -specs=$(SPECS) ...
> 
> and you call "make SPECS=nosys.specs .." ?

Hmm. Turns out marking nosys.specs does not resolve the issue. Thank you for the help, though.

closed 2022-02-28 by embeddedpenguin