blackmagic-debug/blackmagic #996: Cannot connect to EFM32ZG target after writing to SWD GPIO pin registers

rjauc, 2022-03-17:
I am unsure whether this is a missing feature of the probe or if it's a shortcoming of the EFM32, but I wrote some erroneous code for an EFM32ZG108F32 target and accidentally changed the GPIO modes of the SWD interface GPIO pins: I set SWCLK to "Open-drain output with filter and pullup" and SWDIO to "Open-drain output with pullup."

I believe the mistake was writing this code right at the top of the program - without any delays. The only code that gets executed before it is a simple startup assembly file that prepares the C environment, and a single write to a GPIO clock control register. My assumption is that the GPIO modes are changed before the debugger can connect - tried connecting under srst with no success.

The EFM32 also has a UART bootloader on the same pins as the SWD pins, but I can't seem to get into it either, even though it worked before program upload by tying the SWCLK (UART TX) pin high during boot. The [bootloader documentation](https://www.silabs.com/documents/public/application-notes/an0003-efm32-uart-bootloader.pdf) mentions I have to use a 4kiloohm pullup on SWCLK for SWD use - that does nothing and was not required before.

I'm unsure if/how to get access to the chip. The most surprising thing to me is why I can't get into its bootloader, since I feel like that should be checked by the MCU before it starts executing program code, like mentioned in the link above.

rjauc, 2022-03-17:
I... just realised all I needed were pull-up resistors on BOTH of the SWD lines, since they're set to open-drain, I managed to connect to the EFM32 and erase the erroneous program. I need to get myself a rubber ducky!

closed 2022-03-17 by rjauc

esden, 2022-03-17:
I am glad you found the issue. What might have also worked is using the `mon connect_srst enable`. Or manually forcing the mcu to be under reset when you are scanning and connecting to it. This would prevent the mcu from misconfiguring the pins. :)

Good luck with the project. :)

rjauc, 2022-03-18:
I've already tried both of those things - even randomly activating reset while spamming `mon s`, but this MCU disables SWD under reset if I remember correctly. The thing that confuses me most is why it didn't boot into bootloader when I reset/powered it up with the SWD_CLK pin tied to VCC, since, to my understanding at least, it should've worked.

esden, 2022-03-22:
Ok that is annoying. In any case I am glad you found some solution that worked.