blackmagic-debug/blackmagic #1008: Firmware for stlink-v3

escherstair, 2022-04-05:
[Here](https://github.com/blackmagic-debug/blackmagic/tree/master/src/platforms) there are the implementation for some platforms: I see listed STLINK-V1, V2 and V21 but I can't find V3.
Now some new Nucleo board (as an example Nucleo-H743ZI2) has a STLINK-V3 onboard.
In the [stlink subpage](https://github.com/blackmagic-debug/blackmagic/tree/master/src/platforms/stlink) I see some reference to V3, but the overall is not very clear (maybe the README can be improved a little bit, but the situation is not clear to me).
If someone can give details, I can send s PR with some improvements to the README.

UweBonnes, 2022-04-05:
STLINK-V3 can be used with blackmagic hosted. For recent STLINK-V3 there is no known way to flash a custom firmware, as needed for blackmagic firmware. ST encryption has not been reverse engineered yet : https://github.com/RadioOperator/CMSIS-DAP_for_STLINK-V3MINI/issues/1 .Not so recent V3 versions only inhibit Debug access, but a custom version on stlink-tools https://github.com/UweBonnes/stlink-tool/tree/stlinkv21 can flash a custom firmware. Beware for the ST updater with recent firmware, it will lock non-locked devices! Older v3 are fully debug accessible and can be reflashed with a blackmagic firmware. For a blackmagic firmware for V3, see #818.

dragonmux, 2023-01-05:
Hi escherstair! With James Turton's help, the existing ST-Link v3 PR (linked above) has been overhauled and with his help should make it into v1.10. Please keep track of the linked open PR to track this progress

PR #1354 (2023-02-02): Support for ST-Link v3

## Detailed description

Rebased and tested version of #818.
I have verified that all basic functionally works. However I am yet to test SLCAN.
I know the newer stock firmware for the STLink disables the SWD interface on the probe, but for those who haven't updated the stock firmware it does make for a nice probe - especially the [STLINK-V3MINIE](https://www.st.com/en/development-tools/stlink-v3minie.html) which I have been testing with! The major advantage I see with this probe is that the STM32F723 supports USB 2.0/Hi-Speed (480 Mbps) which is magnitudes quicker than  [... cut]
closed 2023-02-02 by dragonmux