blackmagic-debug/blackmagic #1009: mentions of Black Magic Sphere

perigoso, 2022-04-12:
The project distanced itself from "Black Magic Sphere", but the windows .inf file still has mentions

[driver/blackmagic.inf](https://github.com/blackmagic-debug/blackmagic/blob/master/driver/blackmagic.inf)

perigoso, 2022-04-12:
the readme hosted example also shows but thats minor

esden, 2022-06-03:
@perigoso thank you for reporting this. I hope I addressed all the remaining spots. Let me know if I missed anything. :)

perigoso, 2022-06-03:
I just spotted a couple, apparently there were more laying around (:

The only thing that i know of, although it's not part of the actual firmware so it's really just symbolic, is the one I mentioned, in the readme

[image]

which is an example run, but the probe is using an old version firmware that still has the black sphere string

esden, 2022-06-07:
Good point. I ignored it "on purpose" because I was too lazy to regenerate the output. But I probably should deal with it. I am reopening this issue so that I don't forget to make that fix.

commit e4d692bf (2022-05-30): driver: Updated vendor name to Black Magic Debug. Closes #1009
PR #1065 (2022-06-25): readme: sample usage update
closes #1009
closed 2022-06-25 by dragonmux