blackmagic-debug/blackmagic #1013: Flashing RP2040 fails if NRST pin is connected

PetteriAimonen, 2022-05-05:
When I connect NRST pin to RP2040, flashing fails with error `Error erasing flash with vFlashErase packet`:

```
(gdb) mon s
Target voltage: 2.93V
Available Targets:
No. Att Driver
 1      Raspberry RP2040 M0+
 2      Raspberry RP2040 M0+
 3      Raspberry RP2040 Rescue(Attach to reset!) 
(gdb) att 1
...
(gdb) load
Sending packet: $vFlashErase:10000000,0001d000#f0...Ack
Packet received: EFF
Error erasing flash with vFlashErase packet
```

But without NRST connected, everything works fine.

The problem seems to happen at [line 553](https://github.com/blackmagic-debug/blackmagic/blob/master/src/gdb_main.c#L550-L555) in `gdb_main.c`:

```
		if (!flash_mode) {
			/* Reset target if first flash command! */
			/* This saves us if we're interrupted in IRQ context */
			target_reset(cur_target);
			flash_mode = 1;
		}
```

If I comment out the `target_reset()` call, flashing works ok. But even then `run` seems flaky.
It seems the reset by NRST pin causes black magic probe to lose connection to the RP2040, requiring new scan and attach.

I'm not sure what is the correct way to fix this, apart from just disabling NRST pin usage when connected to RP2040.

PetteriAimonen, 2022-05-05:
Newest git (c4869a54733ae920) on ST-LINK/V2 hardware (installed natively to the STM32F103 flash, with BMP bootloader).

PetteriAimonen, 2022-05-05:
@jamesturton @UweBonnes Have either of you tried NRST with RP2040 and BMP by any chance? It seems that it is quite common to only connect SWDIO/SWCLK in RP2040 tutorials.

dragonmux, 2022-05-05:
Thank you for the extra information - as for the issue you're hitting.. we'll be able to do some diagnosis work the start of next week when our RP2040 boards arrive specifically to check for this kind of problem.

PetteriAimonen, 2022-05-06:
Great :)

It's not a big issue and BMP seems to be working well for RP2040 development - it was surprisingly difficult to find a debug probe that supports it.

UweBonnes, 2022-05-21:
Can you please add  "t->target_options |= CORTEXM_TOPT_INHIBIT_SRST;" at the end of  rp_probe(), compile,  test and report back?

dragonmux, 2022-05-21:
Uwe, we recreated this issue, it's not quite that simple and this interacts with some other things. Please let us handle it.

UweBonnes, 2022-05-21:
Can you please explain whaa the issue(s) is/are?

PR #1045 (2022-06-11): RP2040 updates
Fixes #1013. I have done some testing and the suggestion @UweBonnes made to reset using `SYSRESETREQ` seems to be the correct fix here.

I have also fixed up some other oddities with the rp2040's flash handling and added the `erase_sector` command to help when debugging the rp2040.
closed 2022-06-11 by dragonmux