blackmagic-debug/blackmagic #1020: Platforms: any plan for EtherNet or Wi-Fi?

joelguittet, 2022-05-23:
Hello,
Are there any plan for EtherNet or Wi-Fi based platforms to run the BMP firmware?
I now the ctxLink project but it s not updated since a while now. Having the possibility to build and use remote BMP would be great for several use cases.
What about using an esp32 for example for this purpose?
Joel

esden, 2022-05-23:
Hi Joel,

It is currently not core functionality of the system as it is now. Besides @sidprice ctxLink, there is also @xobs who is working on a hardware and firmware fork that, as opposed to ctxLink, is fully hosted on an esp32. I am not sure what the current status of that project is.

Before we can consider adding more host platforms to the project, (especially any that come with a big amount of platform specific uniqueness) we have to do a lot of architectural changes. We do have plans to address those architectural shortcomings, and I am working with @dragonmux towards that goal. We will document our plans when the time is right.

To summarize: In medium term we might consider adding support for remote/wireless probes as long as it will not result in maintenance nightmare. Until then, there are forks of the project that currently or in near future will support remote/wireless use.

I will leave this issue open as a feature request. Let me know if you have more questions. We could eventually rework your question into a proper RFC.

Cheers,
Piotr

xobs, 2022-05-23:
I have my port up at https://github.com/xobs/blackmagic-espidf/ and it mostly works. I have custom hardware that does level translation and voltage conversion from (1.8V-5V) to 3.3V. These boards just got back from the fab, and I'll be testing them next week.

esden, 2022-08-02:
As there are already existing BMP based devices that provide WiFi and we have currently no plans for adding Ethernet I am closing this issue.

closed 2022-08-02 by esden