blackmagic-debug/blackmagic #1022: Windows 11 - Drivers Missing ?

nurquhar, 2022-05-26:
OK I have a Black Magic Probe for a couple of years now, it works great on my Windows 10 laptop. Plug it in and away you go, no usb driver install was required.

I have now upgraded my old Windows 10 machine to a new Dell Laptop which has Windows 11. I connect my usb Black Magic Probe and in Device Manager I see the devices under 'Other Devices' and they both have exclamation marks against the icons. I guess this means the drivers are missing. 

So where to get the drivers or is there another fix ???

I tried Dell Support, they did what I did and checked for any driver updates and then they just gave up as being out of their depth.

Here is what I see in Device Manager.

[image]

dragonmux, 2022-05-26:
Your BMP is already working (the two serial ports show up, the lower numbered is the GDB endpoint, and the higher is the USB UART connected to the UART connections on the back of the board) - the trace capture and firmware upgrade endpoints only need configuring if you need to do tracing or upgrade the firmware.

Our recommended way for now on Windows for setting those up is to use [Zadig](https://zadig.akeo.ie/) and bind both to the WinUSB driver.

nurquhar, 2022-05-28:
So is Zadig now required just to use the program function of BMP ?  As it does not seem to work out the box on Win 11, I get this warning and it hangs. The same happens if I use COM3 or COM4 in my script.
[image]

Removing the other clutter from the USB bus I have this in Device Manger so COM3 & 4 are associated with BMP.
[image]

On the windows 10 machine it programs fine using COM10 in my script.
[image]

However the device manager looks a little different
[image]

dragonmux, 2022-05-28:
Zadig has always been required for Windows to bind drivers to the DFU and Trace/Capture interfaces - this is not a new requirement. Most likely you (or if it's a shared machine, someone else) did this a long time ago on the Windows 10 machine.

However, that's separate from being able to use BMP to program a target which only uses the "COM ports" you see (COM3 for the GDB server which is what you show your script using in the second set of screenshots). You can safely completely ignore the DFU and Trace/Capture interfaces, they're not the cause of your failures. We'd suggest that, as COM10 works with your script already, use the Windows properties dialog for "USB Serial Device (COM3)" to switch it to binding as COM10 and try again.

nurquhar, 2022-05-28:
Ok I used Zadig and now the missing drivers are fixed. I also change COM3 to be COM10 on my Win 11 m/c, not sure why this is necessary as I can edit the port in my script. 
[image]

However even with using COM10 the same thing happens 
[image]

Here is the script also
[image]

dragonmux, 2022-05-28:
Ah, we assumed that changing the port in your script was more complicated (you can put it back if you like) - but fair enough. As for your GDB problems.. without knowing what exact version of GDB you're trying to use and the providence of the toolchain we unfortunately can't be of much further help.

The only extra things we can offer are to check that what is now COM10 is definitely for interface 0 (using the Windows driver properties dialog) to gain some confidence that it's definitely the GDB server interface, and run GDB manually with each of the commands you're passing it with -ex to see what, where, why, and how it's failing with more detail.

Of important note, GDB 11 and 12 will **not** work because of some known bugs in GDB itself, so you must use GDB 10.

nurquhar, 2022-05-28:
I changed back to COM3. Tried manual steps, fails on telling it about COM3 and hangs (same happens for COM4)
[image]

Run same on Win 10 m/c : 
[image]
 To me it looks like I am running the same command on Win 11 as Win 10.
 So how to find the GDB version ?

dragonmux, 2022-05-28:
it's in the output you provided - first line it outputs - GDB 8.0.50 from 2017. Also, note you have an extra backslash on the end of COM3 in your Windows 11 invocation - that won't work.

esden, 2022-05-28:
Hi @nurquhar,

You should try using the other com port. Sometimes windows is being annoying and likes to shuffle the com ports for no reason. I have just installed win11 to test things out. It is a clean install without Zadig drivers installed. (They are not necessary unless you have to update the firmware on the BMP itself as @dragonmux described.)

Here is how that looks like in my Win11 device manager:
[image]

I used the official [arm embedded binary release](https://developer.arm.com/downloads/-/gnu-rm) for testing.

As mentioned above, I had to use the second com port instead of the first one. But it works fine without locking up. When remote connection hangs it is usually because the USBUSART port is selected instead of the GDB serial port.

Here is the output from the Win11 machine.

[image]

I hope you can figure out your setup.

nurquhar, 2022-05-28:
Thanks for all the help guys, I have if figured out and working with COM4 as suggested 👍 The extra slash did not help either, thanks for pointing this out.

[image]

It seems testing with COM3 would fail because it was the wrong COM port. However simply changing the script to COM4 after a failure with COM3 and re-running was insufficient. Re-running the script with COM4 set would also then fail. However COM4 will work after a failure with COM3 if I cycle the USB connection (Unplug and plugin again) !!!

dragonmux, 2022-05-28:
Glad you got it working at last :) going to close the issue for now, but if you encounter any more problems with BMP itself, you'd be welcome to open a new issue.

closed 2022-05-28 by dragonmux

esden, 2022-05-28:
I am also glad to hear that you managed to get things to work. Additionally to what @dragonmux said you can also [join our discord server](https://1bitsquared.com/pages/chat) and ask your questions in the #blackmagic channel where we can give you some real time support. :)