blackmagic-debug/blackmagic #1029: Blackmagic will write 0s on S32K devices, bricking them

xobs, 2022-06-03:
BMP has broken something between 8def28d and f6edb54395d3d4cb2bb186f079cb3e3181eff518 that has caused it to brick S32K devices. This is because the second word written is now `0x00000000` regardless of the actual value. Crucially, address `0x400` is the Flash Configuration Field that locks flash. If this field is anything other than `0xffffffff`, the chip will become locked and cannot be unlocked.

```
(gdb) load
Loading section .interrupts, size 0x400 lma 0x0
Loading section .flash_config, size 0x10 lma 0x400
Loading section .text, size 0x1cb90 lma 0x410
Loading section .ARM, size 0x8 lma 0x1cfa0
Loading section .data, size 0x4b8 lma 0x1cfa8
Start address 0x00000410, load size 119904
Transfer rate: 9 KB/sec, 944 bytes/write.
(gdb) x 0
0x0 <__isr_vector>:     0x2001f000
(gdb) x 4
0x4 <__isr_vector+4>:   0x00000000
(gdb) x 0x400
0x400:  0xffffffff
(gdb) x 0x404
0x404:  0x00000000
(gdb) x 0x408
0x408:  0xffffffff
(gdb)
```

Going back to an older version of BMP, notably 8def28d, resolves the issue for me, meaning something happened between these two versions that caused this failure.

## Recovering a device

If you discover that `0x404` has been overwritten, **do not power off the target**. Instead, issue a `erase_mass`:

```
(gdb) att 2
A program is being debugged already.  Kill it? (y or n) y
Attaching to program: /opt/Ion/gridania-bringup-dvt1/148/_build/gridania-dvt1-148.elf, Remote target
warning: while parsing target description: no element found
warning: Could not load XML target description; ignoring

Program stopped.
warning: while parsing target memory map (at line 1): Required element <memory> is missing
0xc22ea4c2 in ?? ()
(gdb) mon erase_mass
Requesting mass erase (status = 0x1003e)
Mass erase acknowledged
Mass erase complete
(gdb) att 1
A program is being debugged already.  Kill it? (y or n) y
Attaching to program: /opt/Ion/gridania-bringup-dvt1/148/_build/gridania-dvt1-148.elf, Remote target

Program stopped.
0x00000000 in __isr_vector ()
(gdb) x 0
0x0 <__isr_vector>:     0xffffffff
(gdb) x 4
0x4 <__isr_vector+4>:   0xffffffff
(gdb)
```

Because this field is latched on power up, the board will continue to operate with debugging enabled until you power cycle the board.

UweBonnes, 2022-06-03:
If you work with hosted, git bisect will show the offending patch in a resonable time. The device is needed, so please, if you can, give it a try.

xobs, 2022-06-03:
I started a bisect, and ended up bricking a board with this change. I don't know that I can continue to debug this, due to the fact that this bug is really good at destroying Kinetis parts.

As a related note, openocd treats this FCF region specially. There is a command `fcf_source` that must explicitly be called in order to allow writing this region. It may be worth considering doing something similar in Blackmagic: https://github.com/openocd-org/openocd/blob/master/src/flash/nor/kinetis.c#L3018-L3042

dragonmux, 2022-06-03:
@UweBonnes we assigned this to ourself because of the gravity of what's going wrong with the intention to sit down and figure it out today. This isn't a bug for a user to try and figure out.

@xobs don't mind Uwe in this case - we know he's trying to be helpful but we recognise that this is a parts killer bug. While it was marked for v1.9, it's actually getting re-marked for v1.8.1 bugfix release. We will be diving into what's going wrong and dry-running or otherwise investigating why on earth flashing would wipe this memory address, or rather completely set it as 0 is set here.

Thank you for the pointer re special region protection as we can look into implementing that so this can't happen again even with another regression.

xobs, 2022-06-08:
As discussed on Discord, reverting c1a12edb seems to fix this.

xobs, 2022-06-14:
What is the status of this? Will c1a12edbe9a801051a6bfdfde7264be623bd389f be reverted on the master branch?

closed 2022-07-12 by esden

xobs, 2022-07-13:
I've verified that I'm able to program an S32K118 now without having it lock itself.

PR #1098 (2022-07-12): Fix: kinetis Flash control correctness
This PR fixes a part-killing regression found by xobs. It does this by reworking the Kinetis Flash command generation and issuing routines so more closely match the behaviour of the v1.7 ones while retaining the benefits of the revamped approach.

This closes #1029 and should be merged both to main and to v1.8.