blackmagic-debug/blackmagic #1036: Building with `ST_BOOTLOADER=1` fails with "memcpy is out of the bounds"

jounathaen, 2022-06-05:
When I try to build the firmware with `make PROBE_HOST=stlink ST_BOOTLOADER=1` as described in https://github.com/blackmagic-debug/blackmagic/blob/master/src/platforms/stlink/README.md, the build fails with `error: 'memcpy' offset [0, 11] is out of the bounds [0, 0] [-Werror=array-bounds]`.  A build with only `make PROBE_HOST=stlink` succeeds.

Full compilation log:
```
> make PROBE_HOST=stlink ST_BOOTLOADER=1
  GENHDR  stm32/f0
  GENHDR  stm32/f1
  GENHDR  stm32/f2
  GENHDR  stm32/f3
  GENHDR  stm32/f4
  GENHDR  stm32/f7
  GENHDR  stm32/l0
  GENHDR  stm32/l1
  GENHDR  stm32/l4
  GENHDR  stm32/g0
  GENHDR  stm32/g4
  GENHDR  stm32/h7
  GENHDR  gd32/f1x0
  GENHDR  lpc13xx
  GENHDR  lpc17xx
  GENHDR  lpc43xx/m4
  GENHDR  lpc43xx/m0
  GENHDR  lm3s
  GENHDR  msp432/e4
  GENHDR  efm32/tg
  GENHDR  efm32/g
  GENHDR  efm32/lg
  GENHDR  efm32/gg
  GENHDR  efm32/hg
  GENHDR  efm32/wg
  GENHDR  efm32/ezr32wg
  GENHDR  sam/3a
  GENHDR  sam/3n
  GENHDR  sam/3s
  GENHDR  sam/3u
  GENHDR  sam/3x
  GENHDR  sam/4l
  GENHDR  sam/d
  GENHDR  vf6xx
  GENHDR  swm050
  GENHDR  pac55xx
  BUILD   lib/stm32/f1
  CC      adc.c
  CC      adc_common_v1.c
  CC      can.c
  CC      crc_common_all.c
  CC      dac_common_all.c
  CC      desig_common_all.c
  CC      desig_common_v1.c
  CC      dma_common_l1f013.c
  CC      exti_common_all.c
  CC      flash.c
  CC      flash_common_all.c
  CC      flash_common_f.c
  CC      flash_common_f01.c
  CC      gpio.c
  CC      gpio_common_all.c
  CC      i2c_common_v1.c
  CC      iwdg_common_all.c
  CC      pwr_common_v1.c
  CC      rcc.c
  CC      rcc_common_all.c
  CC      rtc.c
  CC      spi_common_all.c
  CC      spi_common_v1.c
  CC      timer.c
  CC      timer_common_all.c
  CC      usart_common_all.c
  CC      usart_common_f124.c
  CC      mac.c
  CC      mac_stm32fxx7.c
  CC      phy.c
  CC      phy_ksz80x1.c
  CC      usb.c
  CC      usb_control.c
  CC      usb_standard.c
  CC      usb_msc.c
  CC      usb_hid.c
  CC      usb_audio.c
  CC      usb_cdc.c
  CC      usb_midi.c
  CC      usb_dwc_common.c
  CC      usb_f107.c
  CC      st_usbfs_core.c
  CC      st_usbfs_v1.c
  CC      vector.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_stm32f1.a
  BUILD   lib/stm32/f4
  CC      adc_common_v1.c
  CC      adc_common_v1_multi.c
  CC      adc_common_f47.c
  CC      can.c
  CC      crc_common_all.c
  CC      crypto_common_f24.c
  CC      crypto.c
  CC      dac_common_all.c
  CC      dcmi_common_f47.c
  CC      desig_common_all.c
  CC      desig_common_v1.c
  CC      dma_common_f24.c
  CC      dma2d_common_f47.c
  CC      dsi_common_f47.c
  CC      exti_common_all.c
  CC      flash.c
  CC      flash_common_all.c
  CC      flash_common_f.c
  CC      flash_common_f24.c
  CC      flash_common_idcache.c
  CC      fmc_common_f47.c
  CC      gpio_common_all.c
  CC      gpio_common_f0234.c
  CC      hash_common_f24.c
  CC      i2c_common_v1.c
  CC      iwdg_common_all.c
  CC      lptimer_common_all.c
  CC      ltdc_common_f47.c
  CC      pwr_common_v1.c
  CC      pwr.c
  CC      rcc_common_all.c
  CC      rcc.c
  CC      rng_common_v1.c
  CC      rtc_common_l1f024.c
  CC      rtc.c
  CC      spi_common_all.c
  CC      spi_common_v1.c
  CC      spi_common_v1_frf.c
  CC      timer_common_all.c
  CC      timer_common_f0234.c
  CC      timer_common_f24.c
  CC      usart_common_all.c
  CC      usart_common_f124.c
  CC      usb.c
  CC      usb_standard.c
  CC      usb_control.c
  CC      usb_msc.c
  CC      usb_hid.c
  CC      usb_audio.c
  CC      usb_cdc.c
  CC      usb_midi.c
  CC      usb_dwc_common.c
  CC      usb_f107.c
  CC      usb_f207.c
  CC      mac.c
  CC      phy.c
  CC      mac_stm32fxx7.c
  CC      phy_ksz80x1.c
  CC      vector.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_stm32f4.a
  BUILD   lib/lm4f
  CC      assert.c
  CC      gpio.c
  CC      rcc.c
  CC      systemcontrol.c
  CC      uart.c
  CC      vector.c
  CC      usb.c
  CC      usb_control.c
  CC      usb_standard.c
  CC      usb_msc.c
  CC      usb_hid.c
  CC      usb_audio.c
  CC      usb_cdc.c
  CC      usb_midi.c
  CC      usb_lm4f.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_lm4f.a
Load address 0x08004000 for original ST-LinkV2 Bootloader
 GIT include/version.h
  CC      target/adiv5.c
  CC      target/adiv5_jtagdp.c
  CC      target/adiv5_swdp.c
  CC      command.c
  CC      target/cortexa.c
  CC      target/cortexm.c
  CC      crc32.c
  CC      target/efm32.c
  CC      exception.c
  CC      platforms/stm32/gdb_if.c
  CC      gdb_main.c
  CC      gdb_hostio.c
  CC      gdb_packet.c
  CC      hex_utils.c
  CC      target/jtag_devs.c
  CC      target/jtag_scan.c
  CC      target/lmi.c
  CC      target/lpc_common.c
  CC      target/lpc11xx.c
  CC      target/lpc17xx.c
  CC      target/lpc15xx.c
  CC      target/lpc43xx.c
  CC      target/lpc546xx.c
  CC      target/kinetis.c
  CC      main.c
  CC      morse.c
  CC      target/msp432.c
  CC      target/nrf51.c
  CC      target/nxpke04.c
  CC      platforms/stlink/platform.c
  CC      remote.c
  CC      target/rp.c
  CC      target/sam3x.c
  CC      target/sam4l.c
  CC      target/samd.c
  CC      target/samx5x.c
  CC      target/stm32f1.c
  CC      target/ch32f1.c
  CC      target/stm32f4.c
  CC      target/stm32h7.c
  CC      target/stm32l0.c
  CC      target/stm32l4.c
  CC      target/stm32g0.c
  CC      target/target.c
  CC      platforms/common/cdcacm.c
  CC      platforms/stm32/usbuart.c
  CC      platforms/stm32/serialno.c
platforms/stm32/serialno.c: In function 'serial_no_read':
platforms/stm32/serialno.c:51:9: error: 'memcpy' offset [0, 11] is out of the bounds [0, 0] [-Werror=array-bounds]
   51 |         memcpy(unique_id, (uint32_t *)DESIG_UNIQUE_ID_BASE, 12);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:106: serialno.o] Error 1
make: *** [Makefile:17: all] Error 2
```

OS: Fedora 36
GCC: `arm-none-eabi-gcc (Fedora 11.1.0-2.fc36) 11.1.0`

Tested on current master and `v1.8.0` (5e3cade)

jounathaen, 2022-06-05:
Might be related: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578

dragonmux, 2022-06-05:
Thank you for the excellent and rather complete issue report! We will look into this in the next couple of days and get back to you once we've had a chance to determine what we can do to fix this

dragonmux, 2022-06-06:
So far we have been able to confirm that this works on GCC 12, lending to it probably being the GCC bug linked.

However, as it looks like this code needs some love anyway and might be expressible in a way that doesn't trigger that bug, we have gone and created [feature/stm32-bootloader-serial-cleanup](https://github.com/blackmagic-debug/blackmagic/tree/feature/stm32-bootloader-serial-cleanup) with the intent that this might well fix the issue you're seeing. Please let us know if it does, and if it does we'll open a PR with these changes.

perigoso, 2022-06-06:
I actually ran into this too on archlinux (was running on non main package gcc)

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 11.2.1 20220111
```

seems to add up

perigoso, 2022-06-06:
this compiled for me, feels a bit silly to swap the memcpy for this, but it should work
```
volatile uint8_t * dst = (void *)unique_id, src = (void *)DESIG_UNIQUE_ID_BASE;
for (size_t i = 0; i < 12; i++)
{
    *dst++ = *src++;
}
```

dragonmux, 2022-06-06:
We're perfectly fine using the swapped version as we can crunch that into the other loops and make the code simpler.. we're unlikely to suffer a perf penalty for that because the compiler should be smart enough to vectorise (and probably more efficiently too!)

closed 2022-06-12 by esden