blackmagic-debug/blackmagic #1042: Can we always prefix the COM port on Windows?

reppad, 2022-06-07:
Do you think it is safe to always use the `\\.\` prefix on Windows, whether the port number is 1 or 2 digits.
The idea is to propose an improvement of the integration into STM32duino on Windows because currently the BMP option only works if the host platform has a port number <= 10.

My personal experience shows that it works but I wonder if you know of any issues it could cause?

esden, 2022-06-07:
As far as I know there is no issue in assuming that all Windows COM ports have to be prefixed with `\\.\`. This is the way you address them.

The fact that stm32duino has an issue with com ports that use numbers that have two digit or more sounds like a bug somewhere in there. You should file a bug report for it. There is no known issue with GDB or Black Magic Debug App in regards to windows com ports and their numbering as far as I know. I hope this helps.

Note: As this is not really related to Black Magic but rather 3rd party tools on Windows I will close this issue...

closed 2022-06-07 by esden

reppad, 2022-06-07:
Indeed, today STM32duino never uses the prefix.
Before submitting the change to add it, I wanted to make sure that it wouldn't cause any known issue to add it all the time on Windows.
Thank you for your reply !

esden, 2022-06-07:
I think I see the context better after reading the Issue on stm32duino project page. The observation in that issue was that one could use just naked `COMx` from 0-9, as soon as a two digit number had to be used a `\\.\` prefix was needed. Here is my answer to this:
> Just to clarify, this is not a limitation of BMP in any way. BMP has nothing to do with any of this. You are using GDB to connect to a remote target com port that is provided by BMP. The correct way to provide remote com port targets to GDB on Windows was pretty much always with the `\\.\` prefix. (At least since Windows NT kernel. The prefix is essentially windows kernel internals leaking into the userspace.)

And for additional context, all single digit COM port names are legacy DOS names. This is why that even works at all. One should always use `\\.\` prefix as this is more reliable and does not have the digit limitation. So not using the prefix can be considered wrong and a bug.

reppad, 2022-06-08:
Many thanks for those clarifications, it's clear now that the prefix should be added all the time.
The fix is ready, it will be solved soon !