blackmagic-debug/blackmagic #1043: macOS: version.h not found

chriskuku, 2022-06-08:
I'm turning back to blackmagic again cause I'm in need to do some programming .
Don't know if I made a mistake, but I did a 

` git clone  https://github.com/blackmagic-debug/blackmagic.git`

and then cd blackmagic followed by this and got this:
`$ pwd
/Users/kuku/Downloads/blackmagic
$ make PROBE_HOST=hosted
 GEN GIT  include/version.h
  CC      target/adiv5.c
  CC      target/adiv5_jtagdp.c
  CC      target/adiv5_swdp.c
  CC      command.c
command.c:34:10: fatal error: 'version.h' file not found
#include "version.h"
         ^~~~~~~~~~~
1 error generated.
make[1]: *** [command.o] Error 1
make: *** [all] Error 2
`

dragonmux, 2022-06-08:
Thank you for the report! We can confirm that is a bug. For the time being, could you try `git revert 8afaedd` then try building again? That should be the commit that broke things.

We will look into how to fix this more permanently.

chriskuku, 2022-06-08:
Thanks. the revert worked in so far that blackmagic now compiled.

make PROBE_HOST=hosted 

worked. But running ./blackmagic gives:

```
$ ./blackmagic
Please implement find_debuggers for MACOS!
```

DanielO, 2022-06-09:
I applied this diff otherwise it could not link to the hidapi library:
```
diff --git a/src/platforms/hosted/Makefile.inc b/src/platforms/hosted/Makefile.inc
index 907b23a..2502539 100644
--- a/src/platforms/hosted/Makefile.inc
+++ b/src/platforms/hosted/Makefile.inc
@@ -40,9 +40,9 @@ LDFLAGS += -lsetupapi
 #https://github.com/dmlc/xgboost/issues/1945 indicates macosx as indicator
 else ifneq (filter, macosx darwin, $(SYS))
 SRC += serial_unix.c
-LDFLAGS += -lhidapi
+LDFLAGS += $(pkg-config --libs hidapi)
 LDFLAGS += -framework CoreFoundation
-CFLAGS += -Ihidapi/hidapi
+CFLAGS += -Ihidapi/hidapi $(pkg-config --cflags hidapi)
 HIDAPILIB = hidapi
 endif
```

As for implementing find_debuggers, perhaps it can be cribbed from Arduino / libserialport:
https://github.com/arduino/libserialport/blob/master/macosx.c

dragonmux, 2022-06-09:
The find_debuggers problem is an open issue (#824) which now we know has user interest still, we'll look at how to solve for this next release.

Qyriad, 2022-06-11:
8afaedd broke compatibility with GNU Make versions before 4.0, which as far as I can tell is when the `$(file)` function was added. macOS comes with Make 3.81 (as that's the last version before GNU Make switched to GPLv3). As a temporary workaround (aside from the revert) you can install GNU's Make with Homebrew (`brew install make`), and then build blackmagic with `gmake`.

dragonmux, 2022-06-11:
We have now opened a PR that addresses this. Please try it and let us know if indeed the MacOS build issues are fixed (we expect, as indicated in the PR, that find_debuggers will still be broken but at least your original problem should be properly solved)

chriskuku, 2022-06-12:
@dragonmux : Thanks a lot for the information. Since I'm not so fluent in git, could you give me a short hint how I can 
keep up with HEAD since I have reverted back to before the version.h issue creeped in.

Normall I would delete the directory and do a new git clone, but I'm sure there is a more elegant way to step up.

dragonmux, 2022-06-12:
fair ask - we'd do the following: `git reset HEAD~ && git fetch && git switch feature/bsd-gen-version-fix`.

The initial git command resets what your copy considers 'master' to be the same as origin/master again, the second then fetches updates from the remote, and the final switches to the feature branch the PR is based on. `git switch master` switches back

chriskuku, 2022-06-12:
Thanks. Since I obvioulsy tinkered with src/Makefile, the above sequence gives:

```
$ git reset HEAD~ && git fetch && git switch feature/bsd-gen-version-fix
Unstaged changes after reset:
M	src/Makefile
remote: Enumerating objects: 54, done.
remote: Counting objects: 100% (54/54), done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 54 (delta 37), reused 45 (delta 35), pack-reused 0
Unpacking objects: 100% (54/54), 22.56 KiB | 405.00 KiB/s, done.
From https://github.com/blackmagic-debug/blackmagic
   483e51d..af48a34  master     -> origin/master
 * [new branch]      feature/bsd-gen-version-fix -> origin/feature/bsd-gen-version-fix
 * [new branch]      feature/cmsis-dap-correctness -> origin/feature/cmsis-dap-correctness
error: Your local changes to the following files would be overwritten by checkout:
	src/Makefile
Please commit your changes or stash them before you switch branches.
Aborting
```
Would `git switch master`now help me out of the mess?

dragonmux, 2022-06-12:
`git stash` (if you want to keep the changes), or `git restore src/Makefile` if you don't would solve that - then re-run the git switch part of that

chriskuku, 2022-06-12:
Thanks. Could you give me a safer recipe. Too much hassle occurs when I try the various restores.
I did no changes that deserve to be kept or saved, So the straight command that gives me the version that I sould test , would be welcome.
```
$ pwd
/Users/kuku
$ cd Downloads/
$ cd blackmagic
$ git restore src/Makefile
$ git reset HEAD~ && git fetch && git switch feature/bsd-gen-version-fix
Unstaged changes after reset:
M	src/platforms/hosted/bmp_libusb.c
error: Your local changes to the following files would be overwritten by checkout:
	src/platforms/hosted/bmp_libusb.c
Please commit your changes or stash them before you switch branches.
Aborting
$ git restore src/platforms/hosted/bmp_libusb.c
$ git reset HEAD~ && git fetch && git switch feature/bsd-gen-version-fix
Unstaged changes after reset:
M	src/platforms/pc/cl_utils.c
error: Your local changes to the following files would be overwritten by checkout:
	src/platforms/pc/cl_utils.c
Please commit your changes or stash them before you switch branches.
Aborting
$ git restore src/platforms/pc/cl_utils.c
$ git reset HEAD~ && git fetch && git switch feature/bsd-gen-version-fix
Unstaged changes after reset:
M	src/platforms/hosted/bmp_libusb.c
error: Your local changes to the following files would be overwritten by checkout:
	src/platforms/hosted/bmp_libusb.c
Please commit your changes or stash them before you switch branches.
Aborting
$ get restore src/platforms/hosted/bmp_libusb.c
-bash: get: command not found
$ git restore src/platforms/hosted/bmp_libusb.c
$ git reset HEAD~ && git fetch && git switch feature/bsd-gen-version-fix
error: The following untracked working tree files would be overwritten by checkout:
	contrib/zsh/_blackmagic
Please move or remove them before you switch branches.
Aborting
$ 
```

dragonmux, 2022-06-12:
ah, we hadn't intended you to run the whole line again 😅 at this point the easiest thing to do is:
`git restore . && git pull && git switch git switch feature/bsd-gen-version-fix` - every time you re-run `git reset HEAD~`, you're walking the history on your 'master' back a step further which we only wanted to do once to undo the `git revert`.

chriskuku, 2022-06-12:
Had there been a typo in the line a few minutes before?:
`git restore . && git pull && git switch feature/bsd-gen-version-fix`
My line (copy/paste had been:
git restore . && git pull && git switch git switch feature/bsd-gen-version-fix

Now it gives:

```
git restore . && git pull && git switch git switch feature/bsd-gen-version-fix
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Updating 958a6d1..af48a34
error: The following untracked working tree files would be overwritten by merge:
	contrib/zsh/_blackmagic
Please move or remove them before you merge.
Aborting
$ pwd
/Users/kuku/Downloads/blackmagic
$ git restore . && git pull && git switch feature/bsd-gen-version-fix
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Updating 958a6d1..af48a34
error: The following untracked working tree files would be overwritten by merge:
	contrib/zsh/_blackmagic
Please move or remove them before you merge.
Aborting
```

chriskuku, 2022-06-12:
I will delete the whole directory tree and start over with git clone (would you give me the git clone URL, please?

dragonmux, 2022-06-12:
Ah, right.. forgot about that.. replace `git restore .` with `git reset --hard` if you want to give that route one last shot.

The clone URL is the normal one for the repo and you want the branch `feature/bsd-gen-version-fix`.

chriskuku, 2022-06-12:
I'm still not there :)

```
$ git reset --hard && git pull && git switch feature/bsd-gen-version-fix
HEAD is now at 958a6d1 hosted: Switched the `bmp_ident` output and printing method from stderr to stdout
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Updating 958a6d1..af48a34
error: The following untracked working tree files would be overwritten by merge:
	contrib/zsh/_blackmagic
Please move or remove them before you merge.
Aborting
```

dragonmux, 2022-06-12:
that's frustrating - oh well, `git clone https://github.com/blackmagic-debug/blackmagic -b feature/bsd-gen-version-fix` to make a fresh copy on the PR branch definitely works, so try that.

chriskuku, 2022-06-12:
```
$ make clean
  CLNHDR  include/libopencm3/stm32/f0/irq.json
  CLNHDR  include/libopencm3/stm32/f1/irq.json
  CLNHDR  include/libopencm3/stm32/f2/irq.json
  CLNHDR  include/libopencm3/stm32/f3/irq.json
  CLNHDR  include/libopencm3/stm32/f4/irq.json
  CLNHDR  include/libopencm3/stm32/f7/irq.json
  CLNHDR  include/libopencm3/stm32/l0/irq.json
  CLNHDR  include/libopencm3/stm32/l1/irq.json
  CLNHDR  include/libopencm3/stm32/l4/irq.json
  CLNHDR  include/libopencm3/stm32/g0/irq.json
  CLNHDR  include/libopencm3/stm32/g4/irq.json
  CLNHDR  include/libopencm3/stm32/h7/irq.json
  CLNHDR  include/libopencm3/gd32/f1x0/irq.json
  CLNHDR  include/libopencm3/lpc13xx/irq.json
  CLNHDR  include/libopencm3/lpc17xx/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m4/irq.json
  CLNHDR  include/libopencm3/lpc43xx/m0/irq.json
  CLNHDR  include/libopencm3/lm3s/irq.json
  CLNHDR  include/libopencm3/msp432/e4/irq.json
  CLNHDR  include/libopencm3/efm32/tg/irq.json
  CLNHDR  include/libopencm3/efm32/g/irq.json
  CLNHDR  include/libopencm3/efm32/lg/irq.json
  CLNHDR  include/libopencm3/efm32/gg/irq.json
  CLNHDR  include/libopencm3/efm32/hg/irq.json
  CLNHDR  include/libopencm3/efm32/wg/irq.json
  CLNHDR  include/libopencm3/efm32/ezr32wg/irq.json
  CLNHDR  include/libopencm3/sam/3a/irq.json
  CLNHDR  include/libopencm3/sam/3n/irq.json
  CLNHDR  include/libopencm3/sam/3s/irq.json
  CLNHDR  include/libopencm3/sam/3u/irq.json
  CLNHDR  include/libopencm3/sam/3x/irq.json
  CLNHDR  include/libopencm3/sam/4l/irq.json
  CLNHDR  include/libopencm3/sam/d/irq.json
  CLNHDR  include/libopencm3/vf6xx/irq.json
  CLNHDR  include/libopencm3/swm050/irq.json
  CLNHDR  include/libopencm3/pac55xx/irq.json
  CLEAN   lib/stm32/f0
  CLEAN   lib/stm32/f1
  CLEAN   lib/stm32/f2
  CLEAN   lib/stm32/f3
  CLEAN   lib/stm32/f4
  CLEAN   lib/stm32/f7
  CLEAN   lib/stm32/l0
  CLEAN   lib/stm32/l1
  CLEAN   lib/stm32/l4
  CLEAN   lib/stm32/g0
  CLEAN   lib/stm32/g4
  CLEAN   lib/stm32/h7
  CLEAN   lib/gd32/f1x0
  CLEAN   lib/lpc13xx
  CLEAN   lib/lpc17xx
  CLEAN   lib/lpc43xx/m4
  CLEAN   lib/lpc43xx/m0
  CLEAN   lib/lm3s
  CLEAN   lib/lm4f
  CLEAN   lib/msp432/e4
  CLEAN   lib/efm32/tg
  CLEAN   lib/efm32/g
  CLEAN   lib/efm32/lg
  CLEAN   lib/efm32/gg
  CLEAN   lib/efm32/hg
  CLEAN   lib/efm32/wg
  CLEAN   lib/efm32/ezr32wg
  CLEAN   lib/sam/3a
  CLEAN   lib/sam/3n
  CLEAN   lib/sam/3s
  CLEAN   lib/sam/3u
  CLEAN   lib/sam/3x
[... 364 lines cut ...]
  CC      target/rp.c
  CC      target/sam3x.c
  CC      target/sam4l.c
  CC      target/samd.c
  CC      target/samx5x.c
  CC      target/stm32f1.c
  CC      target/ch32f1.c
  CC      target/stm32f4.c
  CC      target/stm32h7.c
  CC      target/stm32l0.c
  CC      target/stm32l4.c
  CC      target/stm32g0.c
  CC      target/target.c
  CC      platforms/pc/serial_unix.c
  CC      timing.c
  CC      platforms/pc/cl_utils.c
  CC      platforms/pc/utils.c
  CC      platforms/hosted/bmp_remote.c
  CC      platforms/hosted/remote_swdptap.c
  CC      platforms/hosted/remote_jtagtap.c
  CC      platforms/hosted/bmp_serial.c
  LD      blackmagic
$ 
```
Both worked without a hitch.

dragonmux, 2022-06-12:
Excellent, thank you for confirming that fixes the original issue. Sorry about all the Git problems!

chriskuku, 2022-06-12:
No problem at all :) Excuse my ignorance what git commands /usage is concerned.

PR #1050 (2022-06-12): Fix version.h generation on BSD-like OSes
This PR does two things to fix #1043 - the first part is to switch to the alternative approach we indicated in #1018 as this PR caused a regression.

The second is to apply the HIDAPI pkg-config based library finding fix also presented in #1043 so we get the MacOS flavour of BMDA building again even if an outstanding issue that is the lack of implementation of find_debuggers() for MacOS is still present.
closed 2022-06-12 by esden