blackmagic-debug/blackmagic #1048: Blackpillv2 not working for old STM32F401CCU

Garag, 2022-06-11:
The old version of the Blackpill with a STM32F401CCU controller have only 256K flash and 64K RAM. The current linker script is for the STM32F401CEU with 512K Flash and 96K RAM . As the Stackpointer is after reset above 64K it will hard fault.
If i change the linker script to the lower values BMP works. Maybe it is possible to change the values in the linker script or use conditionally a different linker script for the old model.

During analysis of the problem i found that the pin PA1 is used twice. PA1 is used for JTAG TDI and for LED UART. I think it is better to use different pins.

esden, 2022-06-13:
Thank you for your contribution @Garag The patch that you proposed was attempting to attempting to change the [BlackPill V2](https://stm32-base.org/boards/STM32F411CEU6-WeAct-Black-Pill-V2.0) board. Not V1.2 or V3.0 that you are referring to. So I rejected that patch.

In general the policy is to not tack on platform variants onto each other, unless the firmware will be guaranteed to always work and take advantage of the hardware at hand. Each platform has to stay self contained with a specific pin-out and chip in mind. If changes are needed, you either have to add another target platform sub-directory or implement some runtime system for auto-detection of the host chip you have at hand. The latter would be the most powerful but also most difficult solution to implement. You are welcome to work on that and a patch that consolidates blackpill boards into one single firmware is something we can consider merging. Otherwise, if you want to support your specific blackpill hardware version, please make it a separate platform with the correct linker script, pin out and readme.

perigoso, 2022-06-14:
Of note it seems the smaller flash version is marked as discontinued, so the motivation to support it is smaller

perigoso, 2022-06-14:
They also seem to have another board they call a blackpill, with lower flash and a st "compatible" part

https://github.com/WeActTC/WeActStudio.BlackPill

Garag, 2022-06-14:
The native BMP have only 128KB Flash, so 256K Flash should be no problem.

djix123, 2022-06-27:
WeAct have a new blackpill board based on the Arterytex AT32F403ACGUx ... so maybe 'weact-blackpill-f4' is a better name.

[I have tried the current build on one of these new (AT32F403) boards, but the code enters an infinite loop in the clock configuration routine. If I get time I'll try and debug it]

dragonmux, 2022-06-27:
Given the plans that are in place in v2.0 to rewrite the build system and allow targets to be selectable in the build á la kernel drivers, it is our opinion that it will be best to have a linker script per WeAct board, and keep them as either separate platforms (or at least provide a simple way to select which you're building for).

With where the project is going we can make use of the extra Flash and that will be a useful thing going forward rather than going "let's pretend they're all the oldest slowest smallest processor".

We don't have an existing precedent for including a company name in a platform's name, so we're going to advice use `blackpill-f4`. Until v2 happens, it would be best if that was `blackpillv1-f4`, `blackpillv2-f4`, etc so users can simply `make PROBE_HOST=blackpillv2-f4` or w/e, but once v2 happens we can then merge them down into a single blackpill-f4 platform which has a CPU selector in the menuconfig.

djix123, 2022-06-27:
That makes sense to me

TechLobo, 2022-08-21:
> WeAct have a new blackpill board based on the Arterytex AT32F403ACGUx ... so maybe 'weact-blackpill-f4' is a better name.
>
> [I have tried the current build on one of these new (AT32F403) boards, but [...]

Did you get a chance to debug AT32F403ACGUx usage @djix123 ?
Would be interested to see how BMP works on this faster board.

djix123, 2022-08-21:
I didn't yet - I'm guessing you mean running the BMP firmware on the AT32F403ACGUx. I've been able to use a regular BMP to upload/debug on the AT32F403ACGUx even though it reports ```*** 1 Unknown ARM Cortex-M Designer 43b Partno 4c4 M4```. [Uploading to main flash]

TechLobo, 2022-08-21:
Yeah I did mean running the BMP firmware on the new WeAct blackpill board. Thanks.

djix123, 2022-08-21:
Which version of the firmware are you using - if I can find time I could look at current release ```1.8.2``` or current HEAD.

TechLobo, 2022-08-22:
Current release (1.8.2). 
However don't distract yourself from your own projects - its what keeps us going! ;)

TechLobo, 2022-11-11:
@djix123  - out of curiosity did you ever try to get the current firmware running on the new WeAct blackpill board?

djix123, 2022-11-16:
> @djix123 - out of curiosity did you ever try to get the current firmware running on the new WeAct blackpill board?

I haven't, I'm sorry - I moved on to other things. If I get some time over the next few weeks may try and look again.

djix123, 2022-12-24:
@TechLobo - I did have a play with the AT32F403ACGUx. Not sure this counts as progress - but If you build with PROBE_HOST=stink and load both 'blackmagic_dfu.bin' and 'blackmagic.bin' then the BMP firmware will run on the board. This makes sense since loads of these clones (even when M4 and quicker) have a clock setup and peripheral set that emulates/replicates the STM32F103. This doesn't maximize the system clock, but does appear to run. Next step will be to modify the start up code increase the frequency. I have the examples from the AT SDK download. Need to test and integrate.

djix123, 2022-12-25:
@TechLobo - very rough version working on AT32F403ACGUx with system clock speed 168Mhz. This is based on this libopencm3 [fork / at32f4 branch](https://github.com/beefdeadbeef/libopencm3.git). Note the the external clock isn't used, its all based off the HSI internal.

I've committed the changes necessary to build to my [fork / at32f4 branch](https://github.com/djix123/blackmagic.git) of the Blackmagic repository. Build with PROBE_HOST=stlink. The DFU build will fail, but the main firmware builds. To load you will need a regular (un-patched) stlink DFU build.

Its a bit mess, but did it in a bit of a rush. Hope this is helpful. I've included the necessary files in the attached zip.

[at32f403a_blackmagic.zip](https://github.com/blackmagic-debug/blackmagic/files/10300390/at32f403a_blackmagic.zip)

djix123, 2022-12-25:
Version running at 192Mhz clocked from external 8Mhz source

[at32f403a_blackmagic_ext_src.zip](https://github.com/blackmagic-debug/blackmagic/files/10300443/at32f403a_blackmagic_ext_src.zip)

djix123, 2022-12-25:
If libopencm3 accepts the pull request (which does seem some way off based on the comments), I could submit the build for the AT32F403ACGUx as a new platform as a pull request. Until then I'll try and keep my fork/branch update to date (or, at least periodically).

djix123, 2022-12-26:
Reviewing the AT32F403A/407 Reference Manual (v2.02) the maximum PLL divisor for the USB clock is 4. Therefore, it looks like the maximum PLL clock that allows USB to be enabled is 4*48 = 192Mhz.

TechLobo, 2023-01-01:
> Version running at 192Mhz clocked from external 8Mhz source
> 
> [at32f403a_blackmagic_ext_src.zip](https://github.com/blackmagic-debug/blackmagic/files/10300443/at32f403a_blackmagic_ext_src.zip)

@djix123 thanks for this - away at the moment so will be a few weeks before I can check this, but look forward to trying it!

Does the improved clock speed appear to give a noticeable performance improvement?

djix123, 2023-01-02:
@TechLobo I did enough to confirm the firmware would attach and upload a firmware to another device, but that's it really. I didn't put it through any performance test.

TechLobo, 2023-01-02:
@djix123 no problem, was just curious - thanks. Will find out when I get a chance to try it.

closed 2023-05-11 by dragonmux