blackmagic-debug/blackmagic #1054: Breakpoints no longer work on ESP32 host

xobs, 2022-06-14:
When using an ESP32 host, breakpoints no longer work. Single-stepping and code inspection work, however breakpoints in general never get hit.

Reverting 488df7e6f73c8cf53483a4e856c4aad51fda6330 fixes the issue.

dragonmux, 2022-06-14:
Hi xobs, thanks for reporting this. We'll dig into why this broke breakpoints for you when we've fixed some of the other outstanding issues.

Both the original code and the replacement is full of printf magic, which makes it hard to understand and debug, so we suspect it's going to take focusing for some time and dumping values out to figure out what's going on and why one construct works and the other doesn't.

xobs, 2022-06-15:
This is surprising, and I wouldn't be surprised if other targets had similar issues.

The problem stems from the change from `PRIx32` to `PRIX32`. Because gdb uses lowercase addresses, the uppercase `%X` does not match the lowercase hex addresses used by GDB.

As a side note, the sscanf argument changed from `%x` to `%08X` which may cause additional problems on certain platforms when addresses are not 8 digits, however this did not appear to be an issue on ESP32:

```
I (17654) handle_z_packet: broken sscanf("Z1,80004b4,2", "%*[zZ]%d,%08X,%d"): 1, 00013884, 536969216
I (17654) handle_z_packet: broken sscanf("Z1,80004b4,2", "%*[zZ]%d,%08x,%d"): 1, 080004b4, 2
I (17662) handle_z_packet: broken sscanf("Z1,80004b4,2", "%*[zZ]%d,%X,%d"): 1, 00013884, 2
I (17670) handle_z_packet: broken sscanf("Z1,80004b4,2", "%*[zZ]%d,%x,%d"): 1, 080004b4, 2
I (17678) handle_z_packet: working sscanf(",80004b4,2", ",%x,%d"): (1), 080004b4, 2
```

Simply changing the code from `PRIX32` to `PRIx32` fixes this problem for me.

dragonmux, 2022-06-15:
excellent, good to know. Thank you :)

closed 2022-06-25 by esden

dragonmux, 2022-06-25:
We know this got closed by PR 1066, but @xobs - could you check and verify that your issue is in fact fixed (the change we made restored breakpoints for the targets we have here, but we don't want to discount there being any compounding issues)

xobs, 2022-07-13:
I've verified this is now fixed on my platform as well.

PR #1066 (2022-06-25): Fix a formatting bug that breaks, among other things, breakpoints
This PR fixes a small regression that made it into v1.8 which breaks the ability to set and hit breakpoints. This fixes #1054.

This should be merged into both master and v1.8