blackmagic-debug/blackmagic #1060: black magic dfu driver for windows

TravisRo, 2022-06-21:
I created this with the libusbK inf-wizard.  It is a self-signing driver installer for your **blackmagic_upgrade.inf** and the libusb-win32 driver. Without it, you can't use the dfu-util on windows (MinGW64).  I though you might like to add it somewhere.

I also have the package source if your interested.

https://drive.google.com/file/d/1uQjDVqjnIZhEZV5bKzQ_MZEZhhcF_rOK/view?usp=sharing

TravisRo, 2022-06-22:
Well you can't install drivers on windows 7 x64 and up unless the driver package is signed, your also missing the driver itself libusb0.sys. As for the inf file, there is no difference.  It's identical.  I simply bundled it into an exe installer that self-signs the driver package and installs it with dpinst.exe (A Microsoft tool).  With out this, it's pretty useless your running windows XP or have test signing enabled. This is the same self signing technique used a zadig.

I wrote the libusbK driver and I'm also a maintainer of libusb-win32.  Here is a link to my google drive public folder that contains the package sources. 
https://drive.google.com/drive/folders/1LAQzlqmBcdrDZsq8cAAecI-tT-w1Nlqu?usp=sharing

Nearly everything in the package source was generated by **libusbK-inf-wizard**.  I simply drop in your inf file and run **re-pack-files.cmd**

I would be surprised if anyone has the DFU working on a Windows platform to be honest.  Windows drivers just happen to be my forte. :)

dragonmux, 2023-12-12:
The whole driver directory in the source tree needs some pretty hefty maintenance. You can use Zadig to bind WinUSB to the DFU and Trace Capture interfaces to get them working and happy, though this is not required for your BMP to work.

Blue Pill doesn't required patched firmware - if you've got the project's bootloader on your Blue Pill, you can `make PROBE_HOST=stlink BLUEPILL=1` and then DFU that on and it will just work. This is a well tested pathway.

Trace Capture is for SWO which is an optional feature. See the [using SWO](https://github.com/blackmagic-debug/blackmagic/blob/main/UsingSWO.md) guide for more information.