blackmagic-debug/blackmagic #1062: STLINKv2 blackmagic.bin not flashing

Na-Cly, 2022-06-23:
Hello. I compiled blackmagic.bin using the guide in the stlink folder. I try to use stlink-tool to flash the programmer and this is the output. 

please let me know if i'm doing something dumb or if this is a known issue i'm hitting. 

Thanks.

```
./stlink-tool blackmagic.bin 
Firmware version : V2J29S7
Loader version : 14152
ST-Link ID : 1521090032124353354B4E00
Firmware encryption key : 13FD63335262B06910F29B9ABE16B9BD
Current mode : 1
Loaded firmware : blackmagic.bin, size : 107696 bytes
.................................................Unexpected DFU status : 3
Erase error

```

dragonmux, 2022-06-23:
We can confirm this should not be happening:
```
❯ ./stlink-tool -p
Trying to switch STLINK/Application to bootloader
StlinkV21 Bootloader found
Firmware version : V2J40S27
Loader version : 14152
ST-Link ID : 0670FF535567494867085055
Firmware encryption key : D4E014681B4E998EB5DDFCB1FC5922CA
Current mode : 2
❯ ./stlink-tool ../blackmagic/src/blackmagic.bin
StlinkV21 Bootloader found
Firmware version : V2J40S27
Loader version : 14152
ST-Link ID : 0670FF535567494867085055
Firmware encryption key : D4E014681B4E998EB5DDFCB1FC5922CA
Current mode : 2
Type V2
.....................................................................................................
```

The first thing we did before that though was to upgrade our STLink firmware using the STLinkUpgrade.jar tool mentioned in the guide - you might wish to do that before trying again

UweBonnes, 2022-06-23:
Na-CLy: Are you sure you have a genuine STLINKV2 with a genuine STM32F103x8? It seems your loading aborted about halfways at the 64 kByte boarder and so your part does not have the upper flash that genuine STM32F103x8 has. So no chance for BMD, even with the BMD bootloader, as we fill up nearly all flash of the STM32xB.

UweBonnes, 2022-06-23:
Ah, good to see that V2J40S27 still works and ST did not add more crypto to inhibit reflash on the Stlinkv2 as they did with V3.

dragonmux, 2022-07-05:
Given it's been about two weeks and we couldn't reproduce the issue on genuine hardware, we're going to close this issue now. If you can bring any more information to the table to give us something actionable here or on the stlink-tool repo, please do re-open the issue.

@UweBonnes we have a PR outstanding on your repo  (UweBonnes/stlink-tool#2) which fixes building the tool on a modern compiler. It would be appreciated if you could look at and merge the fix.

closed 2022-07-05 by dragonmux