blackmagic-debug/blackmagic #1104: monitor swdp_scan not working with stlink platform and nrf52 target

MikeDK, 2022-07-13:
When flashing an stlink v2 platform with a blackmagic fw 1.8.0, the command "monitor swdp_scan" does not work - error message is: SW-DP scan failed!

When flashing the same version onto a 1bitsquared native platform, swdp scan is working flawlessly.

The target in both cases is a Nordic NRF52840.

I tried to get some more information about the failing scan, but unfortunately, when I compile the stlink platform fw with ENABLE_DEBUG=1, the resulting binary would exceed flash size and thus fails to link.
^-- is there maybe some compiler switch or macro definition to disable some unnecessary feature(s) in order to get the fw linked?

I will try to debug the bmp fw with a jlink using the .elf file, but this will need some time to do the necessary soldering on the stlink hw ;)

dragonmux, 2022-07-13:
Thank you for reporting this! Re ENABLE_DEBUG - the ability to turn bits of the firmware off like that will be coming in v2 - for now, the patch below should be enough to claw back sufficient Flash for a debug build for nRF52.

Appreciate the extra information, it is most helpful.

disable-lpc.patch:
```patch
Disables the LPC targets to enable `ENABLE_DEBUG=1` builds to fit when debugging other targets

diff --git a/src/Makefile b/src/Makefile
index 3e6b2096..62030fbb 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -9,7 +9,7 @@ Q := @
 endif

 CFLAGS += -Wall -Wextra -Werror -Wno-char-subscripts \
-	-std=gnu99 -g3 -MD -I./target \
+	-Wno-error=unused-function -std=gnu99 -g3 -MD -I./target \
 	-I. -Iinclude -I$(PLATFORM_DIR)

 ifeq ($(ENABLE_DEBUG), 1)
@@ -34,12 +34,6 @@ SRC =			\
 	jtag_devs.c	\
 	jtag_scan.c	\
 	lmi.c		\
-	lpc_common.c	\
-	lpc11xx.c	\
-	lpc17xx.c	\
-	lpc15xx.c	\
-	lpc43xx.c	\
-	lpc546xx.c	\
 	kinetis.c	\
 	main.c		\
 	morse.c		\
diff --git a/src/target/cortexm.c b/src/target/cortexm.c
index 3a60825a..1b969981 100644
--- a/src/target/cortexm.c
+++ b/src/target/cortexm.c
@@ -444,9 +444,6 @@ bool cortexm_probe(ADIv5_AP_t *ap)
 	case AP_DESIGNER_TEXAS:
 		PROBE(msp432_probe);
 		break;
-	case AP_DESIGNER_SPECULAR:
-		PROBE(lpc11xx_probe); /* LPC845 */
-		break;
 	default:
 		if (ap->ap_designer != AP_DESIGNER_ARM) {
 			/* Report unexpected designers */
@@ -461,36 +458,18 @@ bool cortexm_probe(ADIv5_AP_t *ap)
 		if (ap->ap_partno == 0x4c0)  { /* Cortex-M0+ ROM */
 			if ((ap->dp->targetid & 0xfff) == AP_DESIGNER_RASPBERRY)
 				PROBE(rp_probe);
-			PROBE(lpc11xx_probe); /* LPC8 */
 		} else if (ap->ap_partno == 0x4c3)  { /* Cortex-M3 ROM */
 			PROBE(ch32f1_probe);
 			PROBE(stm32f1_probe); /* Care for other STM32F1 clones (?) */
-			PROBE(lpc15xx_probe); /* Thanks to JojoS for testing */
-		} else if (ap->ap_partno == 0x471)  { /* Cortex-M0 ROM */
-			PROBE(lpc11xx_probe); /* LPC24C11 */
-			PROBE(lpc43xx_probe);
 		} else if (ap->ap_partno == 0x4c4) { /* Cortex-M4 ROM */
-			/* The LPC546xx and LPC43xx parts present with the same AP ROM Part
-			Number, so we need to probe both. Unfortunately, when probing for
-			the LPC43xx when the target is actually an LPC546xx, the memory
-			location checked is illegal for the LPC546xx and puts the chip into
-			Lockup, requiring a RST pulse to recover. Instead, make sure to
-			probe for the LPC546xx first, which experimentally doesn't harm
-			LPC43xx detection. */
-			PROBE(lpc546xx_probe);
-
-			PROBE(lpc43xx_probe);
 			PROBE(kinetis_probe); /* Older K-series */
 		} else if (ap->ap_partno == 0x4cb) { /* Cortex-M23 ROM */
 			PROBE(gd32f1_probe); /* GD32E23x uses GD32F1 peripherals */
-		} else if (ap->ap_partno == 0x4c0) { /* Cortex-M0+ ROM */
-			PROBE(lpc11xx_probe); /* some of the LPC8xx series, like LPC802 */
 		}
 		/* Info on PIDR of these parts wanted! */
 		PROBE(sam3x_probe);
 		PROBE(lmi_probe);
 		PROBE(ke04_probe);
-		PROBE(lpc17xx_probe);
 	}
 #undef PROBE
 	return true;

```

MikeDK, 2022-07-13:
Appreciation granted ;)
Thanks, I will try that asap

MikeDK, 2022-07-14:
PS: I just checked if git latest src is working without patch and enable debug -> yep ... and if I compile with patch but without debug enabled, then it works too

so it seems the ENABLE_DEBUG breaks something in an stlink build :/

MikeDK, 2022-07-14:
yeah but I need the ENABLE_DEBUG only because the swdp_scan does not work with 1.8+ bmp fw on an stlink probe :)

and for our project we need the bmp fw running on stlinks, because we already have 50 pieces of them :D

Just to be clear, this Issue is no blocker for us since yesterday, because I could get everything working with the patch from the flash buffer lifetime PR from @dragonmux if I apply it on version 1.7.1

I only wanted to report this so that stlink platform will maybe work again in the next major release ;)

MikeDK, 2022-07-14:
so if i tell gdb to use 0x8002000 as load address (like for the flat bin), it should work?

but even if not, eventually I should be able to flash the original bootloader hex/bin via swd ... if I can find this :D

dragonmux, 2022-07-14:
Exactly. Checking the .elf produced by `make PROBE_HOST=stlink ST_BOOTLOADER=1`, we see:
```
❯ arm-none-eabi-readelf -l src/blackmagic.elf

Elf file type is EXEC (Executable file)
Entry point 0x8014e81
There are 3 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x01d888 0x0801d888 0x0801d888 0x00008 0x00008 R   0x4
  LOAD           0x000000 0x08000000 0x08000000 0x1d890 0x1d890 R E 0x10000
  LOAD           0x020000 0x20000000 0x0801d890 0x00124 0x02d4c RW  0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx
   01     .text .ARM.exidx
   02     .data .bss
```

This might indicate a little work needs doing but, given the .bin is just an objcopy of this same binary and works.. we'd imagine this is correct/fine as it boots and works

MikeDK, 2022-07-14:
ok if i am brave enough I will try this in the evening ...

in the meantime I would have a feature request for the bmp firmware..: the possibility to use the stlink probe without any bootloader ... or maybe with the same bootloader as is compiled for the native hardware ... if this is even possible ;)

UweBonnes, 2022-07-14:
Compiling PROBE_HOST=stlink compiles the BMD bootloader adapted to the stlink platform with 4 kB reserved for the BMP bootloader. BMD starts immediate when plugged. To initially flash the BMD bootloader, you need hardware access to the STLINK SWD pins by soldering pins or wires to the approriate PCB places.

PROBE_HOST=stlink ST_BOOTLOADER=1 uses the ST provided bootloader that uses  8 Kb for the bootloader. You need stlink_tools to flash the BMD binary . When plugged, the ST bootloader starts. Running hosted, stlink-tools  or the ST tools is needed to start the BMD firmware.

A BMP w/o bootloader can only be updated with SWD access. This is a PITA with the stlink platform.So I think your feature request is already satisfied.

I tested main with a L152-Disco board and i could attach with BMD and BMDA. So please, test main with your original problem and if still there, run BMDA to get DEBUG immediate debug output.

MikeDK, 2022-07-14:
@UweBonnes I already have soldered some pins to the CN1 header and with the JLink it seems I got a stable SWD connection with jlinkgdbserver connected to the STM32F103CB.
So when I omit the ST_BOOTLOADER=1 part in the compilation, then the BMP bootloader will be built?
I will try this as it seems like the least complicated thing to do when I already have the SWD connection to the stlink cpu ;)

UweBonnes, 2022-07-14:
As I understand it, than you need to do that 50 times. If you use ST-build stlinks, then in that case I think the time is better spent to get things up without the solder modification to all boards. If you build your own stlink boards with SWD access available, your solution may be faster but will keep the original problem pending for BMD.

MikeDK, 2022-09-13:
update: some hours ago I realized that there is a "mon freq" command (since v1.8 i guess) ... so I first checked at which frequency the bmp is working when plugged into usb (3272727 Hz) ... then I used "mon freq 1000k" (to get it to 1384615Hz actually), and with this lower frequency setting, the swdp_scan command is working for me :)

That is good news, actually ;)

Nevertheless, I will now try to get some logic analyzer measurements of:
a.) working swdp_scan with v1.7.1 (which unintelligibly seems to work with the higher frequency)
b.) failing swdp_scan with v1.8.2 at 3.2 MHz
c.) working swdp_scan with v1.8.2 at 1.3 MHz

MikeDK, 2022-09-13:
ok I managed to get some swd line dumps with sigrok ... from the difference between the two v1.8.2 dumps I can tell that the real swd clock frequency is roughly 4.8 MHz instead of the 3.2 MHz which are shown to me when doing the monitor frequency command.
The real frequency when changing frequency with "monitor frequency 1000k" (which is shown as 1.3 MHz in gdb) is 922 kHz.

Have a look ...:
4.8 MHz:
[image]

922 kHz:
[image]

So it seems the correlation between calculated frequency (the one shown in gdb with mon freq) and the real frequency is somehow broken in the stlink platform (at least)

Attached you can find the two .SR files with the capture of the whole swdp_scan.
[bmp_swd_line_dumps.zip](https://github.com/blackmagic-debug/blackmagic/files/9558010/bmp_swd_line_dumps.zip)

I will also make a dump of the v1.7.1 later ;)

dragonmux, 2022-11-18:
@MikeDK we recently got #1220 merged which theoretically should have fixed the issues you were having - could you please test `main` and see if SWD scan is now working (reliably) for your target?

If it is, we'll close this issue.

MikeDK, 2022-11-21:
Ok I can try but not promise to test it this week as I was recently assigned to another project, but many thanks for the update  ;)

dragonmux, 2023-01-25:
As it's been a couple more months and we've not had anyone else report this issue, we will close it. If the issue is not resolved, please let us know and we'll re-open it and work with you further to solve it.

closed 2023-01-25 by dragonmux