blackmagic-debug/blackmagic #1105: No recovery from SWD parity error over CMSIS-DAP

mubes, 2022-07-13:
When a parity error is received over CMSIS-DAP communication with the target is instantly dropped. There does not appear to be any retry. This is polling after a 'continue' with the '09' representing a correctly received message with a parity error;
```
cmd :   05.00.05.08.00.00.00.00.01.52.00.00.23.05.f0.ed.00.e0.0f.0e.
cmd res:05.05.01.01.00.01.01.01.00.01.01.
memread @ e000edf0 len 4, align 2 , start: 
cmd :   05.00.05.08.00.00.00.00.01.52.00.00.23.05.f0.ed.00.e0.0f.0e.
cmd res:05.05.01.01.00.01.01.01.00.01.01.
memread @ e000edf0 len 4, align 2 , start: 
cmd :   05.00.05.08.00.00.00.00.01.52.00.00.23.05.f0.ed.00.e0.0f.0e.
cmd res:05.05.01.01.00.01.01.01.00.01.01.
memread @ e000edf0 len 4, align 2 , start: 
cmd :   05.00.05.08.00.00.00.00.01.52.00.00.23.05.f0.ed.00.e0.0f.0e.
cmd res:05.05.01.01.00.01.01.01.00.01.01.
memread @ e000edf0 len 4, align 2 , start: 
cmd :   05.00.05.08.00.00.00.00.01.52.00.00.23.05.f0.ed.00.e0.0f.0e.
cmd res:05.05.01.01.00.01.01.01.00.01.01.
memread @ e000edf0 len 4, align 2 , start: 
cmd :   05.00.05.08.00.00.00.00.01.52.00.00.23.05.f0.ed.00.e0.0f.0e.
cmd res:05.05.01.01.00.01.01.01.00.01.01.
memread @ e000edf0 len 4, align 2 , start: 
cmd :   05.00.05.08.00.00.00.00.01.52.00.00.23.05.f0.ed.00.e0.0f.0e.
cmd res:05.05.09.
TARGET LOST.
```

Platform was ORBTrace with STM32L151.

UweBonnes, 2022-07-14:
For a short test for BMDA CMSIS error recovery, can you please try to change in dap.c/wait_word() from
	if(buf[1] == SWDP_ACK_FAULT) {
to
      if(buf[1] >  SWDP_ACK_WAIT) {
Thanks

dragonmux, 2023-01-31:
After talking with mubes about this on Discord, we have agreed to close this and related PRs as fixed with the understanding it can be re-opened in the future if it's not fully resolved.

closed 2023-01-31 by dragonmux