blackmagic-debug/blackmagic #1117: Flashing fails over CMSIS-DAP at high speed

mubes, 2022-07-16:
This may well relate to #1105 

If you flash at high speed over the CMSIS-DAP interface quite often the results are incorrect and fail `compare-sections`;

```
(gdb) file build/STM32F427.elf
(gdb) monitor frequency 25000000
Max SWJ freq 017d7840
(gdb) load
Loading section .isr_vector, size 0x1ac lma 0x8000000
Loading section .text, size 0x14d0 lma 0x80001ac
Loading section .rodata, size 0x4c lma 0x800167c
Loading section .ARM, size 0x8 lma 0x80016c8
Loading section .init_array, size 0x4 lma 0x80016d0
Loading section .fini_array, size 0x4 lma 0x80016d4
Loading section .data, size 0xc lma 0x80016d8
Start address 0x80015b8, load size 5860
Transfer rate: 22 KB/sec, 488 bytes/write.
(gdb) compare-sections 
Section .isr_vector, range 0x8000000 -- 0x80001ac: matched.
Section .text, range 0x80001ac -- 0x800167c: MIS-MATCHED!
Section .rodata, range 0x800167c -- 0x80016c8: matched.
Section .ARM, range 0x80016c8 -- 0x80016d0: matched.
Section .init_array, range 0x80016d0 -- 0x80016d4: matched.
Section .fini_array, range 0x80016d4 -- 0x80016d8: matched.
Section .data, range 0x80016d8 -- 0x80016e4: matched.
warning: One or more sections of the target image does not match
the loaded file

(gdb) 
(gdb) monitor frequency 10000000
Max SWJ freq 00989680
(gdb) load
Loading section .isr_vector, size 0x1ac lma 0x8000000
Loading section .text, size 0x14d0 lma 0x80001ac
Loading section .rodata, size 0x4c lma 0x800167c
Loading section .ARM, size 0x8 lma 0x80016c8
Loading section .init_array, size 0x4 lma 0x80016d0
Loading section .fini_array, size 0x4 lma 0x80016d4
Loading section .data, size 0xc lma 0x80016d8
Start address 0x80015b8, load size 5860
Transfer rate: 23 KB/sec, 488 bytes/write.
(gdb) compare-sections 
Section .isr_vector, range 0x8000000 -- 0x80001ac: matched.
Section .text, range 0x80001ac -- 0x800167c: matched.
Section .rodata, range 0x800167c -- 0x80016c8: matched.
Section .ARM, range 0x80016c8 -- 0x80016d0: matched.
Section .init_array, range 0x80016d0 -- 0x80016d4: matched.
Section .fini_array, range 0x80016d4 -- 0x80016d8: matched.
Section .data, range 0x80016d8 -- 0x80016e4: matched.
(gdb) 
```
This test passes on openocd (with suitable adjusted commands).

Test was conducted using ORBTrace V1.1.0InProgress and STM32F427. Contact me for the elf file if it's useful.

dragonmux, 2022-07-16:
Thank you for reporting this! Going to assume two things here, please correct us if we're wrong..

The first is that similarly to the linked issue which we know you're working on, you have a fix in mind and a way around this you intend to PR.

The second is that this still happens even with the fix/tap-types-and-understandability branch we pushed yesterday which is in testing to be PR'd?

mubes, 2022-07-16:
Hi. Just checked while I've got the rig up. Behaviour appears to be the same on b98b1e9 (tap-types-and-understandabilty) and 1b8e522 (main).

I will dig into this, but not until ORBTrace 1.1.0 is done....these are mostly placeholders for now.  

Users are advised to do a compare-sections after loading over cmsis-dap at any speed as a precaution in the interim.

dragonmux, 2022-07-17:
Thank you for confirming that the branch both works the same and hasn't regressed at least. Shame it didn't fix this though

closed 2022-08-08 by esden