blackmagic-debug/blackmagic #1126: Manual tdi control for lpc82x bootloader

tdaede, 2022-07-20:
lpc82x (and probably other LPC series) have an embedded bootloader that is enabled when a GPIO is pulled low on boot. This happens to be the same GPIO pin mapped to TDI.

If using a connector wired for full JTAG but connecting via SWD, the BMP will by default pull TDI low, which will make the lpc82x always execute its bootloader. However, running a jtag_scan first will leave the TDI pin in a high state, allowing normal program execution.

This is actually sort of handy, because entering the bootloader this way seems to be the only way to recover a device that enters power-down mode where clocks have stopped. It would be nice to have it be an intentional monitor command though, rather than happening by chance if jtag_scan is run.

dragonmux, 2022-07-20:
Thank you for opening this issue! This seems like a valuable enhancement to include in the project. We could do with a small bit of extra information though in order to make the feature this gives rise to truly useful:

You mention TDI state, is this in combination with pinging reset like the scan commands do, or is this something where TDI state needs to be set, BMP plugged in, then the board powered on? We can imagine this working either way, so it would be helpful to have a short walk-through of the intended series of interactions to make use of the feature.

tdaede, 2022-07-20:
Yes, it would be in combination with pinging reset. 

There are really two parts of this feature - one is letting tdi float during power on rather than being set low, which allows the lpc82x to operate normally. This could potentially have an impact on other targets, however considering the pin is not used for SWD, and jtag_scan actively toggles in and leaves it in the opposite state, I think it's not likely.

Of course, because I was intentionally relying on this behavior to help program boards stuck in a low power mode, I'd like some replacement - for example, I'd use a script with some new monitor commands like follows:

```
monitor pin tdi low
monitor hard_srst
monitor pin tdi high
monitor swdp_scan
```

Alternately, the sequence could be one command, for more guaranteed timing.

Note that on my chip, I can't scan if connect_srst is true. I haven't yet figured out if that's a chip feature or BMP bug - if the latter, fixing that would make this unnecessary. I haven't had time to research this yet.

dragonmux, 2022-07-26:
Please give the linked PR a try and tell us if you find it a satisfactory solution.

PR #1136 (2022-07-30): Feature: LPC82x low power reset
This PR implements a new command for doing resets while holding pins in the JTAG interface in specific state. The reason this is necessary is given by LPC82x processors where if in a deep low-power mode do not have running clocks but if reset with TDI held low, will wind up in their bootloaders, allowing BMP to scan for them and attach.

This fixes #1126
closed 2022-07-30 by esden