blackmagic-debug/blackmagic #1130: Allow bypassing JTAG scan sanity checks

ylm, 2022-07-23:
When scanning out IRs IEEE 1149.1 mandates that the first bit output be set to 1.
BMP verify this compliance and errors out should the devices in the chain not comply to IEEE 1149.1.
Sadly, some devices do not comply to the standard and include non-compliant devices in their chains.

dragonmux, 2022-07-26:
Could you verify this is still broken with #1131 which rewrote a significant amount of the JTAG machinery? If it is we'll look at what can be done to work around this issue.

It would be helpful if you could give example devices so we can look at purchasing test hardware to A:B

ylm, 2022-07-26:
Yep, [this line](https://github.com/blackmagic-debug/blackmagic/blob/a0f39c31fb042364f98ba4ba09da14dd3842e813/src/target/jtag_scan.c#L102) and [this line](https://github.com/blackmagic-debug/blackmagic/blob/a0f39c31fb042364f98ba4ba09da14dd3842e813/src/target/jtag_scan.c#L118) are the culprits.

I'm currently targeting iMX6Q on the Novena. I could dig the OpenOCD repo for other targets with such characteristics.

dragonmux, 2022-07-26:
Perfect, thank you :) we'll give this a look and get back to you with a PR patch if we can without breaking lots of stuff

ylm, 2022-07-26:
Emitting a warning, but without the early bail out could do in a pinch.

dragonmux, 2022-07-26:
aye, that's true - actually a decent approach so a user knows they're working with a non-conforming part but we try soldiering on anyway..

Our biggest worry is that it'll result in a lower degree of robustness in the face of potentially very screwed up scan chains or even being connected to not-a-scan-chain and that resulting in the code doing whack random stuff as a result.. but that probably already causes problems that require BMP to be unplugged and re-plugged, so it seems like a reasonable compromise

ylm, 2022-07-26:
Targets supported by OpenOCD which don't respect ieee 1149.1 are the following:
em357, imx25, imx51, imx53, imx6, is5114, samsung_s3c2450, stm32w108xx and u8500.

dragonmux, 2022-07-27:
Please try the above linked PR which should now fix the problem and make the JTAG scan code easier to work with and understand.

ylm, 2022-07-29:
I don't know why yet, but the PR breaks the JTAG scan entirely, and the returned IR value is stuck at one.

dragonmux, 2022-07-29:
Ah, good to know, we'll turn that one into a draft then and dive in further.

ylm, 2022-07-29:
With patch:
```(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon jtag_scan 4 4 5
Resetting TAP
Given list of IR lengths, skipping probe
Change state to Shift-IR
Return to Run-Test/Idle
Change state to Shift-DR
jtag_scan: Sanity check failed: BYPASS dev count doesn't match IR scan
JTAG device scan failed!
Failed
```

Without patch:
```(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) mon jtag_scan 4 4 5
Resetting TAP
Given list of IR lengths, skipping probe
Change state to Shift-IR
check failed: IR[0] != 1
JTAG device scan failed!
Failed
```

Without patch, but early bail-out removed:
```(gdb) mon jtag_scan 4 4 5
Resetting TAP
Given list of IR lengths, skipping probe
Change state to Shift-IR
check failed: IR[0] != 1
Return to Run-Test/Idle
Change state to Shift-DR
Return to Run-Test/Idle
Return to Run-Test/Idle
DPIDR 0x4ba00477 (v0 rev4)
RESET_SEQ failed
Available Targets:
No. Att Driver
```

dragonmux, 2022-07-29:
In the case of run 3 there, is that with #1135 applied too? If so, then we'll just make sure Esden merges that before the linked PR

ylm, 2022-07-29:
No, that's without 1135. I need to test that one too without linked PR.

dragonmux, 2022-07-29:
If you could provide details then on the PR of what the early bailout code you're referring to is, we'll look at updating the PR to include your fix that results in run 3

ylm, 2022-07-29:
This is what I meant:
```diff --git a/src/target/jtag_scan.c b/src/target/jtag_scan.c
index 4addc49..9c55d44 100644
--- a/src/target/jtag_scan.c
+++ b/src/target/jtag_scan.c
@@ -101,7 +101,7 @@ int jtag_scan(const uint8_t *irlens)
                        jtag_proc.jtagtap_tdi_tdo_seq((uint8_t*)&irout, 0, ones, *irlens);
                        if (!(irout & 1)) {
                                DEBUG_WARN("check failed: IR[0] != 1\n");
-                               return -1;
+                               //return -1;
                        }
                        jtag_devs[jtag_dev_count].ir_len = *irlens;
                        jtag_devs[jtag_dev_count].ir_prescan = j;
```

ylm, 2022-07-29:
I checked again and it I erroneously tested 2&3 against master instead of main. Testing against main gives me the same result as 1. So the issue would have been introduced between master's last commit and current main.

PR #1138 (2022-07-31): Fix: Unable to detect non-conforming JTAG processors
In this PR we address that some processors don't have conforming implementations of IEEE 1149.1 (JTAG) which prevents jtag_scan() from functioning correctly for them.

We relax the conformance checks to warnings, and address the state of the JTAG scan code, performing type fixes and nomenclature cleanup.

This has been tested against STM32F4, LPC3470, and TM4C.

This fixes #1130.
closed 2022-07-31 by esden