blackmagic-debug/blackmagic #1140: LPC1343 Problem

oleg-login, 2022-07-28:
Hello is problem LPC1343,

(gdb) monitor swdp_scan
Target voltage: 3.3V
Please report probed device with Designer code 0x8000 and Partno 0x 0
Available Targets:
No. Att Driver
*** 1 Unknown ARM Cortex-M Designer 0x8000 Partno 0x000 M3
(gdb)

Earlier, when there was no check ap->designer_code and ap->ap_partno, everything worked well.

perigoso, 2022-07-28:
@oleg-login can you confirm you were able to probe and detach the LPC1343 before? With what firmware version? the patch you commented on should not have broken anything, so likely something else caused the issue

dragonmux, 2022-07-28:
Thank you for reporting this! With the extra information Perigoso gave that this is a legacy (non-JEP-106) designer code, we'll dive into why this is happening and how to fix it.

For what it's worth, this was probably a smoldering issue that the ADIv5 probe rework showed up - a bug hidden by other bad/buggy code. That it worked before is why we're marking this as a regression.

oleg-login, 2022-07-28:
> 
(gdb) mon version
Black Magic Probe v1.8.0-276-g7c14b62, Hardware Version 3
Copyright (C) 2022 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor swdp_scan
Target voltage: 3.3V
Please report probed device with Designer code 0x8000 and Partno 0x  0
Available Targets:
No. Att Driver
*** 1 Unknown ARM Cortex-M Designer 0x8000 Partno 0x000 M3

But, with version release v1.8.2
(gdb) monitor swdp_scan
Target voltage: 3.3V
Please report probed device with Designer code 0 and Partno  0
Available Targets:
No. Att Driver
*** 1 Unknown ARM Cortex-M Designer 0 Partno 0

perigoso, 2022-07-28:
seems like they are showing the same behavior, with the exception for the designer code change

when you say `Earlier, when there was no check ap->designer_code and ap->ap_partno, everything worked well.` what do you mean?

because if you mean the by now very old code that did not check for which probe to run with the designer code, it would make sense and align with the issue I tried to fix which was actually a regression

dragonmux, 2022-07-28:
Looks just as broken both ways - though at least after Perigoso's patch it's properly detected as a Cortex-M3 core.. If you could op onto the 1b² Discord, we have a debug build of BMP to share with you to get additional information from what's going on.

oleg-login, 2022-07-28:
when you say Earlier, when there was no check ap->designer_code and ap->ap_partno, everything worked well. what do you mean?
 Release v1.7.1

dragonmux, 2022-07-28:
There was a big rewrite by Uwe to the ADIv5 probe code near the start of what became v1.8, this is where your regression comes from and is what Perigoso is referring to with the regression he's been chasing.

oleg-login, 2022-07-28:
[dragonmux], For debugging, I can run your debug build BMP and provide you with a log.

perigoso, 2022-07-28:
@oleg-login can you try the patch #1141 

@dragonmux this is not ideal, there must be a way to fetch info on those devices, but it should at least have things working
(I'm betting it has something to do with the "PIDR")

dragonmux, 2022-07-28:
Correct, oleg - if you can pop onto https://1bitsquared.com/pages/chat and look in the `#blackmagic` channel, you'll find a suitable debug build.

Agreed it has to do with the PIDR, perigoso

oleg-login, 2022-07-28:
Thank you! I will do and provide a report.

oleg-login, 2022-07-29:
Hello. I run debug build BMP.
Received the following data. Maybe it will help to add LPC1343 support?

(gdb) mon version
Black Magic Probe v1.8.0-276-g7c14b62-dirty, Hardware Version 3
Copyright (C) 2022 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
(gdb) monitor debug_bmp enable
Debug mode is enabled
(gdb) monitor swdp_scan
Target voltage: 3.3V
Please report probed device with Designer code 0x8000 and Partno 0x  0
Available Targets:
No. Att Driver
*** 1 Unknown ARM Cortex-M Designer 0x8000 Partno 0x000 M3

***DEBUG PRINT***
DPIDR 0x2ba01477 (v1 rev0)
RESET_SEQ failed
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040 (AHB-AP var1 rev2)
Halt via DHCSR: success 01030003 after 0ms
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 8000 Partno   0
0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x04002bb000  DEVTYPE = 0x0hx ARCHID = 0x0000-> cortexm_probe
CPUID 0x412fc230 (M3 var 2 rev 0)
1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04002bb002  DEVTYPE = 0x0hx ARCHID = 02 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003  DEVTYPE = 0x0hx ARCHID = 3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04002bb001  DEVTYPE = 0x0hx ARCHID 4 0xe0040000: Debug component - Cortex-M3 TPIU (Trace Port Interface Unit) (PIDR = 0x04002bb923  DEVTYPE = 0x0hx ARCHID = 0x0015 0xe0041000: Debug component - Cortex-M3 ETM (Embedded Trace) (PIDR = 0x04002bb924  DEVTYPE = 0x0hx ARCHID = 0x0013)
ROM: Table END

dragonmux, 2022-07-29:
Thank you for the extra output oleg-login, that looks like it should be most useful for digging in and getting started!

oleg-login, 2022-07-29:
And thank you for making a huge contribution to the development of the free firmware and debugger.

perigoso, 2022-08-01:
Hello @oleg-login, would you mind trying the current main branch? this should hopefully be fixed

oleg-login, 2022-08-02:
Hello @perigoso. Okay, I'll try

oleg-login, 2022-08-02:
@perigoso, it is OK. Thank you very much!
 
(gdb) mon version
Black Magic Probe v1.8.0-362-gae23cb8-dirty, Hardware Version 3
Copyright (C) 2022 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor swdp_scan
Target voltage: 3.3V
You are now detached from the previous target.
Available Targets:
No. Att Driver
 1      LPC1343 M3
(gdb)

dragonmux, 2022-08-02:
We'll close this issue as solved then - thank you :)

closed 2022-08-02 by dragonmux