blackmagic-debug/blackmagic #1142: Cannot upgrade firmware

silverv, 2022-07-28:
I am trying to upgrade firmware on my native Blackmagic (https://shop.zubax.com/products/dronecode-probe-v2-2-jtag-swd-adapter-usb-uart-adapter-with-dcd-m-cable?variant=8797640687715) to try and replicate a bug from issue #1128 .

`
silver@silver:~/zubax/blackmagic$ scripts/stm32_mem.py blackmagic.bin
Traceback (most recent call last):
  File "/home/silver/zubax/blackmagic/scripts/stm32_mem.py", line 27, in <module>
    import dfu
  File "/home/silver/zubax/blackmagic/scripts/dfu.py", line 21, in <module>
    import usb
ModuleNotFoundError: No module named 'usb'
silver@silver:~/zubax/blackmagic$ pip^Cripts/stm32_mem.py blackmagic.bin
silver@silver:~/zubax/blackmagic$ pip install pyusb
Defaulting to user installation because normal site-packages is not writeable
Collecting pyusb
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Installing collected packages: pyusb
Successfully installed pyusb-1.2.1
silver@silver:~/zubax/blackmagic$ scripts/stm32_mem.py blackmagic.bin
-
USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
-
Device ID:       1d50:6018
Manufacturer:    Black Sphere Technologies
Product:         Black Magic Probe
Serial:          C2D6BBE4
Failed to read device state! Assuming APP_IDLE
Invoking DFU Device
Error: DFU device did not appear
silver@silver:~/zubax/blackmagic$
`

silverv, 2022-07-28:
This is what I see with dmesg
```
[ 5068.360617] usb 1-3.4: usbfs: interface 0 claimed by cdc_acm while 'python3' sets config #1
[ 5068.365997] usb 1-3.4: usbfs: interface 0 claimed by cdc_acm while 'python3' sets config #1
[ 5068.397702] usb 1-3.4: USB disconnect, device number 10
[ 5068.684704] usb 1-3.4: new full-speed USB device number 11 using xhci_hcd
[ 5068.831011] usb 1-3.4: New USB device found, idVendor=1d50, idProduct=6017, bcdDevice= 1.00
[ 5068.831030] usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5068.831034] usb 1-3.4: Product: Black Magic Probe (Upgrade)
[ 5068.831038] usb 1-3.4: Manufacturer: Black Sphere Technologies
[ 5068.831041] usb 1-3.4: SerialNumber: C2D6BBE4
```

silverv, 2022-07-28:
So it does make the device go into bootloader or this upgrade mode but then fails.

dragonmux, 2022-07-28:
Rather than using the stm32_mem.py script which should be considered deprecated for native platform BMPs, please dfu-util as in `dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D src/blackmagic.bin`. It is better tested, maintained and more robust.

silverv, 2022-07-28:
Okay, I believe it worked.
```
silver@silver:~/zubax/blackmagic$ sudo dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D blackmagic-native-v1_8_2.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1d50:6017
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08002000, size = 104608
Download        [=========================] 100%       104608 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

dragonmux, 2022-07-28:
correct, that's a good update.

dragonmux, 2022-07-28:
Closing as won't fix as the stm32_mem.py script is provided only as historical interest and not maintained by the project at this time due to internal developments. Users can always use dfu-util to update their firmware instead.

closed 2022-08-09 by esden