blackmagic-debug/blackmagic #1158: BMP hangs debug logic on iMX6

ylm, 2022-08-04:
When enumerating entries in the ROM table, memory accesses to debug peripheral related to disabled cores hangs the memory bus deadlocking the AP. AP accesses return the WAIT status, AP's CSW register report a transaction in progress and DP's CTRL/STAT register returns an unclearable STICKYERR despite the ABORT instruction being sent.
Accesses from the processor rather than the AP exhibit the same behaviour.

I'm looking for guidance on how to solve this with BMP (if possible).

So far, I see two possible solutions:

1. Enable all the processors prior ROM table enumeration (either with BMP or manually through some other means)
2. Limit the number of discoverable targets (with a monitor command) and return early once we've reached the limit.

I'll wait for feedback on this. For now, I think I can live with option 1, but I'm tempted to implement option 2. Let me know what you think.

dragonmux, 2022-08-04:
So, we have two questions we'd like answered before we weigh in directly on the question presented - and that is:

1. Is the first core guaranteed to be running?
2. Does BMP have any guaranteed way on the target to read out via the DP or first AP which processors are enabled and so avoid the ones that aren't during scan?

Depending on the options that leaves us with we'd say either option 1 or a third option which is "Skip APs in the DP that are for disabled processors"

ylm, 2022-08-04:
1. Core 0 can never be disabled. It always present and always functional.
2. Untested, but I can think of a way that would be doable. This method would be target dependant.

To clarify, the debug peripherals are all accessed through the same AP, but through different ROM entries. Otherwise it wouldn't be an issue: the affected APs would deadlock themselves, but could become available as the processors would become enabled and wouldn't block access to the other debug components.

To work properly in my case, option 3 would need to be "skip ROM table entries in this specific AP related to disabled processors" at which point I don't need to enumerate the ROM table. I just hardcode the necessary AP data in BMP and use that instead.

dragonmux, 2023-01-30:
👋🏼  ylm, any news on this?

ylm, 2023-01-30:
I personally avoided the issue by manually enabling all the processors before ROM initialization with u-boot's CLI.
If there is an interest for it, I can help automating the process, but the changes would be significant.

dragonmux, 2023-01-30:
Given that might help with dealing with things like i.MXRT power down states when `ONOFF` has not been triggered.. and generally improve the readiness of the code base in future.. we'd be interested in a patch for this