blackmagic-debug/blackmagic #1159: stm32h750 target wrong part_id read as 0x4500 not 0x450

klenSA, 2022-08-04:
(gdb) tar ext /dev/kgp/bmp
Remote debugging using /dev/kgp/bmp
(gdb) mon auto
Target voltage: 0.0V
JTAG scan found no devices, trying SWD!
Please report unknown device with Designer 20 Part ID 450
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x4500 M7
(gdb) mon revision
Target does not support this command.
(gdb)

dragonmux, 2022-08-04:
Thank you for reporting this. As shown in the message in the middle, the part ID is read out correctly (need to look into why it's formatted wrong for the driver string but that's a separate problem).

What it looks like is happening here is that the stm32h7_probe routine failed in some manner.
It's also interesting that your probe is reporting a target voltage of 0V

Can we confirm that this is with latest v1.8.2 firmware or main?

If you are able to do a debug-enabled build of the firmware, that would be very helpful in figuring out what's really going wrong here. To make such a build requires building with ENABLE_DEBUG=1 and then once Flashed, running `mon debug en` just before invoking auto-scan while monitoring the debug serial port with minicom or similar (with the project udev rules that looks like: `minicom -D /dev/ttyBmpTarg -8`).

perigoso, 2022-08-05:
This is very likely a regression of #1146

The different Part IDs can be explained by the fact that I missed a change from ´ap->partno' -> `t->part_id` (will PR a fixup in sec)
The Id actually being used to probe is `t->part_id` which is as of the new change 0x4500, and 0x450 would be the old ID (`ap->partno`)

Without more information we can't know for certain what is happening, but I can make an assumption I think is likely:
The `stm32h750` has a valid TARGETID register and as such the PARTNO present in it is being used to run the probe routine, but this PARTNO has one extra nibble over the one in the ap, I was hoping they were the same as to not break any probe routines, but it seems they're not, the first 3 nibbles match, it just has the extra nibble that shows as an extra 0 at the end

There's a couple ways we could approach a fix for this, and it really depends on how other targets behave, what I think is the most correct probably is to update the ids used in `stm32h7_probe` and other targets that have TARGETID available to match it, the problem is that if a device is probed by jtag we won't have it has the register is fetched only on swd scan routines, we really ought to fetch both DPIDR and TARGETID if available on `adiv5_init` as these are features of the ADI, not SWD, and should be available on JTAG too, but currently only use the JTAG IDCODE

perigoso, 2022-08-05:
note:
PARTNO has different sizes on the different components
JTAG IDCODE: 4 nibbles
DP DPIDR: 2 nibbles
AP PIDR: 3 nibbles
DP TARGETID: 4 nibbles

dragonmux, 2022-08-05:
Ah, thanks for that Perigoso :) We'll review the PR v. shortly then

dragonmux, 2022-08-05:
That's the unknown part printout regression fixed

dragonmux, 2022-08-05:
We would suggest that updating part IDs and making the readout of key part identification information specific to ADIv5 should indeed be in the non-protocol-specific adiv5.c routines - the way it's currently split and how that gives rise to different behaviour based on underlying protocol seems really bad and something we'd like to see fixed in v1.9 as it's been biting us with the LPC43xx work too

closed 2022-08-06 by dragonmux