blackmagic-debug/blackmagic #1165: update problem: scripts/stm32_mem.py no compatible device found

klenSA, 2022-08-05:
Attempting to update results in a message that compatible devices were not found

`klen@klen-station:/klen/home/src/bmp$ python scripts/stm32_mem.py src/blackmagic.bin
-
USB Device Firmware Upgrade - Host Utility -- version 1.2
Copyright (C) 2011  Black Sphere Technologies
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
-
No DFU devices found!
`

klenSA, 2022-08-05:
proposed patch:
```diff --git a/scripts/stm32_mem.py b/scripts/stm32_mem.py
index a33c12f..dc3e9c2 100755
--- a/scripts/stm32_mem.py
+++ b/scripts/stm32_mem.py
@@ -101,7 +101,12 @@ def stm32_scan(args, test):
                        continue
 
                man = dfudev.handle.getString(dfudev.dev.iManufacturer, 30)
-               if man == b"Black Sphere Technologies":
+
+               # esden committed on 31 May 5e3cadec20a5fa3ec6eef53f23520707d0db5380
+               #    https://github.com/blackmagic-debug/blackmagic/commit/5e3cadec20a5fa3ec6eef53f23520707d0db5380
+               # Since this commit manufacturer is not "Black Sphere Technologies", It was
+               # renamed to "Black Magic Debug" instead. Saved the ability to update outdated firmware.
+               if man == b"Black Sphere Technologies" or b"Black Magic Debug":
                        bmp = bmp + 1
                        bmp_devs.append(dev)
```

dragonmux, 2022-08-05:
It's on our list of things to fix - this is actually a documentation issue and that script is being deprecated. Please use dfu-util á la `dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D src/blackmagic.bin`.

We will be updating that script to emit a deprecation warning and fixing the documentation in the next week.

PR #1185 (2022-08-09): Fix: Add deprecation warnings and remove stale CI configuration

## Detailed description

This PR removes the stale Travis CI configuration as we no longer use that service. It also introduces deprecation notices into the stm32_mem.py script and blackmagic_upgrade utility as these are to be slated for removal in v2 and alternatives in [bmputil](https://github.com/blackmagic-debug/bmputil) and dfu-util already exist.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [gu [... cut]
closed 2022-08-09 by esden