blackmagic-debug/blackmagic #1204: BMPA build broken on OSX: aliase are not supported on darwin

djix123, 2022-08-22:
Building the latest HEAD revision is broken on OSX. Build fails with the following messages:

```
target/target_probe.c:50:58: error: aliases are not supported on darwin
bool cortexa_probe(ADIv5_AP_t *apb, uint32_t debug_base) CORTEXA_PROBE_WEAK_NOP;
                                                         ^
target/target_probe.c:22:53: note: expanded from macro 'CORTEXA_PROBE_WEAK_NOP'
#define CORTEXA_PROBE_WEAK_NOP __attribute__((weak, alias("cortexa_probe_nop")))
```

dragonmux, 2022-08-28:
Thank you for bringing this to our attention. Amy has kindly written a fix for this in the attached PR - if you'd be kind enough to test that fixes things for you that'd be grand.

This issue is set to close with that PR merging, please re-open this if it is not fixed after that point.

closed 2022-08-28 by dragonmux

djix123, 2022-08-28:
Build on OSX confirmed as working

commit 5e98f30f (2022-08-28): target: fix weak aliasing of probe NOPs on macOS

AppleClang has several incompatibilities with __attribute__((weak,
alias)) that prevent sane GCC-like usage on macOS:

- it does not support __attribute__((alias)) at all, only the #pragma
  weak symbol1 = symbol2 syntax [1]

- the compiler removes unused static functions prior to aliasing, which
  renders the above into an error [2].

To fix this, I implemented weak aliasing using the #pragma directive and
the recommended indirection macros from the GCC docs [3] and removing
the static inline specifiers from the functions if building under mac [... cut]