blackmagic-debug/blackmagic #1230: LPC812,824,844 do not program

Squantor, 2022-09-02:
On git head (TODO hash) I am unable to program the following devices:
LPC812, LPC824, LPC844

When I run the following debug commands:
```
set print pretty on
set architecture armv6
file bin/LPC824M201HI33_blinky/debug/LPC824M201HI33_blinky.elf
target extended-remote /dev/ttyACM0
monitor tpwr enable
shell sleep 1
monitor swdp_scan
attach 1
set mem inaccessible-by-default off
set {int}0x40048000 = 2
load bin/LPC824M201HI33_blinky/debug/LPC824M201HI33_blinky.elf
set {int}0x40048000 = 2
compare-sections
b main
```
It hangs at the Loading section
```
For help, type "help".
Type "apropos word" to search for commands related to "word".
The target architecture is set to "armv6".
Enabling target power
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x M0+
0x00000164 in main () at projects/LPC824M201HI33_blinky/src/main.cpp:25
25          var = var ^ 0x55;
Loading section .text, size 0x280 lma 0x0
```

Using git bisect from the last good known hash (6C700F7) but could not find the commit that is responsible for the break as I got some compilation errors (hashes 0a6621a3, 8eea6110, 9b6d9c47) .

jerry200202, 2022-09-02:
31adb2f works fine.
After 31adb2f many chips not work any more.

Squantor, 2022-09-05:
I have been using the commit jerry200202 provided (31adb2f94db8b17212a104c83fe61d42b42c568c) to do a new git bisect and the result is that aec7460426893953f2a0ee6bb4766c92d09359c1 seems to be the culprit.

Squantor, 2022-09-05:
@perigoso I have tried that commit number and it works. When doing the git bisect I did encounter a build error so I just did filled in the last git bisect result to "skip" over the sections that have compilation errors.

dragonmux, 2023-02-12:
👋🏼  sorry this seems to have slipped under the radar. A lot of improvements and changes were made since this issue was raised in regards Flash function return values and fixing small logic mistakes that were made when cleaning up all that code.

@Squantor could you let us know if this is still an issue in v1.9, or if we can close this?

Squantor, 2023-02-12:
I will make a test suite with the LPC devices that share common code, as I use them semi regularly (switching to RP2040 anyways due to chip shortage). I will recheck all the devices and a few in the same family as I have a bunch of devkits with these devices. Do you want me to only check 1.9 release?

I will check LPC812, 824, 844 etc and report here what the results are. I have to dig up these devkits and make a nice test setup, so in the next few days.

dragonmux, 2023-02-12:
v1.9 release or even current main should both be sufficient as we've not poked that LPC code further since the release was made. If you test both for good measure, we'll be able to know if it's definitely still an issue in the current dev state or not as well as the release

Squantor, 2023-02-19:
I have tested the following versions:
main: output from mon version: ```Black Magic Probe v1.9.0-178-ga7364d1a, Hardware Version 4```
1.9.0: output from mon version: ```Black Magic Probe v1.9.0, Hardware Version 4```
known good: output from mon version: ```Black Magic Probe v1.7.1-230-g41719c55, Hardware Version 3```

Hardware used is a an official Black magic probe V2.1e

This is the script that I used for all of my tests:
```
set print pretty on
set architecture armv6
file (file for each board)
target extended-remote /dev/ttyACM0
monitor tpwr enable
shell sleep 1
monitor swdp_scan
attach 1
set mem inaccessible-by-default off
set {int}0x40048000 = 2
load (file for each board)
set {int}0x40048000 = 2
compare-sections
```
Results:
LPC812M101DH20: git main: fails with a hang, git tag 1.9.0: fails with a hang, known good works
LPC824M201HI33: git main: fails with a hang, git tag 1.9.0: fails with a hang, known good works
LPC844M201BD64: git main: fails with a hang, git tag 1.9.0: fails with a hang, known good works

A hang is described as hanging in gdb at the line ```Loading section .text size 0xsomething lma 0x0``` this hangt persists until I CTRL-C, I have waited for 10 minutes+.

Reconnecting after one such hang results in:
```
observation: after disconnect, reconnect I get these messages: 
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
```
What can I do to help you further? I have the following tools that I can use:

- FT2322H board, I have tried the semihosting variant but I could not get it to work to get more datapoints
- Jlink segger, segfaults when using git main and semihosting
- High speed logic analyser (400MHz 4 channels)
- 100MHz oscilloscope

I am also willing to send a few boards like a LPC812, LPC824 and such to somebody, these are minimal development boards and can be connected via 1.27mm pitch header to the black magic probe without issue. The SWD pins can easely be probed via broken out 2.54mm pinheaders.

dragonmux, 2023-02-19:
If you could, please build fix/arm-jtag-regressions with `ENABLE_DEBUG=1` (you'll need to delete some target support .c lines from src/Makefile for it to fit) and retry while monitoring the secondary serial connection (/dev/ttyBmpTarg if on Linux using the project udev rules) and adding `monitor debug enable` after `target extended-remote` in your script. This should provide a significantly greater insight into what's going on and avoid some bugs on main which we've fixed since asking you to test.

This looks like possibly a logic issue introduced with the Flash function rewrite in v1.9 (v1.8.2 should work - it would be useful check to help A-B when this broke)

Squantor, 2023-02-20:
Thanks for the update and further instructions, I will do the following:
I will test v1.9, v1.8.2 and fix/arm-jtag-regressions with ```ENABLE_DEBUG=1``` and the LPC812 as I believe most of the LPC800 series have similar handling. I will put/attach serial traces here from the secondary serial connection.

Squantor, 2023-02-24:
I have done some tests with LPC824 with 1.8.2, 1.9, fix/arm-jtag-regressions. The results in short are works, hang, hang. I have included the traces below:
LPC824 with 1.8.2 debug trace:
```
DPIDR 0x0bc11477 (v1 MINDP rev0)
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=f0000003 CSW=03000040 (AHB-AP var3 rev0
Halt via DHCSR: success 01030003 after 0ms
ROM: Table BASE=0xf0000000 SYSMEM=0x00000001, designer 501 Partno   0
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, designer 43b Partno 4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008  DEVTYPE = 0x0hx ARCHID = 0x000CPUID 0x410cc601 (M0+ var 0 rev 1)
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a  DEVTYPE = 0x0hx ARCHID =  2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b  DEVTYPE = 0x0hx ARCHID = 0x0000)
 ROM: Table END
ROM: Table END
```
Runs normally with single stepping and continue.

LPC824 with 1.9 debug trace:
```
scan_multidrop: false
DP DPIDR 0x0bc11477 (v1 MINDP rev0) designer 0x43b partno 0xbc
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=f0000003 CSW=03000040 (AHB-AP var3 rev0)
Halt via DHCSR: success 00030003 after 0ms
ROM: Table BASE=0xf0000000 SYSMEM=0x00000001, Manufacturer 501 Partno   0
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x00000004000bb008  DEVTYPE = 0x00 ARCHID = 0x0000)
 -> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x00000004000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x00000004000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM: Table END
ROM: Table END
```
Hangs, no output from debug serial port.

LPC824 with fix/arm-jtag-regressions debug trace:
```
scan_multidrop: false
DP DPIDR 0x0bc11477 (v1 MINDP rev0) designer 0x43b partno 0xbc
RESET_SEQ failed
AP   0: IDR=04770031 CFG=00000000 BASE=f0000003 CSW=03000040 (AHB-AP var3 rev0)
Halt via DHCSR(00030003): success after 0ms
ROM: Table BASE=0xf0000000 SYSMEM=0x00000001, Manufacturer 501 Partno   0
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c0
 0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x00000004000bb008  DEVTYPE = 0x00 ARCHID = 0x0000)
 -> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
 1 0xe0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x00000004000bb00a  DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0xe0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x00000004000bb00b  DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM: Table END
ROM: Table END
```
Hangs, no output from debug serial port.

Can I try anything else?

dragonmux, 2023-03-04:
Sorry it's taken a while to get back to you!

Given you're testing atop the fix/arm-jtag-regressions branch; with the debug build of the firmware on your probe, please do a debug build of BMDA (`make PROBE_HOST=hosted ENABLE_DEBUG=1`) and run it with `src/blackmagic -v 21 2>lpc824-bmda-bmp.log` as the LPC support being so silent on the matter is rather problematic for debugging. -v 21 will enable INFO, TARGET and WIRE messages from BMDA which should tell what's going on. Likewise for the v1.8.2 firmware + BMDA so we have a point of comparison.

Hopefully the transaction traces should shed some light on what's going on here and allow us to fix this

Squantor, 2023-03-04:
I have tested with the LPC824M201HI33:
BMDA on fix/arm-jtag-regressions and firmware on fix/arm-jtag-regressions. Results in a hang and quickly growing log file, it continued to grow after I stopped the GDB session and had to CTRL-C the BMDA software. [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10889659/LPC824-FWfix-BMDAfix.log)

BMDA on fix/arm-jtag-regressions and firmware on 1.8.2. Results in ```Floating point exception (core dumped)``` from the BMDA and a error in GDB on the line ```load bin/LPC824M201HI33_tests/debug/LPC824M201HI33_tests.elf```.
[LPC824-FW182-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10889664/LPC824-FW182-BMDAfix.log)

I hope the traces are useful. If you want me to try any other combinations or something else, feel free to ask.

dragonmux, 2023-03-04:
If you can do firmware + BMDA both as v1.8.2 that would be handy - we'll look at fixing that FPE on fix/arm-jtag-regressions as that's a bug on that branch then

Squantor, 2023-03-04:
BMDA on 1.8.2 and firmware on 1.8.2. Results in normal operation, I only loaded the firmware, no single stepping nor running. [LPC824-FW182-BMDA182.log](https://github.com/blackmagic-debug/blackmagic/files/10889704/LPC824-FW182-BMDA182.log)

dragonmux, 2023-03-04:
ah, something of note for when you wake: it looks like you're on an old version of fix/arm-jtag-regressions given that firmware answer behaviour.

Please `git fetch --all --prune` and then `git switch -C fix/arm-jtag-regressions origin/fix/arm-jtag-regressions` (or upstream if that's what the remote for this repo is called on your working copy). We'd had to force-push the branch a few times to fix regressions the fixes caused, and that DHCSR halt behaviour is looking like the memory write bug that we since fixed on that branch.

Squantor, 2023-03-05:
Whoops, offcourse this is a pretty lively branch and forgot to update. I ran all your commands and redid the tests. The FW on 1.8.2 and BMD on 1.8.2 I did not rerun as that works.

FW on fix branch and BMDA on fix branch results in a floating point exception on the BMDA software. [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10891383/LPC824-FWfix-BMDAfix.log)
FW on 1.8.2 and BMDA on fix branch results in a floating point exception on the BMDA software. [LPC824-FW182-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10891387/LPC824-FW182-BMDAfix.log)
FW on fix and BMDA on 1.8.2 results in gdb error on the line ```attach 1```, BMDA software does not crash. [LPC824-FWfix-BMDA182.log](https://github.com/blackmagic-debug/blackmagic/files/10891391/LPC824-FWfix-BMDA182.log)

dragonmux, 2023-03-05:
Been trying to figure out where and how that branch is SIGFPE'ing and we can't honestly tell.. it seems to detect the processor correctly and then just.. stops.. so; could we please ask you to run BMDA in GDB w/ fix branch firmware on the probe and using the fix branch built BMDA?

Something like `make PROBE_HOST=hosted ENABLE_DEBUG=1 OPT_FLAGS='-Og -g'` (from clean) then `gdb src/blackmagic -tv 1` and post the backtrace given where the exception triggers

compuphase, 2023-03-07:
I have the same problem on an LPC11U14.
Attached is a log of running BMDA in GDB, It shows a SIGFPE in target_add_flash(). Is this what you mean?
[issue1230-gdb.log](https://github.com/blackmagic-debug/blackmagic/files/10911503/issue1230-gdb.log)

I have been bold enough to add an assert at the start of the function. Then it ends with:
```
ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 471
0 0xe000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x00000004000bb008  DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x410cc200 (M0 var 0 rev 0)
Calling lpc11xx_probe
blackmagic: target/target.c:192: target_add_flash: Assertion `f->blocksize != 0' failed.

Thread 1 "blackmagic" received signal SIGABRT, Aborted.
```
I think this points us in the right direction.

dragonmux, 2023-03-07:
Please give fix/lpc11xx-lpc8xx-identification a go which should now fix the FPE and allow us to debug the underlying issue here (this branch is based on top of fix/cortexm-halt-resume-correctness for now to allow testing of the combined fixes)

compuphase, 2023-03-08:
I confirm that this fixes the issue.
You probably already know this, but the "propagation of writebufsize" part of the fix (commit 966b2ec2e9a019ff1505c530d0f0e40405c205b8) also applies to lpc15xx.c, lpc17xx.c, lpc43xx.c and lpc546.c.

dragonmux, 2023-03-08:
aye, we wanted to be sure we'd fixed the correct thing before spending the time and effort to rewrite lpc_add_flash() to do the right thing here, fixing all of the LPC support at once

dragonmux, 2023-03-09:
We have now updated the fix/lpc11xx-lpc8xx-identification branch with a more comprehensive fix for all LPC parts now we know we've correctly identified the problem that was causing SIGFPE. If either of you could please provide us logs using that branch of the Flash programming problem we will endeavour to work on a fix for that too

Squantor, 2023-03-10:
I am back, sorry for not finding enough time to test but it was a busy few weeks. So here are some test results:
With firmware on 1.8.2 and BMDA on 1.8.2 I get the following in gdb:
```
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x M0+
0x10000340 in defaultTestTable ()
Loading section .text, size 0x1738 lma 0x0
Loading section .data, size 0xa lma 0x1738
Start address 0x000000c0, load size 5954
Transfer rate: 4 KB/sec, 744 bytes/write.
Section .text, range 0x0 -- 0x1738: matched.
Section .data, range 0x1738 -- 0x1742: matched.
```
Everything works fine as before, but I had an issue after programming fix branch, see notes below. Included Log: [LPC824-FW182-BMDA182.log](https://github.com/blackmagic-debug/blackmagic/files/10941478/LPC824-FW182-BMDA182.log)

With firmware on 1.8.2 and BMDA on fix/lpc11xx-lpc8xx-identification I get the following in gdb:
```
Target voltage: 3.3V
Available Targets:
Available Targets:
No. Att Driver
 1      LPC82x M0+
Dummy_Handler () at targets/src/LPC824.cpp:22
22        while (1)
Loading section .text, size 0x1738 lma 0x0
Loading section .data, size 0xa lma 0x1738
Start address 0x000000c0, load size 5954
Transfer rate: 988 bytes/sec, 744 bytes/write.
Section .text, range 0x0 -- 0x1738: matched.
Section .data, range 0x1738 -- 0x1742: matched.
```
I did a continue and everything seems to work fine, log: [LPC824-FW182-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10941517/LPC824-FW182-BMDAfix.log)

With firmware on fix/lpc11xx-lpc8xx-identification and BMDA on fix/lpc11xx-lpc8xx-identification I get the following in gdb:
```
Target voltage: 3.3V
Available Targets:
Available Targets:
No. Att Driver
 1      LPC82x M0+
0xfffffffe in ?? ()
Loading section .text, size 0x1738 lma 0x0
Loading section .data, size 0xa lma 0x1738
Start address 0x000000c0, load size 5954
Transfer rate: 2 KB/sec, 744 bytes/write.
Section .text, range 0x0 -- 0x1738: MIS-MATCHED!
Section .data, range 0x1738 -- 0x1742: MIS-MATCHED!
warning: One or more sections of the target image does not match
the loaded file
```
Included log: [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10941355/LPC824-FWfix-BMDAfix.log)
When I use FW on fix branch and BMDA on the fix branch combination and then tried the 1.8.2 Firmware and BMDA combination I was unable to program the LPC824M201. I had to resort to a segger to access and erase the chip. I have added the log of FW 1.8.2 and BMDA 1.8.2 trying and failing to program the LPC824: [LPC824-FW182-BMDA182-stuck.log](https://github.com/blackmagic-debug/blackmagic/files/10941493/LPC824-FW182-BMDA182-stuck.log)

I am using the following gdb script:
```
set print pretty on
set architecture armv6
file bin/LPC824M201HI33_tests/debug/LPC824M201HI33_tests.elf
target extended-remote localhost:2000
shell sleep 1
monitor swdp_scan
attach 1
set mem inaccessible-by-default off
set {int}0x40048000 = 2
load bin/LPC824M201HI33_tests/debug/LPC824M201HI33_tests.elf
set {int}0x40048000 = 2
compare-sections
```
The ```set {int}0x40048000 = 2``` are needed as at boot the boot ROM is placed in the memory space.

Hope this helps.

compuphase, 2023-03-10:
Attached is the log of BMDA running in test mode, under GDB.
[issue1230-gdb-2.log](https://github.com/blackmagic-debug/blackmagic/files/10941507/issue1230-gdb-2.log)

Also attached are the logs of BMDA when running it with options "-pv 1", and then running GDB to download a "blinky" program into an LPC11U14. This was successful (though slow).
[issue1230-bmda-3.log](https://github.com/blackmagic-debug/blackmagic/files/10941693/issue1230-bmda-3.log)
[issue1230-gdb-3.log](https://github.com/blackmagic-debug/blackmagic/files/10941694/issue1230-gdb-3.log)

Please note the double line "Available Targets:" in the GDB log. This also happens when I put this firmware in the native probe and use it directly with GDB (so, without BMDA in the middle). There is a delay after the line "Target voltage: 3.3V". When issuing `set debug remote 1` first, I see that GDB signals a timeout after that line and sends a `NAK`. Right after that, the rest of the `swdp_scan` text follows.

Nota Bene (Squantor posted his reply while I was typing mine), my native BMP is hardware version 3 (BMP v2.1) and firmware version 1.7.1.

dragonmux, 2023-03-10:
For some weird and intriguing reason @Squantor your run with the fix branch firmware and BMDA resulted in DHCSR halt failing which is why part of the results you saw. We have just rebased that branch to flow in some fixes done v. late last night (though that was for CMSIS-DAP) but it's /possible/ that'll impact this..
The extra logs are fantastically useful though for tracing what's going on here, so thank you

dragonmux, 2023-03-11:
We have just pushed a pile of correctness fixes and a rewrite of a chunk of the IAP code to fix/lpc-iap-correctness, which fix/lpc11xx-lpc8xx-identification is rebased on top of. This may not fix the Flash programming problems, but it should make the whole mess better behaved and stop it triggering undefined behaviour, making results more repeatable.

Squantor, 2023-03-11:
@dragonmux Shall I proceed to retest with fix/lpc-iap-correctness or have the changes there been pulled into fix/lpc11xx-lpc8xx-identification?

dragonmux, 2023-03-11:
The changes there are reflowed such that you'll get them on fix/lpc11xx-lpc8xx-identification. Given the bug we're looking to isolate here, we suggest only retesting the updated fix branch firmware and fix branch BMDA as that's the log we're focused on comparing against the v1.8.2 firmware+BMDA log

Squantor, 2023-03-11:
Will do! I have some time today to retest all three combinations, FW on fix, BMDA on fix and both on fix. Will report later during the day when I have done these tests. I really need to automate them :-).

Squantor, 2023-03-11:
Here I am with some results:
With firmware on 1.8.2 and BMDA on fix/lpc11xx-lpc8xx-identification I get the following in gdb:
```
Target voltage: 3.3V
Available Targets:
Available Targets:
No. Att Driver
 1      LPC82x M0+
Dummy_Handler () at targets/src/LPC824.cpp:22
22        while (1)
./gdb_scripts/LPC824M201HI33_tests_debug.txt:10: Error in sourced command file:
Error erasing flash with vFlashErase packet
```
Log: [LPC824-FW182-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10948245/LPC824-FW182-BMDAfix.log)

With firmware on fix and BMDA on 1.8.2 I get the following in gdb:
```
Target voltage: 3.3V
SW-DP scan failed!
Failed
./gdb_scripts/LPC824M201HI33_tests_debug.txt:7: Error in sourced command file:
Attaching to Remote target failed
```
Log: [LPC824-FWfix-BMDA182.log](https://github.com/blackmagic-debug/blackmagic/files/10948282/LPC824-FWfix-BMDA182.log)

With firmware on fix and BMDA on fix I get the following in gdb:
```
Target voltage: 3.3V
Available Targets:
Available Targets:
No. Att Driver
 1      LPC82x M0+
0x1fff00d6 in ?? ()
./gdb_scripts/LPC824M201HI33_tests_debug.txt:10: Error in sourced command file:
Error erasing flash with vFlashErase packet
```
Log: [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10948284/LPC824-FWfix-BMDAfix.log)

dragonmux, 2023-03-11:
It looks like there's an additional issue with how lpc_add_flash() is working and the state it's leaving the lpc_flash_s object it creates in.. We've updated fix/lpc11xx-lpc8xx-identification with some additional diagnostics in lpc_flash_erase() to help track down what's going on there. Please give it a whirl - you only need to retest the fix firmware+BMDA combination.

The fix firmware + v1.8.2 BMDA will be broken without passing `-H` to BMDA as that BMDA doesn't know how to talk to the newer firmware's remote protocol

Squantor, 2023-03-11:
I did a ```git fetch --all --prune``` then a ```git pull```

With firmware on fix and BMDA on fix I get the following in gdb:
```
Target voltage: 3.3V
Available Targets:
Available Targets:
No. Att Driver
 1      LPC82x M0+
0x1fff00da in ?? ()
./gdb_scripts/LPC824M201HI33_tests_debug.txt:10: Error in sourced command file:
Error erasing flash with vFlashErase packet
```
Log: [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/10948390/LPC824-FWfix-BMDAfix.log)

I am away for the rest of the day, hopefully I have some time this evening to do any tests you want me to do.

Squantor, 2023-03-11:
With firmware on fix and BMDA on 1.8.2 with the -H flag I get the following in gdb:
```
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x M0+
0x1fff00d6 in ?? ()
Loading section .text, size 0x1738 lma 0x0
Loading section .data, size 0xa lma 0x1738
Start address 0x000000c0, load size 5954
Transfer rate: 1 KB/sec, 744 bytes/write.
Section .text, range 0x0 -- 0x1738: matched.
Section .data, range 0x1738 -- 0x1742: matched.
```
Log: [LPC824-FWfix-BMDA182.log](https://github.com/blackmagic-debug/blackmagic/files/10949370/LPC824-FWfix-BMDA182.log)

Note: I also did some single steps and a continue in that log to just check functionality, it all works fine. Dont know how useful this test is.

dragonmux, 2023-03-15:
We've spent a while doing a pile of sleuthing in the IAP handling and found and fixed a couple of bugs + greatly improved the error reporting - please give fix/lpc11xx-lpc8xx-identification another whirl after updating your working copy (we had to force-push again, so you'll need to do a suitable dance with Git).

we only need the FWfix+BMDAfix combination re-tested.

Squantor, 2023-03-17:
I am back from a week at embedded world, ready to test. 
I have done a ```git fetch --all --prune``` and a ```git pull``` to make sure I am up to date.

FW on fix, BMDA on fix. GDB output:
```
Target voltage: 3.3V
Available Targets:
Available Targets:
No. Att Driver
 1      LPC82x M0+
Dummy_Handler () at targets/src/LPC824.cpp:22
22        while (1)
./gdb_scripts/LPC824M201HI33_tests_debug.txt:10: Error in sourced command file:
Error erasing flash with vFlashErase packet
```
Output log: [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/11006012/LPC824-FWfix-BMDAfix.log)

dragonmux, 2023-03-26:
There have been a series of small fixes here and there on the lineage of branches that fix/lpc-iap-correctness lives on solving various things including an issues with DHCSR halt and an IAP ROM issue with the LPC8xx parts. 

We don't think it's quite fixed in full yet thanks to the work and contribution of @TomCrypto on Discord, but if you could re-test all the same (fix firmware + fix BMDA on branch fix/lpc11xx-lpc8xx-identification) and provide us with the logs that would be incredibly helpful to figuring out this final piece of the puzzle on what's going wrong

Squantor, 2023-03-28:
I have done a git fetch --all --prune and a git pull to make sure I am up to date.
FW on fix, BMDA on fix. GDB output:
```
The target architecture is set to "armv6".
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x M0+
Dummy_Handler () at targets/src/LPC824.cpp:22
22        while (1)
Loading section .text, size 0x1738 lma 0x0
Loading section .data, size 0xa lma 0x1738
Start address 0x000000c0, load size 5954
Transfer rate: 1 KB/sec, 744 bytes/write.
Section .text, range 0x0 -- 0x1738: matched.
Section .data, range 0x1738 -- 0x1742: matched.
(gdb) s
54        dst = &_data_start;
(gdb) s
55        while (dst < &_data_end) *dst++ = *src++;
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
main () at projects/tests_LPC824M201HI33/src/main.cpp:30
30        while (1) __BKPT(0xFE);
(gdb) quit
```
Output log: [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/11092249/LPC824-FWfix-BMDAfix.log)

As you can see by the output, IT WORKS! Let me try some other microcontrollers:

LPC812: Works too, log: [LPC812-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/11092269/LPC812-FWfix-BMDAfix.log)
LPC844: [LPC844-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/11092284/LPC844-FWfix-BMDAfix.log)

dragonmux, 2023-03-28:
Most excellent! @TomCrypto found an additional small thing that needs doing to make this reliable across LPC8xx parts, but we'll take that win.

We'll open a PR for this once the backlog's been merged, and tag the issue when we do. Thank you for your time and patience with us as we worked through this and developed a fix.

Squantor, 2023-03-29:
I will test LPC1114 and 1225 when I have time and report the results here.

dragonmux, 2023-04-04:
We've now opened a draft PR for the IAP corrections - once we've got your yay/nay that it works properly, we'll convert it over to ready to review

Squantor, 2023-04-07:
I have done a ```git fetch --all --prune```, and afterwards i saw that ```fix/lpc11xx-lpc8xx-identification``` is gone from origin, is this intentional?
I have tried two branches, ```fix/lpc11xx-lpc8xx-identification``` and ```fix/lpc-iap-correctness```.
I had to add the device signature to lpc11xx.c where LPC11xx-XL devices are detected so I can flash LPC112X devices. Maybe an idea to add these signiatures?
```
	case 0x00140040U: /* LPC1124/303 - 32K Flash 8K SRAM */
	case 0x00150080U: /* LPC1125/303 - 64K Flash 8K SRAM */
```

flashing LPC824M201JHI33 with just a BMP running ```fix/lpc11xx-lpc8xx-identification``` works.
flashing LPC812M101JDH20 with just a BMP running ```fix/lpc11xx-lpc8xx-identification``` works.
flashing LPC1114BD48/323 with just a BMP running ```fix/lpc11xx-lpc8xx-identification``` works.
flashing LPC1125BD48/303 with just a BMP running ```fix/lpc11xx-lpc8xx-identification``` works. 

flashing LPC824M201JHI33 with BMDA BMP combination running ```fix/lpc11xx-lpc8xx-identification``` works.
flashing LPC812M101JDH20 with BMDA BMP combination running ```fix/lpc11xx-lpc8xx-identification``` works.
flashing LPC1114BD48/323 with BMDA BMP combination running ```fix/lpc11xx-lpc8xx-identification``` works.
flashing LPC1125BD48/303 with BMDA BMP combination running ```fix/lpc11xx-lpc8xx-identification``` works. 

flashing LPC824M201JHI33 with just a BMP running ```fix/lpc-iap-correctness``` fails, gdb output:
```
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x M0+
0x000001e8 in main () at projects/LPC824M201HI33_blinky/src/main.cpp:25
25          var = var ^ 0x55;
./gdb_scripts/LPC824M201HI33_blinky_debug.txt:8: Error in sourced command file:
Error erasing flash with vFlashErase packet
```
flashing LPC824M201JHI33 with BMDA BMP combination running ```fix/lpc11xx-lpc8xx-identification``` fails the same, log included: [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/11178126/LPC824-FWfix-BMDAfix.log)

dragonmux, 2023-04-07:
Hi Squantor, thanks for the testing - fix/lpc11xx-lpc8xx-identification went away because the branch it was based on (fix/lpc-flash-registration) got merged, so the correct branch to test is fix/lpc-iap-correctness

curious and frustrating that the underlying fix branch is broken some more again - we'll look into it.

dragonmux, 2023-04-07:
Those new targets have been added. The extra handling around the Flash enter/exit mode callbacks is taken care of. Please give the fix/lpc-iap-correctness branch a test and let us know how you get on.

Squantor, 2023-04-07:
after doing a git fetch --all --prune I am getting a build error:
```
target/lpc40xx.c:61:16: error: redefinition of 'struct iap_result'
   61 | typedef struct iap_result {
      |                ^~~~~~~~~~
In file included from target/lpc40xx.c:25:
target/lpc_common.h:71:16: note: originally defined here
   71 | typedef struct iap_result {
      |                ^~~~~~~~~~
target/lpc40xx.c:64:3: error: conflicting types for 'iap_result_s'
   64 | } iap_result_s;
      |   ^~~~~~~~~~~~
In file included from target/lpc40xx.c:25:
target/lpc_common.h:74:3: note: previous declaration of 'iap_result_s' was here
   74 | } iap_result_s;
      |   ^~~~~~~~~~~~
make[1]: *** [Makefile:128: lpc40xx.o] Error 1
```

dragonmux, 2023-04-07:
Apologies, that was a rebasing mistake which is now fixed. We did have to force-push the branch to fix that correctly

Squantor, 2023-04-07:
flashing LPC824M201JHI33 with just a BMP running ```fix/lpc-iap-correctness``` works!
flashing LPC824M201JHI33 with BMD and BMP running ```fix/lpc-iap-correctness``` does not work: 
```
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x M0+
0xfffffffe in ?? ()
./gdb_scripts/LPC824M201HI33_blinky_debug.txt:9: Error in sourced command file:
Error erasing flash with vFlashErase packet
```
Log: [LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/11179590/LPC824-FWfix-BMDAfix.log)

dragonmux, 2023-04-07:
Turns out that was being caused by lpc_iap_call() over-eagerly trapping faults as errors - the faulting IAP call was stopping on our inserted breakpoint instruction. That's now fixed, so please give it another try (`git pull` should be sufficient this time).

Squantor, 2023-04-07:
Did the pull, results:
flashing LPC824M201JHI33 with BMD and BMP running fix/lpc-iap-correctness does not work, flashing seems to work, stepping also but when I do a continue the program does not run and CTRL-C does not work in GDB:
```
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      LPC82x M0+
0x000001ec in main () at projects/LPC824M201HI33_blinky/src/main.cpp:25
25          var = var ^ 0x55;
Loading section .text, size 0x73c lma 0x0
Start address 0x000000c0, load size 1852
Transfer rate: 1 KB/sec, 926 bytes/write.
Section .text, range 0x0 -- 0x73c: matched.
(gdb) s
53        src = &_flash_data;
(gdb) s
54        dst = &_data_start;
(gdb) c
Continuing.
^C^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
Disconnected from target.
(gdb) 
```
Log:
[LPC824-FWfix-BMDAfix.log](https://github.com/blackmagic-debug/blackmagic/files/11180033/LPC824-FWfix-BMDAfix.log)

PR #1433 (2023-04-19): Fix: LPC IAP correctness

## Detailed description

This PR addresses the breakage that occurred between v1.8.2 and v1.9 in the IAP routines used by the LPC8xx parts, among others. While investigating this it turned out that the `lpc_iap_call` was invoking undefined behaviour and was quite poorly documented.

We've addressed this with a near complete rewrite of the routines affected and adding handling for the LPC8xx parts support to handle their needs around `target_reset()` and entering/exiting Flash mode

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https:// [... cut]
closed 2023-04-19 by esden