blackmagic-debug/blackmagic #1235: [bug] USB-enumeraton fails after initial disconnect for Black Magc Probe (FW 1.8.2)

Nightwalker-87, 2022-09-03:
Hardware: Black Magic Probe Mini V2.1 (BMP21)
Firmware: v1.8.2
Operating System: Debian GNU/Linux Bookworm
Kernel: 5.18.0-4-amd64 (x86_64)
C-Library: GNU C Library / (Debian GLIBC 2.34-7) 2.34

The current precompiled v1.8.2 binary (.bin) seems to have introduced a bug regarding USB enumeration at least for BMP21 devices.

Following the official documentation

 `sudo dfu-util -d 1d50:6018,:6017 -s 0x08002000:leave -D blackmagic-native-v1.8.x.bin`

results in the following outputs:

Previous Firmware (v1.8.1):

```
[10279.355611] usb 1-4.2: USB disconnect, device number 37
[10279.580796] usb 1-4.2: new full-speed USB device number 38 using xhci_hcd
[10279.683382] usb 1-4.2: New USB device found, idVendor=1d50, idProduct=6018, bcdDevice= 1.00
[10279.683396] usb 1-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10279.683402] usb 1-4.2: Product: Black Magic Probe  v1.8.2
[10279.683407] usb 1-4.2: Manufacturer: Black Magic Debug
[10279.683410] usb 1-4.2: SerialNumber: BFE7D7EE
[10279.701890] cdc_acm 1-4.2:1.0: ttyACM0: USB ACM device
[10279.702869] cdc_acm 1-4.2:1.2: ttyACM1: USB ACM device
```

This is reproducible upon connect-disconnect-connect routines. Flashing this previous version restores the expected behaviour.

Current Firmware (v1.8.2):

```
[10290.212589] usb 1-4.1: new full-speed USB device number 39 using xhci_hcd
[10290.292601] usb 1-4.1: device descriptor read/64, error -32
[10290.484518] usb 1-4.1: device descriptor read/64, error -32
[10290.668573] usb 1-4.1: new full-speed USB device number 40 using xhci_hcd
[10290.748606] usb 1-4.1: device descriptor read/64, error -32
[10290.936588] usb 1-4.1: device descriptor read/64, error -32
[10291.044691] usb 1-4-port1: attempt power cycle
[10291.648490] usb 1-4.1: new full-speed USB device number 41 using xhci_hcd
[10291.648564] usb 1-4.1: Device not responding to setup address.
[10291.856695] usb 1-4.1: Device not responding to setup address.
[10292.064555] usb 1-4.1: device not accepting address 41, error -71
[10292.145466] usb 1-4.1: new full-speed USB device number 42 using xhci_hcd
[10292.145537] usb 1-4.1: Device not responding to setup address.
[10292.352706] usb 1-4.1: Device not responding to setup address.
[10292.560542] usb 1-4.1: device not accepting address 42, error -71
[10292.560669] usb 1-4-port1: unable to enumerate USB device
[10292.640549] usb 1-4.2: new full-speed USB device number 43 using xhci_hcd
[10292.743063] usb 1-4.2: New USB device found, idVendor=1d50, idProduct=6018, bcdDevice= 1.00
[10292.743076] usb 1-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10292.743082] usb 1-4.2: Product: Black Magic Probe  v1.8.2
[10292.743086] usb 1-4.2: Manufacturer: Black Magic Debug
[10292.743089] usb 1-4.2: SerialNumber: BFE7D7EE
[10292.761012] cdc_acm 1-4.2:1.0: ttyACM0: USB ACM device
[10292.762083] cdc_acm 1-4.2:1.2: ttyACM1: USB ACM device
[10292.840525] usb 1-4.4: new full-speed USB device number 44 using xhci_hcd
[10292.920566] usb 1-4.4: device descriptor read/64, error -32
[10293.108578] usb 1-4.4: device descriptor read/64, error -32
[10293.300536] usb 1-4.4: new full-speed USB device number 45 using xhci_hcd
[10293.380541] usb 1-4.4: device descriptor read/64, error -32
[10293.572557] usb 1-4.4: device descriptor read/64, error -32
[10293.680689] usb 1-4-port4: attempt power cycle
[10294.284518] usb 1-4.4: new full-speed USB device number 46 using xhci_hcd
[10294.284677] usb 1-4.4: Device not responding to setup address.
[10294.492659] usb 1-4.4: Device not responding to setup address.
[10294.700503] usb 1-4.4: device not accepting address 46, error -71
[10294.780526] usb 1-4.4: new full-speed USB device number 47 using xhci_hcd
[10294.780674] usb 1-4.4: Device not responding to setup address.
[10294.988653] usb 1-4.4: Device not responding to setup address.
[10295.196500] usb 1-4.4: device not accepting address 47, error -71
[10295.196640] usb 1-4-port4: unable to enumerate USB device
```

It would be great, if someone on the development side could dig into this in order to resolve this error / fix this bug ASAP.
If necessary additional testing may take place on the current system on which the results were obtained.

Thank You.

Nightwalker-87, 2022-09-03:
I have found the cause for the described behaviour. It is not related to the firmware itself but to used USB hardware.
USB device enumeration may fail as soon as the BMP is connected to a USB hub and not directly to a host USB port.
In the current setup the following bus-powered USB hub was used:

_ANKER 4-Port Ultra Slim USB 3.0 Data Hub (Model: A7516)_

I can't give any further details on the related hardware issue yet, but may leave a note it I come across anything useful.
For now I hope this may help others along. :wink:

closed 2022-09-03 by Nightwalker-87

esden, 2022-09-03:
Thank you @Nightwalker-87 for following up on this. I use an Anker 10port hub (Model: A7515) for all the BMP production programming and testing without any noticeable problems. But in general USB and hubs can introduce hard to track down issues. If you figure something out let us know.

tkornack, 2022-12-01:
I also hit a USB enumeration issue like this on my CalDigit Thunderbolt USB hub. It was resolved with lots of unplugging and replugging. Oddly, the BMP is the only USB-C device I have that exhibits this particular issue. Why might that be?