blackmagic-debug/blackmagic #1238: Black Magic Probe(STLINK/V2) freezes Raspberry Pi Pico(rp2040) when GDB is attached

turtiustrek, 2022-09-11:
When any of the sleep functions are called from the RPI sdk and the debugger is attached, the program fails to continue execution. 
Here is the GDB log for the following code:
```
#include <stdio.h>
#include "pico/stdlib.h"

int main()
{
    stdio_init_all();
    gpio_init(25);
    gpio_set_dir(25, true);
    while (true)
    {

        printf("LED on\n");
        gpio_put(25, 1);
        busy_wait_ms(1);
        printf("LED off\n");
        gpio_put(25, 0);
        busy_wait_ms(1);
    }
    return 0;
}
```

```
Reading symbols from ./hello_world.elf...
(gdb) tar ext /dev/ttyACM3
Remote debugging using /dev/ttyACM3
(gdb) mon s
Target voltage: 3.16V
Available Targets:
No. Att Driver
 1      Raspberry RP2040 M0+
 2      Raspberry RP2040 M0+
 3      Raspberry RP2040 Rescue (Attach to reset!) 
(gdb) att 1
Attaching to program: /home/turtius/Programming/Hardware/test/build/hello_world.elf, Remote target
0x0000018a in ?? ()
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x0000018a in ?? ()
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/turtius/Programming/Hardware/test/build/hello_world.elf 
^C
Program received signal SIGINT, Interrupt.
0x10000c18 in busy_wait_ms ()
(gdb) c
Continuing.

^C
Program received signal SIGINT, Interrupt.
0x10000c12 in busy_wait_ms ()
```
Even continuing execution, it seems to get stuck in ```busy_wait_ms```  regardless of the value passed to it.
The program works normally not debugging.
```
(gdb) mon v
Black Magic Probe(STLINK/V2) v1.8.0-639-g3e8f296d, Hardware Version 0
Copyright (C) 2022 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) 
```

jamesturton, 2022-09-12:
Hi @turtiustrek! This seems to be a duplicate of #1228 which @dragonmux and I are working on a fix for in #1229, so hopefully this will be resolved soon.

dragonmux, 2022-09-12:
We'll link this issue to the mentioned PR then so that merging that will automatically close this

closed 2022-09-19 by dragonmux