blackmagic-debug/blackmagic #1243: Support for Renesas RA2A1 ?

mtk11, 2022-09-19:
Trying to use it with Renesas RA2A1 and getting the following:

```
WARNING: "Debug component" !match expected "Generic IP component"
Found renesas chip (R7FA2A1AB3CNF   ) with Flash Root Table and unsupported Part ID 0x1b0 please report it
0x80000000: 0x00000004000a3004 Non ARM component ignored
```

Support for this series ?

dragonmux, 2022-09-19:
There is experimental support for Renesas parts thanks to @perigoso's #1133, we'll take their lead on how to deal with that. v1.9 is currently in a feature freeze pending release of v1.9rc1, so this will get targeted to v1.10.

mtk11, 2022-09-19:
> does it not show up as an attachable target?

```
(gdb) mon s
Available Targets:
No. Att Driver
 1      R7FA2A1AB3CNF    M23
(gdb) att 1
Attaching to program: /tmp/ra2a1-main/blinky/main.hex, Remote target
0xeffffffe in ?? ()
(gdb) load main.hex
Loading section .sec1, size 0xc0 lma 0x0
Load failed
```

dragonmux, 2023-04-11:
We have re-opened the issue and will modify the tittle accordingly.

PR #1244 (2022-09-19): target/renesas: add known id for RA2A1

Adds id for RA2A1, from #1243 

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/blackmagic-debug/blackmagic/blob/main/CONTRIBUTING.md) to this repository
* [x] It builds for hardware native (`make PROBE_HOST=native`)
* [x] It builds as BMDA (`make PROBE_HOST=hosted`)
* [ ] I've tested it to the best of my ability (no hardware, 3rd party testing necessary)
* [x] My commit messages  [... cut]