blackmagic-debug/blackmagic #1248: RTT for F072 (STMF072) platform not working

lucasgcbkhomp, 2022-09-23:
I have a black magic debugger with STM32F072 on a custom board. RTT logging doesn't seem to work. It works for erasing and writing just fine currently.  Tried it with both v1.8.1 and the 9f9d42861dd2772b25557fb13cbae7e56c09d20c commit in main.

I've followed the instructions in UsingRTT.md with minicom/picocom with gdb; I can attach and enable rtt but as soon as I run it my debugger goes silent and the target will not respond to interrupt requests. The same commands for a STM32 "bluepill" board seems to work just fine.

Has anyone managed to get RTT logging to work for STM32F072? Is it a missing feature or perhaps just a problem on my end? Here's my output; it takes a long time for the F072 to work again.

```bash
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
(gdb) target external-remote /dev/ttyACM0
Undefined target command: "external-remote /dev/ttyACM0".  Try "help target".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor swdp_scan
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0003014e in ?? ()
(gdb) mon rtt enable
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program:  
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
^C^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
Disconnected from target.
(gdb) mon rtt status
"monitor" command not supported by this target.
```

Thanks

koendv, 2022-10-13:
OK. Connect to target, and read the target memory map.

```
 $ arm-none-eabi-gdb 
(gdb) tar ext /dev/ttyBmpGdb
(gdb) mon swd
(gdb) att 1
(gdb) info mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs 
0   y  	0x08000000 0x08020000 flash blocksize 0x400 nocache 
1   y  	0x20000000 0x20005000 rw nocache 
```
gdb shows two memory regions: flash and ram. ram is tagged "rw". gdb's idea of available ram has to correspond exactly to physical ram on your target. I assume this is not the case, and gdb thinks there is more ram than physically present. Correct?

As a quick fix, in the bmp source, go to the source for your target, and hardcode ram size. Go to src/target/nrf51.c. Look for 
target_add_ram() for your target, nrf52, and hardcode ram address and size, e.g.  `target_add_ram(t, 0x20000000, 24 * 1024);` for an nrf52 with 24k ram. Update bmp, check target memory map, try rtt.

lucasgcbkhomp, 2022-10-14:
> OK. Connect to target, and read the target memory map.
> [...]
> gdb shows two memory regions: flash and ram. ram is tagged "rw". gdb's idea of available ram has to correspond exactly to physical ram on your target. I assume this is not the case, and gdb thinks there is more ram than physically present. Correct?
> 
> I have attached a patch to bmp that allows you to limit the address range scanned by rtt. [ram.txt patch](https://github.com/blackmagic-debug/blackmagic/files/9771678/ram.txt)
> [...]
> ```

Thanks for the attention.

I've rebased my repository to the upstream BMP main for this task and applied the patch you've attached with no issues.

Running the commands as instructed, I get 0x40000 bytes for the target (in this case a nRF52840).

I limited the ram as you've specified, and provided it with the ram span of 0x20003e80 for the F072. While it doesn't get stuck anymore after running `mon rtt`, it does not seem to find the RTT block.

Running it with the full nRF52840 (the default) size or above 0x20003e80 simply freezes the BMP which requires me to power cycle it.

```bash
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x000229f2 in ?? ()
(gdb) mon swdp_scan
Target voltage: ABSENT!
You are now detached from the previous target.
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) att 1
A program is being debugged already.  Kill it? (y or n) y
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0002334c in ?? ()
0x0002334c in ?? ()
(gdb) info mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs 
0   y   0x00000000 0x00100000 flash blocksize 0x1000 nocache 
1   y   0x10001000 0x10002000 flash blocksize 0x1000 nocache 
2   y   0x20000000 0x20040000 rw nocache 
(gdb)  mon rtt ram 0x20000000 0x20003E80
(gdb) mon rtt status
rtt: on found: no ident: "off" halt: off channels: auto ram: 0x20000000 0x20003e80
max poll ms: 256 min poll ms: 8 max errs: 10
(gdb) mon rtt 
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program:  
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
^C
Program received signal SIGINT, Interrupt.
0x000229f2 in ?? ()
(gdb) mon rtt status
rtt: on found: no ident: "off" halt: off channels: auto ram: 0x20000000 0x20003e80
max poll ms: 256 min poll ms: 8 max errs: 10
```

## An additional strange thing:
- Running minicom as per de UsingRTT.md file doesn't allow one to target extended. If I put minicom after the target extended, it slows down a ton to the point and barfs packets. I'm not sure if this is expected?
  ```
  (gdb) tar ext /dev/ttyACM0
  Remote debugging using /dev/ttyACM0
  Remote connection closed
  (gdb) tar ext /dev/ttyACM0
  Remote debugging using /dev/ttyACM0
  Remote connection closed
  (gdb) tar ext /dev/ttyACM0
  Remote debugging using /dev/ttyACM0
  warning: No executable has been specified and target does not support
  determining executable automatically.  Try using the "file" command.
  0x000229f2 in ?? ()
  (gdb) mon swdp_scan
  Target voltage: ABSENT!
  Available Targets:
  No. Att Driver
   1      Nordic nRF52 M4
   2      Nordic nRF52 Access Port 
  (gdb) att 1
  A program is being debugged already.  Kill it? (y or n) y
  Attaching to Remote target
  Ignoring packet error, continuing...
  Ignoring packet error, continuing...
  Remote 'g' packet reply is of odd length: 03000000f83b00204c210020883a0020c11600003603000008b1012001000000143400200100000001000000bc2c002068440020d8b00120f1ffffff4c3302000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  Ignoring packet error, continuing...
  Remote 'g' packet reply is of odd length: 03000000f83b00204c210020883a0020c11600003603000008b1012001000000143400200100000001000000bc2c002068440020d8b00120f1ffffff4c330200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003000061d8b0012090a901200100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  (gdb) Ignoring packet error, continuing...
  Remote connection closed
  (gdb) Quit
  ```

dragonmux, 2022-10-14:
The problem with the RTT code isn't the RAM block size stuff - at least, not directly.. it's the fact the RTT code isn't calling `target_check_error()` or using read timeouts so never finds out that the reads it's trying are off the end of RAM (or indeed, just providing accurate maps in the first place.. but this is a bit complicated for nRF51/52). We will be addressing this once we've completed the clang-format effort and it is on our radar (if you will).

Regarding the packet problems, we notice you're using a very old GDB (8.3) - it would be interesting to see if the behaviour you're seeing changes with GDB 10.3 from the [gnu-rm](https://developer.arm.com/downloads/-/gnu-rm) suite. We're half wondering if BMP is not fully compatible with GDB 8 any more.

koendv, 2022-10-14:
OK. If the rtt code doesn't check ``target_check_error()`` that needs to be fixed. But looking at rtt.c, it seems the return code of ``target_mem_read()`` is checked, and the return code of ``target_mem_read()`` is ``target_check_error()``, I think.

lucas,

On the rtt side, there are two things I'd like you to try. The first is slowing down rtt to a crawl. Configure rtt for polling between 1 and 4 times per second, and stopping after 10 errors:
```
(gdb) mon rtt poll 1024 256 10
(gdb) mon rtt stat
rtt: on found: no ident: no channels: auto 
max poll ms: 1024 min poll ms: 256 max errs: 10
```
 If that solves the problem, then maybe the default values are too much for the cpu of the F072. 

Also, just to make sure: please take the linker map of your firmware, and grep for RTT somewhere at the beginning of ram. A block with size a multiple of 24 decimal. Size depends upon number of channels configured. 
```
grep 2000 HelloWorld.ino.map | grep RTT
```
That is the rtt control block. The rtt control block ought to exist, be within the output of ``(gdb)info mem``, and if ``mon rtt ram`` has been specified, also within the address range of ``mon rtt ram``; else rtt will never find the control block.

dragonmux, 2022-10-14:
Ah, fair point re `target_mem_read()` - looks like we shot our mouth about the wrong call.. it's the target halt processing that does The Bad.. along with a couple of other things that need addressing. The big one that in `poll_rtt()`, `target_halt_poll()` is called in a while loop with no checking if the target's even still attached, which in Lucas' instance he's detaching from BMP which will necessarily detach the target, meaning that loop might never end. There are a couple of other unsound things the code is doing. Apologies that we picked on the wrong call

koendv, 2022-10-15:
> The big one that in `poll_rtt()`, `target_halt_poll()` is called in a while loop with no checking if the target's even still attached

OK. I assume this is line 407; /* briefly halt target during target memory access */ ?
This is for risc-v. 

With rtt the user program writes output to a circular buffer, and the debugger reads output from the circular buffer and prints it. 
The is attractive because, on arm processors, the debugger can read and write target memory without stopping the processor.
The arm spec requires the debugger to be able to access memory while the processor is running.

On risc-v this is not the case.
The riscv spec has no requirement for debugger to be able to read and write target memory while the processor is running.
This saves some gates, and makes the hardware a bit cheaper.
I did some tests with the riscv branch of bmp, and found the debugger had to stop the processor before accessing memory.
So there's a check if the cpu is riscv "RVDBG" in target_no_background_memory_access().
If the cpu is riscv the cpu is halted before rtt accesses memory and resumed after. On arm this code is not executed, I think.
Still: you're right, the target_halt_poll() code needs to check if the target is gone.

dragonmux, 2022-10-16:
Yeah, that approach could work and eliminates some of the other potential pitfalls that code had.. might be interesting to see if we can pull some of those concepts out into their own `static inline` functions to make it easier to read and reason about but that does at least solve for the main problems we saw. Checking over the rtt_if implementations for soundness is a second factor that needs checking here and not something we've done yet.

If you could (as it's your code originally!) do this and open a PR with this patch and anything else you spot that the code's doing that's not completely sound, that would give us better confidence that lucas' problems entirely stem from targets defining memory regions too loosely and them not having a findable RTT header block in RAM.

Our few other bits of feedback that could then be addressed are that:

* We're not hurting for characters in identifiers, so `fastsrch` and `memsrch` should please be named using full words and not letter-dropped shortenings (+ the differences between them clarified by their naming)
* static adding to the constants at the top of fastsrch to reduce overhead and memory usage
* both `target_aligned_mem_read` and `target_no_background_memory_access` either moved into the target API implementation, or not named with `target_` in their names as presently this is unsound to do
* `target_no_background_memory_access` might be better named as `target_mem_access_needs_halt` and either the value constructed directly returned, or if still assigning to a variable, the value made const.

These are items of clean-up we had in mind to do anyway but if you address them in a PR in a way that makes the most sense for your contribution then it'll give us more confidence we've not subtly broken some part of the RTT support.

koendv, 2022-10-16:
OK. I agree there is a need for a PR, but I'd like to first solve this issue.

If you are using Segger source in your target, then the control block is called _SEGGER_RTT.
You ought to be able to find this symbol in the linker map.

If you are using rust rtt-target, the same symbol name is used.

Try beginning with a small "Hello, world!" program.

In your source, try explicitly calling SEGGER_RTT_Init() early in your code to initialize the control block. This ought to result in bmp finding the control block. "mon rtt stat"

On the bmp side, "mon rtt cblock" displays the control block (if found).
This is a cached copy of the control block in the target. You could fool bmp by re-allocating the buffers, changing buffer address etc.

koendv, 2022-10-24:
Lucas,

You are using Arduino, correct? Please find enclosed compiled binaries for nRF52840, 256kb ram.
[ArduinoSketchbook.zip](https://github.com/blackmagic-debug/blackmagic/files/9850965/ArduinoSketchbook.zip)
Begin with the sketch called "minimal". 

Looking at the map file, I see following addresses for the control block, depending upon sketch:
0x20007028, 0x20007014, 0x20006ffc
so perhaps use
```
mon rtt ram 0x20006000 0x20008000
```
but you need to check whether this is ok for your specific nRF52 device.

I also add a typescript of a similar session with a blue pill as target:
[typescript.txt](https://github.com/blackmagic-debug/blackmagic/files/9851044/typescript.txt)
Specifically, look for the following:
- ``mon rtt status`` gives ``rtt: on found: yes``
- ``mon rtt cblock`` gives ``cbaddr: 0x2000XXXX``, where 0x2000XXXX is the same address as found in the map file. 

Thank you.

koen

lucasgcbkhomp, 2022-10-24:
> Lucas,
> 
> You are using Arduino, correct? Please find enclosed compiled binaries for nRF52840, 256kb ram. [ArduinoSketchbook.zip](https://github.com/blackmagic-debug/blackmagic/files/9851184/ArduinoSketchbook.zip) Begin with the sketch called "minimal".
> 
> Looking at the map file, I see following addresses for the control block, depending upon sketch: 0x20007028, 0x20007014, 0x20006ffc so perhaps use
> 
> ```
> mon rtt ram 0x20006000 0x20008000
> ```
> 
> but you need to check whether this is ok for your specific nRF52 device.
> 
> I also add a typescript of a similar session with a blue pill as target: [typescript.txt](https://github.com/blackmagic-debug/blackmagic/files/9851044/typescript.txt) Specifically, look for the following: If you interrupt the program with Ctrl-C,
> 
> * `mon rtt status` gives `rtt: on found: yes`
> * `mon rtt cblock` gives `cbaddr: 0x2000XXXX`, where 0x2000XXXX is the same address as found in the map file.
> 
> Thank you.
> 
> koen

Hello koen, thank you also for your patience. 

I am not using Arduino. I'm using Zephyr 2.7.3 as of now. 

Both the BMP and my nRF52840 are my custom layouts.

I've compiled a tiny Zephyr Project with RTT enabled and tested it with my J-Link. So that works.
The block is very early on in RAM, so I set the range to `0x20000100 0x20001000`. No cigar, the BMP still freezes.

It would be nice if someone else had this board so they could verify RTT is working and I'm not messing something up, which is a possibility. I've had some strange behavior on `main` where writing to flash with GDB 10 leaves a process running which freezes my BMP when killed.

### grep Zephyr map:

```
$ grep 2000 zephyr.map | grep RTT
                0x00000000200000f0       0x14 modules/segger/libmodules__segger.a(SEGGER_RTT_zephyr.c.obj)
                0x0000000020000934       0xa8 modules/segger/libmodules__segger.a(SEGGER_RTT.c.obj)
                0x0000000020000934                _SEGGER_RTT
                0x0000000020000a41       0x10 modules/segger/libmodules__segger.a(SEGGER_RTT.c.obj)
                0x0000000020000a51      0x400 modules/segger/libmodules__segger.a(SEGGER_RTT.c.obj)
```

### info mem on the nRF52840 through the STM32F072 BMP:
```
(gdb) info mem
Using memory regions provided by the target.
Num Enb Low Addr   High Addr  Attrs 
0   y   0x00000000 0x00100000 flash blocksize 0x1000 nocache 
1   y   0x10001000 0x10002000 flash blocksize 0x1000 nocache 
2   y   0x20000000 0x20040000 rw nocache 
```

The commands after I've flashed the program:
```bash
(gdb) tar ext /dev/ttyACM5
Remote debugging using /dev/ttyACM5
(gdb) mon swdp_scan
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.

Program stopped.
0x0000df22 in ?? ()
(gdb) mon rtt ram 0x20000100 0x20001000
(gdb) mon rtt stat
rtt: off found: no ident: "off" halt: off channels: auto ram: 0x20000100 0x20001000
max poll ms: 256 min poll ms: 8 max errs: 10
(gdb)  mon rtt poll 1024 256 10
(gdb) mon rtt stat
rtt: off found: no ident: "off" halt: off channels: auto ram: 0x20000100 0x20001000
max poll ms: 1024 min poll ms: 256 max errs: 10
(gdb) mon rtt enable
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program:  
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
^C^CThe target is not responding to interrupt requests.
```

dragonmux, 2022-10-24:
`ident: "off"` looks wrong - that suggests a bug in the RTT monitor command handler, possibly something we introduced. We'll review that code and see if we can find anything obvious tomorrow.

koendv, 2022-10-25:
Lucas,

Does bmp also hang if you wait until the program has initialized to switch on rtt? Like this:
```
 $ /opt/xpack-arm-none-eabi-gcc-10.3.1-2.3/bin/arm-none-eabi-gdb -q
(gdb) tar ext /dev/ttyBmpGdb
(gdb) mon swd
(gdb) att 1
(gdb) mon rtt disable
(gdb) r
```
Once the program is running, wait a few seconds, then ctrl-c, switch on rtt and continue:
```
^C
Program received signal SIGINT, Interrupt.
0x08000326 in ?? ()
(gdb) mon rtt enable
(gdb) mon rtt ram 0x20000100 0x20001000
(gdb) continue
Continuing.
```

koendv, 2022-10-27:
I've opened a PR for this; https://github.com/blackmagic-debug/blackmagic/pull/1276

lucasgcbkhomp, 2022-10-27:
> OK. This is becoming interesting. `mon rtt poll 1024 256 10` does not seem to be doing anything, we can forget that.
> 
> Please send me your Zephyr sources. Can you try unzipping ArduinoSketchbook.zip, and loading the file minimal/build/adafruit.nrf52.pca10056/minimal.ino.elf ? Can you set a breakpoint after SEGGER_RTT_Init() and, in gdb, dump _SEGGER_RTT and the next 7*24=168 bytes?
> 
> A small checklist. To keep things simple:
> 
> * call SEGGER_RTT_Init() early on in the program
> * do not use flags such as SEGGER_RTT_MODE_BLOCK_IF_FIFO_FULL
> * do not change buffers using SEGGER_RTT_AllocUpBuffer or SEGGER_RTT_ConfigUpBuffer
> * do not change defaults in SEGGER_RTT_Conf.h
> * do not define RTT_USE_ASM.
> 
> If you had a STM32F103, you could set up two experiments: once with the F103 as bmp, the nRF52 as target, and once with the F072 as bmp, the F103 as target, and see if the error persists.
> 
> Dragonmux, I'll take a look at ident: "off" tomorrow and clean up.
> 
> I'm calling it a day. Good night.
> 
> koen

Hello, koen

STM32F102 works although it hangs sometimes. I could not get the minimal arduino example to do what you've requested (it segfaults when loading and running). You can get the Zephyr sources I used here: 
[hello_world_zephyr.zip](https://github.com/blackmagic-debug/blackmagic/files/9880291/hello_world_zephyr.zip)

> Lucas,
> 
> Does bmp also hang if you wait until the program has initialized to switch rtt on? Like this:
> 
> ```
>  $ /opt/xpack-arm-none-eabi-gcc-10.3.1-2.3/bin/arm-none-eabi-gdb -q
> (gdb) tar ext /dev/ttyBmpGdb
> (gdb) mon swd
> (gdb) att 1
> (gdb) mon rtt disable
> (gdb) r
> ```
> 
> Once the program is running, wait a few seconds, then ctrl-c, switch on rtt and continue:
> 
> ```
> ^C
> Program received signal SIGINT, Interrupt.
> 0x08000326 in ?? ()
> (gdb) mon rtt enable
> (gdb) mon rtt ram 0x20000100 0x20001000
> (gdb) continue
> Continuing.
> ```

I've attempted this with the Zephyr hello world, unfortunately it still freezes.

```
gdb) tar ext /dev/ttyBmpGdbB206134542038
Remote debugging using /dev/ttyBmpGdbB206134542038
(gdb) mon swd
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0000df22 in ?? ()
(gdb) mon rtt disable
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program:  
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
^C
Program received signal SIGINT, Interrupt.
0x0000df22 in ?? ()
(gdb) mon rtt enable 
(gdb) mon rtt ram 0x20000100 0x20001000
(gdb) continue
Continuing.
^C^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
```

I've attempted the same commands after merging #1276 locally and unfortunately little results ensued.

```
(gdb) tar ext /dev/ttyBmpGdbB206134542038
Remote debugging using /dev/ttyBmpGdbB206134542038
(gdb) mon swdp_scan
Target voltage: ABSENT!
Available Targets:
No. Att Driver
 1      Nordic nRF52 M4
 2      Nordic nRF52 Access Port 
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.

Program stopped.
0x0000df22 in ?? ()
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
(gdb) mon rtt disable
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program:  
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
^C
Program received signal SIGINT, Interrupt.
0x0000df22 in ?? ()
(gdb) mon rtt enable
(gdb) mon rtt ram 0x20000100 0x20001000
(gdb) mon rtt status
rtt: on found: no ident: off halt: off channels: auto ram: 0x20000100 0x20001000
max poll ms: 256 min poll ms: 8 max errs: 10
(gdb) continue
Continuing.
^C^CThe target is not responding to interrupt requests.
Stop debugging it? (y or n) y
Disconnected from target.
```

Again, I'm not discarding the possibility of it being a problem on my end because I have a somewhat abnormal setup. We could benefit from another tester to validate these responses.

koendv, 2022-10-27:
It's probably unnecessary to ask, but the logic voltages are both 3.3V?

lucasgcbkhomp, 2022-10-27:
> It's probably unnecessary to ask, but the logic voltages are both 3.3V?

On the SWD, yes. My VCC on it is 3v3.

koendv, 2022-10-27:
OK. If you want, you can do printf debugging like this:
```
diff --git a/src/rtt.c b/src/rtt.c
index 984a569..ecfb854 100644
--- a/src/rtt.c
+++ b/src/rtt.c
@@ -421,6 +421,7 @@ void poll_rtt(target *cur_target)
                }
                /* do rtt i/o if control block found */
                if (rtt_found && rtt_cbaddr) {
+                       gdb_out("here!\r\n");
                        /* copy control block from target */
                        uint32_t rtt_cblock_size = (sizeof rtt_channel[0]) * (rtt_num_up_chan + rtt_num_down_chan);
                        if (target_mem_read(cur_target, rtt_channel, rtt_cbaddr + 24, rtt_cblock_size)) {
```
The output goes to gdb. There's also gdb_outf() if you want printf-like output.
If the messages come too fast and flood the terminal, use "mon rtt poll" to lower the frequency.
Each time, try halving the number of lines where the error could occur.

koendv, 2022-10-30:
nrf52 datasheet: "Access port protection blocks the debugger from read and write access to all CPU registers and memory-
mapped addresses. See the UICR register APPROTECT"

When connected using bmp, are you able you read and write memory from the gdb command prompt? 
Example:

```
(gdb) mon rtt cb
cbaddr: 0x2000071c
...
(gdb) x/16c 0x2000071c
```

dragonmux, 2022-11-01:
This issue should now be solved by the RTT patch that's just been merged, however if it does not and Koen's suggestion about protected mode on the nRF target doesn't help, please re-open this issue.

koendv, 2022-11-06:
rtt can be unresponsive when using wfi and sleep modes that switch the swd port off.

koendv-test, 2022-12-18:
I'd like to re-visit this issue. 
Please try the fix/cortexm-swd-attach-regression branch and see if this fixes the issue. E.g.:
``
git clone -b fix/cortexm-swd-attach-regression https://github.com/blackmagic-debug/blackmagic
cd blackmagic
make PROBE_HOST=f072 ENABLE_RTT=1
``

lucasgcbkhomp, 2022-12-19:
Hell> I'd like to re-visit this issue. Please try the fix/cortexm-swd-attach-regression branch and see if this fixes the issue. E.g.:
> 
> ```
> git clone -b fix/cortexm-swd-attach-regression https://github.com/blackmagic-debug/blackmagic
> cd blackmagic
> make PROBE_HOST=f072 ENABLE_RTT=1
> ```
> 
> Thank you.

Hello, thanks for looking further into this. Any particular steps you'd like me to attempt? Or just the same steps ones we've tried before on an nRF52?

koendv, 2022-12-19:
On Mon, 19 Dec 2022 07:33:46 -0800
lucasgcbkhomp ***@***.***> wrote:

> Hell> I'd like to re-visit this issue. Please try the
> Hell> fix/cortexm-swd-attach-regression branch and see if this fixes
> Hell> the issue. E.g.:
> > 
> > ```
> > git clone -b fix/cortexm-swd-attach-regression
> > https://github.com/blackmagic-debug/blackmagic cd blackmagic
> > make PROBE_HOST=f072 ENABLE_RTT=1
> > ```
> > 
> > Thank you.  
> 
> Hello, thanks for looking further into this. Any particular steps
> you'd like me to attempt? Or just the same steps ones we've tried
> before on an nRF52?
> 

Everything ought to work. 

koen

koendv, 2022-12-19:
On Mon, 19 Dec 2022 16:55:54 +0100
Koen De Vleeschauwer ***@***.***> wrote:

> On Mon, 19 Dec 2022 07:33:46 -0800
> lucasgcbkhomp ***@***.***> wrote:
>
>
> Everything ought to work.
>
> koen

You will have to tell bmp the amount of ram in your nrf, with
mon rtt ram 0x... 0x...
but apart from that, the defaults ought to do just fine.

koen

koendv-test, 2022-12-20:
Lucas,

If bmp hangs, one of the things you can do is run a debugger on the
f072 with bmp:

$ arm-none-eabi-gdb -q
(gdb) tar ext /dev/ttyACM0
(gdb) mon swd
You are now detached from the previous target.
Available Targets:
No. Att Driver
 1      STM32F07 M0
(gdb) at 1
(gdb) file src/blackmagic.elf 
(gdb) c
Continuing.

This way, if there is a trap, you will see it.
And if bmp hangs, you can at least do a 'where'.

koen

giobauermeister, 2023-01-06:
Hello @koendv. I tested one of the most recent commits from main: 98915144944d46cb2dc1daedc734b0134dbde7fb and RTT on F072 seems to be working now. I was helping @lucasgcbkhomp to test this. Will notify him when he gets back to the office next week. Thank you very much!

dragonmux, 2023-01-06:
Are folks happy for this issue to be reclosed then?

koendv-test, 2023-01-07:
On Fri, 06 Jan 2023 07:37:12 -0800
Giovanni Bauermeister ***@***.***> wrote:

> RTT on F072 seems to be working now

Glad to hear.

On Fri, 06 Jan 2023 13:58:09 -0800
Rachel Mant ***@***.***> wrote:

> Are folks happy for this issue to be reclosed then?
> 

Yes. User says it works.  

koen

commit fdc97386 (2022-11-01): rtt: fix #1248, RTT for F072 (STM32F072) platform not working
closed 2023-01-07 by dragonmux