blackmagic-debug/blackmagic #1252: load fails on Bluepill clone

henla464, 2022-09-30:
I am having problem with "load" hanging with my Bluepill; after reading https://github.com/blackmagic-debug/blackmagic/issues/1200 this I thought the latest version would solve it but it didn't. 

I don't know what clone it is (it says STM32... on the chip). Maybe someone can identify it from the logs below?
On v1.8.2 monitor swdp_scan identified it as a CH32F103 I believe. Now on cf2bd66 it says "STM32F1 (clone) medium density M3". Any ideas why it hangs, am I doing something wrong? Is there anything I can test or try?

> (gdb) tar ext /dev/ttyACM0
> Remote debugging using /dev/ttyACM0
> Sending packet: $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+#df...Ack
> Packet received: PacketSize=400;qXfer:memory-map:read+;qXfer:features:read+
> Packet qSupported (supported-packets) is supported
> Sending packet: $vMustReplyEmpty#3a...Ack
> Packet received: 
> Sending packet: $!#21...Ack
> Packet received: OK
> Sending packet: $Hg0#df...Ack
> Packet received: OK
> Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
> Packet received: E01
> Sending packet: $qTStatus#49...Ack
> Packet received: 
> Packet qTStatus (trace-status) is NOT supported
> Sending packet: $?#3f...Ack
> Packet received: W00
> (gdb) mon tpwr enable
> Sending packet: $qRcmd,7470777220656e61626c65#07...Ack
> Packet received: O456e61626c696e672074617267657420706f7765720a
> Enabling target power
> Packet received: OK
> (gdb) mon swdp_scan
> Sending packet: $qRcmd,737764705f7363616e#3c...Ack
> Packet received: O54617267657420766f6c746167653a20332e33560a
> Target voltage: 3.3V
> Packet received: O417661696c61626c6520546172676574733a0a
> Available Targets:
> Packet received: O4e6f2e20417474204472697665720a
> No. Att Driver
> Packet received: O203120202020202053544d333246312028636c6f6e6529206d656469756d2064656e73697479204d330a
>  1      STM32F1 (clone) medium density M3
> Packet received: OK
> (gdb) att 1
> Attaching to program: /home/henla464/Documents/STM32CubeIDE/workspace_1.10.1/BluepillTest/Debug/BluepillTest.elf, Remote target
> Sending packet: $vAttach;1#37...Ack
> Packet received: T05thread:1;
> Packet vAttach (attach) is supported
> Sending packet: $qC#b4...Ack
> Packet received: QC1
> Sending packet: $Hg1#e0...Ack
> Packet received: OK
> Sending packet: $qXfer:features:read:target.xml:0,3fb#46...Ack
> Packet received: m<?xml version="1.0"?><!DOCTYPE target SYSTEM "gdb-target.dtd"><target>  <architecture>arm</architecture> <feature name="org.gnu.gdb.arm.m-profile"><reg name="r0" bitsize="32"/><reg name="r1" bitsize="32"/><reg name="r2" bitsize="32"/><reg name="r3" bitsize="32"/><reg name="r4" bitsize="32"/><reg name="r5" bitsize="32"/><reg name="r6" bitsize="32"/><reg name="r7" bitsize="32"/><reg name="r8" bitsize="32"/><reg name="r9" bitsize="32"/><reg name="r10" bitsize="32"/><reg name="r11" bitsize="32"/><reg name="r12[508 bytes omitted]
> Sending packet: $qXfer:features:read:target.xml:3fb,3fb#11...putpkt: Junk: 
> Ack
> Packet received: me="no"/></feature></target>
> Sending packet: $qXfer:features:read:target.xml:416,3fb#b1...Ack
> Packet received: l
> Sending packet: $g#67...Ack
> Packet received: 20a107000020000020a107004b9006002c000020e554951c421710c9e04f00200d0b3565f2b4c3020201097c4409fa4a11745ecae04f00206f010008e002000800000001e04f0020d060262c00000000
> Sending packet: $qOffsets#4b...Ack
> Packet received: 
> Sending packet: $qSymbol::#5b...Ack
> Packet received: 
> Packet qSymbol (symbol-lookup) is NOT supported
> Sending packet: $qfThreadInfo#bb...Ack
> Packet received: m1
> Sending packet: $qsThreadInfo#c8...Ack
> Packet received: l
> Sending packet: $qXfer:memory-map:read::0,3fb#e5...Ack
> Packet received: m<memory-map><memory type="ram" start="0x20000000" length="0x5000"/><memory type="flash" start="0x08000000" length="0x20000"><property name="blocksize">0x400</property></memory></memory-map>
> Sending packet: $qXfer:memory-map:read::bd,3fb#7b...Ack
> Packet received: l
> Sending packet: $m80002e0,4#5c...Ack
> Packet received: 0133fb60
> 0x080002e0 in delay (Sending packet: $m800016e,4#61...Ack
> Packet received: 01224ff4
> Sending packet: $m20004fc0,40#ec...Ack
> Packet received: 20a107000020000020a10700a68e060011745eca6f010008e0020008000000010200020020a107000c9106204b900600f84f0020f84f0020da405004eb030008
> x=500000) at ../Core/Src/main.c:205
> 205	  for (i=0 ; i < x ; i++)
> (gdb) set mem inaccessible-by-default off
> (gdb) load BluepillTest.elf
> Sending packet: $vFlashErase:08000000,00001400#c7...Ack
> Packet received: EFF
> Error erasing flash with vFlashErase packet

henla464, 2022-09-30:
Discord pointed me to https://github.com/blackmagic-debug/blackmagic/pull/1249 and I tested:
https://github.com/fk0815/blackmagic/tree/main

But still couldn't get it to work...

dragonmux, 2022-10-01:
Sorry to hear you're having trouble. There are two things that would be helpful here to understand the problem better and suggest a way forward.

The first is that, helpful as a remote protocol trace can be, that does not supply the debug information needed in this instance - please do a build of main with `ENABLE_DEBUG=1`. You might need to disable one or two targets such as efm32.c and nrf51.c to make the build fit. Once you have the build on your probe, please then run `mon debug en` after attaching to the probe with `tar ext /dev/ttyACM0` but before doing any of the others steps. Attach to the debug serial port with something like minicom by running `minicom -D /dev/ttyACM1 -8` in another console and provide the output from minicom. This will allow us to see what's going on during each of the steps you're performing.

The second would be confirming what make line you're using to build the firmware for your probe and what source tree you're using and at what revision - this is mostly so we know where you are exactly and so can dive in deeper.

henla464, 2022-10-01:
Thank you!

I did: git clone --recursive https://github.com/blackmagic-debug/blackmagic.git and followed your instructions and got this result in minicom:

> scan_multidrop: false
> DP DPIDR 0x2ba01477 (v1 rev0) designer 0x43b partno 0xba
> RESET_SEQ failed
> AP   0: IDR=24770011 CFG=00000000 BASE=e00ff003 CSW=23000040 (AHB-AP var1 rev2)
> Halt via DHCSR: success 00030003 after 0ms
> ROM: Table BASE=0xe00ff000 SYSMEM=0x00000001, Manufacturer 43b Partno 4c3
> 0 0xe000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x00000004000bb000  DEVTYPE = 0x00 ARCHID = 0x0000)
> -> cortexm_probe
> CPUID 0x412fc231 (M3 var 2 rev 1)
> DBGMCU_IDCODE 0x20036410, DEVID 0x410, REVID 0x2003 
> Detected clone STM32F1
> 1 0xe0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x00000004003bb002  DEVTYPE = 0x00 ARCHID = 0x0000)
> 2 0xe0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x00000004002bb003  DEVTYPE = 0x00 ARCHID = 0x0000)
> 3 0xe0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x00000004003bb001  DEVTYPE = 0x00 ARCHID = 0x0000)
> 4 0xe0040000: Debug component - Cortex-M3 TPIU (Trace Port Interface Unit) (PIDR = 0x00000004003bb923  DEVTYPE = 0x11 ARCHID = 0x0000)
> 5 Entry 0xfff42002 -> Not present
> ROM: Table END
> stm32f1 flash error 0x10
> stm32f1 flash error 0x10
> stm32f1 flash error 0x10
> stm32f1 flash error 0x10
> stm32f1 flash error 0x10
>

dragonmux, 2022-10-06:
Seems like you're not alone in having STM32F1 unhappy.. #1257 seems to be related. Could you please try the patch suggested by that PR (mean00@4c9b7d70) and let us know how you get on with that?

We're going to be applying something similar in the process of cleaning up in stm32f1.c so it'll be useful to know if that makes your issue go away or if there's something further to investigate.

henla464, 2022-10-09:
I have tested today just now with latest main but I get exactly the same as before, same error and the same output in minicom.

dragonmux, 2022-10-10:
We're guessing that means the issue is unchanged by #1263 which addresses the issue we were trigging on the clones and even real devices by over-eagerly testing for errors

dragonmux, 2023-04-21:
There have been a variety of fixes made to the STM32F1 code since this was last touched, could we ask you to check again and let us know if you still have issues on latest main?

We would note too that you must use an official Arm compiler, *not* one from your distribution's package manager (Ubuntu's miscompiles the code for example), and for a Bluepill you will want to use `make PROBE_HOST=stlink BLUEPILL=1`.

dragonmux, 2023-12-31:
Given you haven't responded and we've done a lot of work to fix the STM32F1 Flash code since this was opened, we are going to assume it is fixed. Please let us know if this is wrong and we will re-open the issue.

closed 2023-12-31 by dragonmux