blackmagic-debug/blackmagic #1257: STM32F1xx: Potential flash erase/write regression

mean00, 2022-10-04:
Hi,
I looked into why git/main was not working on my bluepill (v1.8.2 works without problem)
When a flash operation is performed (write/erase) the logic has changed to check success
- Before : Wait till the busy bit goes to zero THEN check for errors
- Now : Wait till the busy bit goes to zero WHILE checking for errors

With the current behaviour, the very first erase fails (it tries to continue erasing anyway btw ?)
Restoring the old behaviour in stm32f1_flash_busy_wait() makes it work again (just have to move the the while 4 lines up)

The gigadevice spec (did not find the relevant part in STM doc) says to wait for BUSY to go to zero before checking for errors.

Should i try to create another broken merge request or is this just noise due to my setup and should be ignored ?
Thank you

dragonmux, 2022-10-04:
Does #1249 fix things for you? If it does, we'll mark this issue as fixed by that and work on getting that PR merged ASAP

mean00, 2022-10-04:
I'll test it tomorrow but i dont think that will work, my debug code was checking the EOP already and it was not set.
For information, that does work for me (tm):
--- a/src/target/stm32f1.c
+++ b/src/target/stm32f1.c
@@ -363,11 +363,11 @@ static bool stm32f1_flash_busy_wait(target *t, uint32_t bank_offset)
        uint32_t sr;
        do {
                sr = target_mem_read32(t, FLASH_SR + bank_offset);
-               if ((sr & SR_ERROR_MASK) || !(sr & SR_EOP) || target_check_error(t)) {
-                       DEBUG_WARN("stm32f1 flash error 0x%" PRIx32 "\n", sr);
-                       return false;
-               }
        } while (sr & FLASH_SR_BSY);
+       if ((sr & SR_ERROR_MASK) || !(sr & SR_EOP) || target_check_error(t)) {
+               DEBUG_WARN("stm32f1 flash error 0x%" PRIx32 "\n", sr);
+               return false;
+       }
 
        return true;
 }

mean00, 2022-10-04:
I'm not saying this is a correct fix, just that the root cause seems to be checking for error_mask while busy bit  is still on

dragonmux, 2022-10-04:
Ok, we'll figure out some combination of the fixes present then.

With SWD scan, does #1220 work for you? Seeing how that's another known bug we've tried to address but which needs people to let us know if the rewritten code works or regresses. Do note that this must be tested as a firmware update, it is not sufficient to use BMDA to check.

mean00, 2022-10-05:
After a quick test,   using a bluepill based BMP to debug a STM32F103 board : 

- PR #1220 + my simple patch : SWDP scan works, erase / write flash works
- PR #1220 +PR #1249 : SWDP scan works, erase / write flash works
- 
So both solutions are making the BMP work again for me on this simple test
Thank you

mean00, 2022-10-05:
(tested with reflashing the BMP, NOT through hosted mode)

mean00, 2022-10-05:
on a side note,  gdb11 seems to work much better, can debug rust now. llgdb still no go.

dragonmux, 2022-10-05:
We'll work on landing fk0815's patch and then addressing the GigaDevice datasheet item so the code works correctly with the clones then. Many thanks for this great feedback and testing :)

mean00, 2022-10-05:
Great !
Just to be clear the problem also happens with the genuine STM32F103 part (that's what i was using)
it's just that i did not find the relevant documentation in the STM32 datasheet, but it was pretty clear in the gigadevice user manual, so it confirmed that the expected behaviour was to wait for the busy bit to clear THEN check for errors.

dragonmux, 2022-10-10:
With the fix from fk0815 merged, and having written a patch based on the information provided here, could you  please give the patch (linked above) a whirl and tell us if it now works correctly for you?

Many thanks

mean00, 2022-10-10:
Hi, 
I did a quick check with git/main on a stm32f103 (using swlink bmp) => all seems to work fine
Thank you

mean00, 2022-10-11:
a bit more quick tests : 
- GD32F3  : Ok
- CH32F1  : Ko, not detected as a CH32 chip, flashing fails

dragonmux, 2022-10-11:
With which combination(s) of PRs is that the case?

mean00, 2022-10-11:
just git/main

dragonmux, 2022-10-11:
It would be helpful please to test #1263 as this was targeted at your original issue, and then likewise please open a new issue with details for the CH32F1 issues - debug information such as that provided in #1252 would be ideal to include in such an issue so we can track, debug and understand what's going on so it can be fixed.

We are glad to know that merging fk0815's fix has fixed a variety of targets for you though.

mean00, 2022-10-11:
git/main + https://github.com/blackmagic-debug/blackmagic/pull/1263/files

STM31F1  : Ok
GD32F303 : Ok
CH32F103 : Ko

(very quick test)

PR #1263 (2022-11-02): Fix: stm32f1 Flash routine functionality fixes

## Detailed description

In this PR we address the issues found in the STM32F1 Flash operation wait code which cause early exits and incomplete operations erroneously. This also includes some documentation on the chosen steps for operation and mode of operation to hopefully stave off future breakage by well intentioned hands.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.co [... cut]
closed 2022-11-02 by esden