blackmagic-debug/blackmagic #1288: Scanning a device with STICKYERR already set fails

zyp, 2022-11-16:
If a device is already in a state with the STICKYERR bit set in the DP, blackmagic doesn't detect and clear this condition before attempting to scan the device and thus fails:

```
$ ./blackmagic -t
Black Magic Debug App v1.8.0-709-g5092cd14
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 1209:3443 31908092A2189006 Orbcode
 Orbtrace
Using bulk transfer
Ver 2.1.0, Cap (0x 3): SWD/JTAG, DAP_SWD_Sequence
Running in Test Mode
Target voltage: (null) Volt
Speed set to  4.0000 MHz for SWD
IN error: -7
dap_write_reg 08 data 00000002:fault
dap_write_reg 08 data 00000000:fault
scan_multidrop: true
No target found
```

dragonmux, 2022-11-16:
We'll look into how or if we can solve that in the ADIv5 layer once the current cleanup is sorted out and merged. Depending on what we find we can do, we'll either aim a fix for v1.9 or, if it works out to be too complicated for a quick fix, v1.10 - but we'd hope v1.9

If, in the mean time, anyone else comes up with a fix, we'd be most happy to accept a pull request for this issue.

zyp, 2022-11-16:
The fix should be as simple as checking the CTRL/STAT register for the STICKYERR bit as soon as the DP is detected, before scanning for any APs, and clearing it if set.

dragonmux, 2022-11-16:
During scan, sure.. but is it possible to end up set in other situations due to things BMP is doing? we were contemplating if part of doing an AP access should check the DP for a sticky error or indeed adding a check and reset at the start of the AP loop so for each new AP we reset it just in case.

We'd like to consider a holistic fix and not just "what's in front of us right now"

zyp, 2022-11-16:
The correct solution for any normal case is to check and clear it as soon as a transfer returns a failure status so that we don't leave the device in this state in the first place (hence why I also filed #1289), but we should still check this when scanning a device, because we can't trust whatever came before us to have left it in a sane state.

dragonmux, 2022-11-19:
We've opened a PR that should fix this issue. Please test the linked PR

PR #1303 (2022-11-19): Fix: adiv5 sticky error handling

## Detailed description

Currently BMP gets "stuck" if trying to evaluate an ARM micro with a DP that has the sticky error bit set and declares it a lost cause. This PR addresses this by adding sticky bit reset sequences just after the DP is discovered, and just after a failed AP scan, which should allow normal scan to proceed on otherwise scuffed targets.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing]( [... cut]
closed 2022-11-19 by esden