blackmagic-debug/blackmagic #1310: Incorrect response on connection & handshake of the virtual UART port

compuphase, 2022-11-26:
Use a serial terminal to reproduce the issue. The steps in connecting to GDBserver via a (virtual) serial port are to 1) open the connection, and 2) set the DTR line (Data Terminal Ready).

This is how version 1.8.2 handles it (and this is the correct output):
- connect to GDBserver port: no output, DSR & CTS set
- toggle DTR on: output = "$OK#9A" (repeated 3 times, because we fail to acknowledge reception)

On the secondary UART, we can also connect and toggle DTR, and nothing is received (which is correct).

However, on version 1.9 (mainline):
- connect to GDBserver port: ouput = [a1 20 00 00 00 00 02 00 03 00 a1 20 00 00 00 00 02 00 03 00 a1 20 00 00 00 00 02 00 03 00]; DSR & CTS *not* set
- toggle DTR on: output = [a1 20 00 00 00 00 02 00 03 00] (DSR & CTS still not set)
- toggle DTR off: same output

- on secondary UART, exactly the same thing happens: a stream of binary data as soon as you open a connection (and exactly the same byte sequence as in the case for the GDBserver port), and a repeat of this binary stream when you toggle DTR.

Hypothesis: The byte sequence looks eerily similar to a SERIAL_STATE notification for the state of the (virtual) control lines, which should go over the control endpoint. It is as if this notification is transmitted on the IN endpoint.

Notes: 

1) GDB is unaffected to this error. When it does not receive the expected "OK" return, it first retries toggling DTR. When that fails too, it carries on and requests the capabilities of the debug probe. There is a good response on that command, so GDB concludes the connection is successful.

2) On version 1.9: if you transmit data to the BMP with DTR off (i.e. not set), serial communication hangs on Windows (on that port). Not an issue with GDB (because it always sets DTR), but not truly an appropriate response either. Version 1.8.2 does not show this behaviour.

dragonmux, 2022-11-26:
We are a bit confused as to what exactly you're trying to convey here (and please don't use the issue title as an integral part of the report - the UI flow makes reading it and the description as a single document very difficult).

As best as we're able to make out, you're trying to say that:

* Connecting to either USB serial port on windows results in the binary data for a USB packet being emitted into your virtual terminal
* GDB handles this fine but this is more by accident than intent
* You are not seeing the expected behaviour from the debug serial port as you're expecting DTR toggling to work
* Everything else about the behaviour between the two, save for DTR, is the same

Please let us know if that's a reasonable summary

compuphase, 2022-11-26:
This behaviour occurs in both Linux and Windows.
Only the last note may be Windows-specific (I have not yet tested this on Linux).

Relating to the snippet:
https://github.com/blackmagic-debug/blackmagic/blob/97b5c115e99e3e4de7a8224680eb059c21155241/src/platforms/common/usb_serial.c#L97-L100

It appears to me that `CDCACM_GDB_ENDPOINT` (on line 98) is the DATA endpoint. My hunch is, it should be `CDCACM_GDB_ENDPOINT + 1`. See:
https://github.com/blackmagic-debug/blackmagic/blob/1de56884bd903325c2958c90501043f18e82f26d/src/platforms/common/usb_descriptors.h#L62-L69

dragonmux, 2022-11-26:
As noted on Discord, please see the fix branch we already created and the edit in our original reply. A PR is waiting once we know how much of the problem reported here is fixed by that branch

compuphase, 2022-11-28:
The fix/usb-cdc-notification-regression patch fixes all points in this issue:
- The garbage output is gone on both ports, both on connection and on toggling DTR.
- Writing to the GDBserver port with DTR off, no longer hangs.

On toggling DTR (on the GDBserver port), there is still no "$OK#9A" handshake sent (as is does with firmware 1.8.2). However, I have been re-reading the RSP chapter in the GDB manual, and I don't see that GDB requires (or expects) this. Thus, my remark on the missing handshake is a non-error.

dragonmux, 2022-11-28:
Many thanks, we'll convert the PR from a draft then and get Esden to review and merge it ASAP

PR #1314 (2022-11-28): fix: USB serial DTR handling and resulting USB packet sent on the wrong endpoint regression

## Detailed description

Fixed a regression caused in #1190 that puts the CDC notification replies out over the wrong endpoint.

The effect of this regression is that BMP erroneously replies to state change requests, like asking for DTR to be set, over the wrong endpoint channel - using the serial data IN endpoint rather than the notification endpoint.

Many thanks to @compuphase for the data in their bug report allowing us to so quickly pinpoint this.

## Your checklist for this pull request

* [x] I' [... cut]
closed 2022-11-28 by esden