blackmagic-debug/blackmagic #1315: Cannot read out TRACESWO on Arduino Due (SAM3X8E)

RavuAlHemio, 2022-11-28:
Greetings,

I am having trouble convincing the official hardware to talk to an Arduino Due (SAM3X8E).

I have recently extended the [SerialWireOutput](/koendv/SerialWireOutput) library for Arduino to include Arduino Due (Microchip/Atmel SAM3X8E, Cortex-M3 core) support; it previously only supported some STM32-based Arduino-like boards.

Initial attempts were promising: I was able to convince OpenOCD and my J-Link EDU (which only supports Async TRACESWO) to transmit to my PC. However, I never managed to get my BMP (official hardware V2.3b, first with the latest official firmware, then with some of the latest firmware revisions built by GitHub Actions) to work; I did reconfigure the project to use Manchester TRACESWO instead, since that's the only TRACESWO protocol supported by the official BMP hardware.

I'm running `GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git` as it ships with the Arduino IDE on Windows; my startup commands are:

```
targ ext COM9
moni freq 100k
moni swdp_scan
attach 1
moni traceswo decode
```

Opening the other serial port provided by the probe, COM12 in my configuration, causes the terminal emulator to stop responding to user input and eventually give up with the error message `The semaphore timeout period has expired.` (apparently Windows error `ERROR_SEM_TIMEOUT` = 121 = 0x79).

Also, if I attempt a connection with COM12 or follow `moni traceswo decode` with `cont` in gdb, the probe stops blinking its LEDs (only the green LED remains lit) and it stops responding to `gdb` commands.

Am I missing something obvious?

Thanks in advance!

dragonmux, 2024-11-09:
👋🏼 We finally got around to looking at the SWO subsystem in the firmware. This issue should have been fixed by #1934 - please let us know how you get on with the latest `main` and if there are any further issues you encounter.

dragonmux, 2025-03-06:
Given we've heard nothing back on this and nobody complaining that SWO is super broken out the box from the release candidates, we're going to close this as fixed.

closed 2025-03-06 by dragonmux