blackmagic-debug/blackmagic #1319: S32K148: Device fails to attach

xobs, 2022-11-29:
With `Black Magic Probev1.9.0-rc0-3-gd5e59b94, Hardware Version 6`, the S32K simply fails to attach. It scans just fine, but attaching does not work anymore.

```
(gdb) mon ver
Black Magic Probev1.9.0-rc0-3-gd5e59b94, Hardware Version 6
Copyright (C) 2022 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) mon s
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      S32K14x M4
 2      Kinetis Recovery (MDM-AP)
(gdb) att 1
Attaching to Remote target
Attaching to Remote target failed
(gdb) mon s
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      S32K14x M4
 2      Kinetis Recovery (MDM-AP)
(gdb) att 1
Attaching to Remote target
Attaching to Remote target failed
(gdb) att 2
Attaching to Remote target
Attaching to Remote target failed
(gdb)
```

Loading an older version fixes things.

```
(gdb) tar ext \\.\COM5
Already connected to a remote target.  Disconnect? (y or n) y
Remote debugging using \\.\COM5
(gdb) mon s
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      S32K14x M4
 2      Kinetis Recovery (MDM-AP)
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x000182d2 in ?? ()
(gdb) mon ver
Black Magic Probe v1.8.0-682-g12bda1e0, Hardware Version 6
Copyright (C) 2022 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb)
```

dragonmux, 2022-11-29:
Hi xobs, could we get you to provide debug traces (preferably by using BMDA built with `ENABLE_DEBUG=1` and run with `-v 5` to get the register read/write trace) for each firmware version so we can figure out how the attach routine for that part broke please?

xobs, 2022-12-01:
I'm on Windows, so I installed a WSL1 distro and rebuilt it in there. Curiously, `12bda1e0` didn't work when using `hosted` mode. Instead, the BMP stopped working much in the same way it's doing with `main`. I checked out 2b06f045c442c676ecd8bac515b54f4d7321c89f which did end up working.

[log-bad-d5e59b947608060a4e0cc62fb21d9c6dac023950.txt](https://github.com/blackmagic-debug/blackmagic/files/10129379/log-bad-d5e59b947608060a4e0cc62fb21d9c6dac023950.txt)
[log-good-2b06f045c442c676ecd8bac515b54f4d7321c89f.txt](https://github.com/blackmagic-debug/blackmagic/files/10129380/log-good-2b06f045c442c676ecd8bac515b54f4d7321c89f.txt)

perigoso, 2022-12-01:
Make sure you're not running into #636

xobs, 2022-12-07:
I've tried building under MSYS2 (or is it MINGW64? MSYS2 didn't build successfully). The "good" commit also doesn't work there.

[log-bad-19d9111a834e53b0be6bda1b1294c03a7f31431c.txt](https://github.com/blackmagic-debug/blackmagic/files/10172052/log-bad-19d9111a834e53b0be6bda1b1294c03a7f31431c.txt)
[log-good-12bda1e09f2c7b39d25e95e9632bd7a6aca33453.txt](https://github.com/blackmagic-debug/blackmagic/files/10172065/log-good-12bda1e09f2c7b39d25e95e9632bd7a6aca33453.txt)

dragonmux, 2022-12-07:
We'll look into build problems as a separate issue, though you'll be happy to know that this issue appears to be the same as another that was reported to us on Discord affecting STM32F0/F4/etc parts and which we've been able to reproduce locally. and we're investigating what's going wrong and how to possibly fix it

dragonmux, 2022-12-09:
So, a little update.. in trying to deep-dive this and locate the breaking commit, we've found that this issue is timing-dependant.

If you run GDB in batch mode, eg, `arm-none-eabi-gdb -batch -ex "tar ext :2000" -ex "mon tpwr en" -ex "mon s" -ex "att 1"` then the bug should not trigger.

If you run GDB normally and have a human amount of time occur between `mon s` and` att 1`, bug triggers (reliably)

xobs, 2022-12-09:
That would explain why I don't see it on Farpatch, which uses the same SWD code as BMP.

dragonmux, 2022-12-09:
Lovely, observing the bus also cures the problem.. so we are unable to get a logic capture of what's going wrong to dive into and debug

dragonmux, 2022-12-12:
Noting that we know BMDA is currently broken on this branch, please could you test fix/cortexm-swd-attach-regression and let us know if you still have issues attaching to your S32K148 device please

xobs, 2022-12-13:
It is able to attach and load programs with the fix/cortexm-swd-attach-regression branch.

closed 2022-12-28 by esden