blackmagic-debug/blackmagic #1328: Potential defect in `stm32f4_probe()`

stoyan-shopov, 2022-12-14:
https://github.com/blackmagic-debug/blackmagic/blob/5ac4378b77095326e7f096c0a440f1060200a63b/src/target/stm32f4.c#L234

Hi, team
I suspect there could be a defect with this code, in file `stm32f4.c`, function `stm32f4_probe()`, line 234:
```
		t->driver = stm32f4_get_chip_name(t->part_id);
		t->part_id = device_id;
```
I think these two lines of code should be interchanged, but I may be wrong.

I am getting an **STM32F411** device reported ig **gdb** as **STM32F2 M4**.
I happen to have some probe with old firmware, and that one reports in **gdb** **STM32F40x M4**

Feel free to close this issue if inappropriate

Thanks

perigoso, 2022-12-14:
Hi, you are correct in that that looks wrong, but it might be more appropriate to rewrite `t->driver = stm32f4_get_chip_name(t->part_id);` as `t->driver = stm32f4_get_chip_name(device_id);`

perigoso, 2022-12-14:
can you try #1329?

stoyan-shopov, 2022-12-14:
Thank you, @perigoso

I confirm that your change works

I just tried your change on an **STM32F411 blackpill** as a BMP host.
It successfully recognizes another **STM32F411 blackpill**:
```
(gdb) mon s
Available Targets:
No. Att Driver
 1      STM32F411 M4
```
I also tried to connect to an old **blackpill**, with an **STM32F401** part:
```
(gdb) mon s
Available Targets:
No. Att Driver
 1      STM32F401C M4
```

(For the record, with an **STM32F411 blackpill**, pressing the reset button takes the board into DFU mode - so, in order to actually reset the probe, power needs to be cycled. This has nothing to do with this change, just mentioning. I will have a look at the probe initialization code, maybe there is some discrepancy with the probes I use.)

Thanks a lot!

PR #1329 (2022-12-14): target/stm32f4: correct code used to look up part name
t->part_id was being used, but that code is unlikely to match the idcode used internally by ST leading to a bad look up, use device_id which is read from the internal ST register

## Detailed description

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/blackmagic-debug/blackmagic/blob/main/CONTRIBUTING.md) to this repository
* [x] It builds for hardware native (`mak [... cut]
closed 2022-12-14 by dragonmux