blackmagic-debug/blackmagic #1336: BMDA segfaults attempting to use command line flash operations on unknown chip

grevaillot, 2022-12-22:
Hi!

I've just attempted to use an stm32wb15 nucleo board, with embedded board st-link interface and a fresh build of BMDA (567b670ce67cefdf624638d35430c6599ba75c7e,  `make PROBE_HOST=hosted`), and it seems that command line flash operations are broken if probe failed due to unknown device. 

1.9.0rc0 tag shows the same behaviour

```
 % ./src/blackmagic -v8 -E
Black Magic Debug App v1.9.0-rc0-13-g567b670c
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 0483:3752 0670FF363730554157082027 STMicroelectronics
 STM32 STLink
Divisor for 4.0000 MHz is 0
Caching SELECT 0x02
Caching SELECT 0x00
Open AP 0
stlink_readmem from e000edfc to 0x7fff44a3bd3c, len 4
stlink_readmem from e000edf0 to 0x7fff44a3bd3c, len 4
stlink_readmem from e00ffff0 to 0x7fff44a3bca0, len 16
stlink_readmem from e00fffd0 to 0x7fff44a3bc60, len 16
stlink_readmem from e00fffe0 to 0x7fff44a3bc60, len 16
stlink_readmem from e00fffcc to 0x7fff44a3bc9c, len 4
stlink_readmem from e00ff000 to 0x7fff44a3bc9c, len 4
stlink_readmem from e000eff0 to 0x7fff44a3bc00, len 16
stlink_readmem from e000efd0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e000efe0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e000ed00 to 0x7fff44a3bbcc, len 4
stlink_readmem from e000ed88 to 0x7fff44a3bbcc, len 4
stlink_readmem from e000ed88 to 0x7fff44a3bbcc, len 4
stlink_readmem from e000ed7c to 0x7fff44a3bbcc, len 4
stlink_readmem from e0042000 to 0x7fff44a3bb8c, len 4
stlink_readmem from e0042000 to 0x7fff44a3bbac, len 4
Please report unknown device with Designer 0x20 Part ID 0x4940
stlink_readmem from e00ff004 to 0x7fff44a3bc9c, len 4
stlink_readmem from e0001ff0 to 0x7fff44a3bc00, len 16
stlink_readmem from e0001fd0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e0001fe0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e00ff008 to 0x7fff44a3bc9c, len 4
stlink_readmem from e0002ff0 to 0x7fff44a3bc00, len 16
stlink_readmem from e0002fd0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e0002fe0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e00ff00c to 0x7fff44a3bc9c, len 4
stlink_readmem from e0000ff0 to 0x7fff44a3bc00, len 16
stlink_readmem from e0000fd0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e0000fe0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e00ff010 to 0x7fff44a3bc9c, len 4
stlink_readmem from e0040ff0 to 0x7fff44a3bc00, len 16
stlink_readmem from e0040fd0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e0040fe0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e0040fcc to 0x7fff44a3bbfc, len 4
stlink_readmem from e0040fbc to 0x7fff44a3bbfc, len 4
stlink_readmem from e00ff014 to 0x7fff44a3bc9c, len 4
stlink_readmem from e00ff018 to 0x7fff44a3bc9c, len 4
stlink_readmem from e0043ff0 to 0x7fff44a3bc00, len 16
stlink_readmem from e0043fd0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e0043fe0 to 0x7fff44a3bbc0, len 16
stlink_readmem from e0043fcc to 0x7fff44a3bbfc, len 4
stlink_readmem from e0043fbc to 0x7fff44a3bbfc, len 4
stlink_readmem from e00ff01c to 0x7fff44a3bc9c, len 4
Open AP 1
stlink_readmem from f0000ff0 to 0x7fff44a3bca0, len 16
Open AP 2
Close AP 2
Open AP 3
Close AP 3
Open AP 4
Close AP 4
Open AP 5
Close AP 5
Open AP 6
Close AP 6
Open AP 7
Close AP 7
Close AP 8
Close AP 9
stlink_readmem from e0002000 to 0x7fff44a3bebc, len 4
stlink_readmem from e0001000 to 0x7fff44a3bebc, len 4
stlink_readmem from e000edf0 to 0x7fff44a3bebc, len 4
stlink_readmem from e000edf0 to 0x7fff44a3bebc, len 4
stlink_readmem from e000edf0 to 0x7fff44a3be7c, len 4
stlink_readmem from e000edf0 to 0x7fff44a3be7c, len 4
stlink_readmem from e000edf0 to 0x7fff44a3be7c, len 4
zsh: segmentation fault (core dumped)  ./src/blackmagic -v8 -E
```

```
Core was generated by `./src/blackmagic -v8 -E'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00005555ea60a962 in flash_done (f=0x0) at target/target_flash.c:103
103		if (!f->ready)
[Current thread is 1 (Thread 0x7f544d5d6740 (LWP 806387))]
(gdb) bt
#0  0x00005555ea60a962 in flash_done (f=0x0) at target/target_flash.c:103
#1  0x00005555ea60aaa2 in target_flash_erase (t=t@entry=0x5555eb629bf0, addr=4294967295, len=0) at target/target_flash.c:156
#2  0x00005555ea60df6e in cl_execute (opt=opt@entry=0x5555ea62fa40 <cl_opts>) at platforms/hosted/cli.c:555
#3  0x00005555ea5feed7 in platform_init (argc=<optimized out>, argv=<optimized out>) at platforms/hosted/platform.c:127
#4  0x00005555ea5f07fd in main (argc=<optimized out>, argv=<optimized out>) at main.c:34
(gdb) 
```

grevaillot, 2022-12-22:
i'll have a look at the issue myself once i've fixed the probe issue on wb15 :)

dragonmux, 2022-12-22:
Thanks for reporting this! It looks like something isn't doing a `NULL` check there as no Flash regions are defined when using the generic Cortex-M support, which leads to `NULL` being passed to the Flash handling routines (hence this crash). Unsure if this is strictly a BMDA-only bug or if one can trigger this in the firmware too.. pretty sure it's not a regression, but this is a serious enough bug in the new target Flash API that we'll mark it for v1.9

The stack trace is sufficient information for us to trace what's going on here, thank you :)

perigoso, 2022-12-22:
@grevaillot could you check if #1338 fixes the issue?

grevaillot, 2022-12-22:
> @grevaillot could you check if #1338 fixes the issue?

```
Black Magic Debug App v1.9.0-rc0-14-g4e184f41
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 0483:3752 0670FF363730554157082027 STMicroelectronics
 STM32 STLink
Divisor for 4.0000 MHz is 0
Please report unknown device with Designer 0x20 Part ID 0x4940
Erasure failed!
```

looks good :)

PR #1338 (2022-12-23): target_flash: fix missing null check on target_flash_erase

## Detailed description

self explanatory, see  #1336

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [x] I've read the [guidelines for contributing](https://github.com/blackmagic-debug/blackmagic/blob/main/CONTRIBUTING.md) to this repository
* [x] It builds for hardware native (`make PROBE_HOST=native`)
* [x] It builds as BMDA (`make PROBE_HOST=hosted`)
* [ ] I've tested it to the best of my ability
* [x] My commit messages  [... cut]
closed 2022-12-23 by dragonmux