blackmagic-debug/blackmagic #1337: stm32wb10 & wb15 targets are not detected

grevaillot, 2022-12-22:
Hi!

STM32WB support is not yet handling the part id used for wb10 and wb15 chips

```
Target voltage: 3.25V
JTAG scan found no devices, trying SWD!
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x4940 M4
```

```
 % ./src/blackmagic -E
Black Magic Debug App v1.9.0-rc0-13-g567b670c
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, JLink and libftdi/MPSSE
Using 0483:3752 0670FF363730554157082027 STMicroelectronics
 STM32 STLink
Divisor for 4.0000 MHz is 0
Please report unknown device with Designer 0x20 Part ID 0x4940
```

I'll have a look at the l4 driver to add them :)

dragonmux, 2022-12-22:
Thank you for reporting this! Hopefully you'll find the comments we left in the device ID enum help you figure out how to do that. We look forward to seeing the PR for this.

Given this is technically not a regression, but rather new part support, we will earmark this for v1.10.

grevaillot, 2022-12-22:
> Thank you for reporting this! Hopefully you'll find the comments we left in the device ID enum help you figure out how to do that. We look forward to seeing the PR for this.

yep, filling the blanks regarding detection was easy and debugging seem to be ok - now i need to dig a bit deeper regarding flash as, at least, flash pages size are different between wb1x and bigger parts, so it wont work out of the box.

PR #1339 (2023-01-30): stm32l4: add stm32wb1x parts detection.
stm32wb10 & stm32wb15, stm32wb family smallest parts are reporting as 0x4940 and were not detected (see #1337)

Add missing id, chip info and handle part specific 2kb page size (bigger chips have 4k pages)

Detection tested on BMDA + stlink against stm32wb15cc and stm32wb10cc chip, debugging and loading code checked against stm32wb15cc on a nucleo board.

## Your checklist for this pull request

* [X] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE_OF_CONDUCT.md)
* [X] I've read the [guidelines for contributi [... cut]
closed 2023-01-30 by dragonmux