blackmagic-debug/blackmagic #1345: Memory leak in cortexm_attach

walmis, 2022-12-28:
I've noticed latest (1.8x, v1.9.0-rc0) blackmagic leaks memory on my esp8266 port, traced it on hosted build and found out it most likely leaks here

this is from tag v1.9.0-rc0
```
==3083162== 7,248 bytes in 4 blocks are definitely lost in loss record 29 of 29
==3083162==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==3083162==    by 0x1161B3: cortexm_attach (cortexm.c:755)
==3083162==    by 0x12BD01: target_attach (target.c:185)
==3083162==    by 0x119685: handle_v_packet (gdb_main.c:551)
==3083162==    by 0x119685: gdb_main_loop (gdb_main.c:354)
==3083162==    by 0x11008E: main (main.c:44)
```

Steps to reproduce
```
target extended-remote localhost:2000
monitor swdp_scan
att 1
monitor swdp_scan
att 1
monitor swdp_scan
att 1
monitor swdp_scan
att 1
monitor swdp_scan
att 1

etc.
```
cortexm_detatch is never called which should free `t->tdesc`

dragonmux, 2022-12-28:
This was fixed in #1335 merged today. Please check again with latest main.

dragonmux, 2023-01-25:
As the PR that addresses this has been merged, we will close this issue. If the fix is insufficient, please let us know and we will re-open the issue and work with you to better solve it.

closed 2023-01-25 by dragonmux

walmis, 2023-02-01:
Thanks, this issue appears to be fixed.